OctoPrint/OctoPrint #4007: rjsmin minifier breaks some regex

FormerLurker, 2021-02-06:
**The problem**

1. Enable plugin bundling and minification in config.yaml
2. Load showdown.min.js (just needs to be included in any plugin's get_assets)
3. Restart OctoPrint and reload all javascript
4. Look in the console and see the *Uncaught SyntaxError: nothing to repeat* error.

I have dug in deep and it appears to be a problem with the minifier's handling of some regex that includes spaces.  This isn't an issue with OctoPrint, but rather the underlying minifier.  See this issue:  https://github.com/ndparker/rjsmin/issues/17

Here's a better description using a very similar js library to what I'm using: https://github.com/django-compressor/django-compressor/issues/1011

This also appears to only be a problem when minifying previously minified files.  If there is a way to skip min.js files somehow (might not be possible), that too would resolve this issue.  As far as I can tell, this issue has not been fixed in the rjsmin github.

You can grab the showdown.min.js file I'm using [here](http://showdownjs.com/) and give it a go if you'd like.

**Did the issue persist even in safe mode?**

I cannot test this issue in safe mode (state why below)

**Version of OctoPrint**

1.5.3

**Operating system running OctoPrint**

Both Windows 10 and OctoPi 0.17 (updated regularly)

**Printer model & used firmware incl. version**

NA

**Browser and version of browser, operating system running browser**

Latest chrome for windows dosen't throw the error, but latest FF does.  Tested in win 10 only.

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Here is the console error: ```Uncaught SyntaxError: nothing to repeat```

I can give you anything else you need.  Not sure what can be done here, if anything, unless min.js files can be omitted somehow.  Otherwise this is upstream.

foosel, 2021-02-08:
I gotta admit that I'm not really surprised that double minification would cause an issue at some point. 

>  If there is a way to skip min.js files somehow (might not be possible), that too would resolve this issue.

This is exactly what I've done on `maintenance`, see the above commit. If you could give that a whirl from your end and see if you can verify it solves the issue, that would be great!

FormerLurker, 2021-02-08:
Wow, that was fast!

> I gotta admit that I'm not really surprised that double minification would cause an issue at some point.

Yeah, it's not exactly surprising.  It probably should work, though, at least in theory.  Def easier just to skip them  completely.

I will try it out ASAP, but it might need to wait until the weekend unless work dies down somewhat.  We seem to be on opposite schedules, lol!

Thank you!

FormerLurker, 2021-02-08:
On, one more note, and it is unrelated to this issue:  The new issue forms are AWESOME!  Can't wait until they open the feature up!

foosel, 2021-02-08:
Yeah, I'm very happy about them myself :)

commit 2913eb5f (2021-02-08): 🥅 Only minify non-minified JS

If the file ends on ".min.js" we'll keep our hands off of it.

Closes #4007
closed 2021-04-27 by foosel