OctoPrint/OctoPrint #4008: Connection refused after update.

ebaystudio, 2021-02-07:
**The problem**

Refuses to connect wirelessly or at all really. I get this error in the terminal.

I tried the "Apply parity double open workaround" and that doesn't work either. 

Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Connecting"
Connected to: Serial<id=0xae3636f0, open=True>(port='/dev/ttyS0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
No answer from the printer within the connection timeout, trying another hello
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from "Connecting" to "Offline"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyS0, baudrate 115200
Changing monitoring state from "Opening serial connection" to "Error: Connection error, see Terminal tab"
Changing monitoring state from "Error: Connection error, see Terminal tab" to "Offline (Error: Connection error, see Terminal tab)"
Unexpected error while connecting to serial port /dev/ttyS0, baudrate 115200 from hook default: error: '(22, 'Invalid argument')' @ comm.py:_open_serial:3670

**Did the issue persist even in safe mode?**

Yes, it did persist

**Version of OctoPrint**

 OctoPrint 1.5.3

**Operating system running OctoPrint**

Python 2.7.13

**Printer model & used firmware incl. version**

Prusa MK3s 3.9.0-3421

**Browser and version of browser, operating system running browser**

Chrome

**Checklist of files to include below**

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[plugin_softwareupdate_console.log](https://github.com/OctoPrint/OctoPrint/files/5939668/plugin_softwareupdate_console.log)
[image]

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5939670/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/5939671/serial.log)

jneilliii, 2021-02-07:
@ebaystudio could you please enabled serial logging in OctoPrint's settings and try again. That logging is not enabled by default and the one you have provided is empty. I don't think it's going to be helpful in this case though. Can you provide some additional details as to how you are connecting your pi to the Prusa printer? I know there are a couple of approaches via uart or USB just need to know which you are using. If you are using USB the option on the printer for RPi Port needs to be disabled. If you are using a direct serial connection (not USB) then that option needs to be turned on.

cp2004, 2021-02-07:
This is probably more suited to the community forums, since it's not usually a problem in OctoPrint when the printer is not connecting. Usually USB/Printer/other hardware.

And as a side note, whenever you say 'after an update' always mention what you updated from, it helps us work out if anything remotely connected to the issue was changed.

ebaystudio, 2021-02-07:
Sorry, I don't know the version before the update. The current version is what's listed above. As for how it connects it's via the pins on the board and it is turned on in the settings. It's a PiW on the Ensy board of a Prusa MK3s. Its been working fine for about 10 months until the update today. As for how I connect to the pi it's wireless. 

[octoprint (2).log](https://github.com/OctoPrint/OctoPrint/files/5940033/octoprint.2.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/5940045/serial.log)
Here is the system info if it matters. 

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.150 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 1
env.hardware.freq: 1000
env.hardware.ram: 388599808
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi Zero W Rev 1.1
env.plugins.pi_support.octopi_version: 0.15.0PE
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 9.0.3
env.python.version: 2.7.13
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3

foosel, 2021-02-07:
The Zero and Zero 0w are not supported.

And yes, I agree, this is more suited to the community forums, please get in touch there.

closed 2021-02-07 by foosel

ebaystudio, 2021-02-07:
What is this?  "Zero and Zero 0w" 

Oh, wait so you're saying because I'm using a Pi-Zero its not going to work and I'll need to swap all 3 of my printers over to full-sized pi's.

 Again, it was working perfectly fine until the update, and why I opened the bug report. I noticed the warning about the Z last update and disabled the warning as suggested in the update.

ebaystudio, 2021-02-07:
So that's it, you just close it? no suggestion or solution?

fieldOfView, 2021-02-07:
The suggestion is to get help at the community forum.