OctoPrint/OctoPrint #4009: [Request]Don't check for updates while printing

MarkusNaeher, 2021-02-08:
**Is your feature request related to a problem? Please describe.**
Checking for updates while printing does not make sense, because (at least to my experience so far), updates require a OctoPrint restart. Restarting would kill a print.  
It was pure luck that I did my first update **not** while printing. And now I know that I have to be careful. But maybe other new users are not that lucky.

**Describe the solution you'd like**
I would like OctoPrint to disable update checks while printing.  
And (for performance reasons) maybe all other communication with the outer world as well, like checking for news in the blog etc. Maybe this should have a switch in the settingsl.

**Describe alternatives you've considered**
Sorry, I have no idea about alternatives.

**Additional context**
None.

foosel, 2021-02-08:
> It was pure luck that I did my first update **not** while printing.

It wasn't. OctoPrint will refuse to apply updates while printing, precisely because it would kill a print. 

The only thing it does while printing is retrieve update information, and it doesn't do that automatically in the background but only when triggered through a user action. That could be you loading the page for the first time after the update cache ran out, or you explicitly clicking on "check now". The performance overhead of that is minimal (it's primarily waiting for i/o from a network request). The same goes for announcement checks and similar things.

On the other hand, disabling all these things while printing, without a concrete reason to do so, would confuse people who (understandably) would question why something as querying data from a remote endpoint is forbidden while printing. Transparently reporting back on WHY something is disabled right now would be necessary and blow all this completely out of proportions regarding its usefulness.

So - this will not be implemented. As I said, updates are already prevented while printing.

closed 2021-02-08 by foosel

MarkusNaeher, 2021-02-09:
I don't know how I triggered it. Only thing I know for sure is that I did **not** klick on "check for updates".  During printing, I always keep my hands off the settings. :-)

I only switched between the main window tabs (Temperature / Control / GCode Viewer / Terminal / Timelapse ) and I've set check marks in a few "Suppress ..." options in the Terminal tab. And suddenly the "update available" notification popped up.

I have a second OctoPrint running on a different computer. There, I tried what happens when I confirm the update and learned that a plugin update requires restart.

I hope through these explanations you can see how I've got the impression that it is possible to kill a print accidentally.