OctoPrint/OctoPrint #4010: [Request] Setting: Display/Don't Display "Back to Top" arrow

KenLucke, 2021-02-08:
**Is your feature request related to a problem? Please describe.**
The annoying (to me) appearing and disappearing "Back to Top" arrow.  I know some people like it, but it's not as if the web page is dozens of screens long. The scroll wheel/scrollbar is perfectly capable of dealing with that short distance faster than finding that and clicking on it, and it blinking on and off as I scroll up and down is what annoys me the most. If it were a permanent button at the bottom, that would be one thing, but the constant appear/disappear routine is distracting.

**Describe the solution you'd like**
A  checkbox setting in either Features or Appearances that would allow a user to disable the Back To Top arrow.

**Describe alternatives you've considered**
Not many alternatives but live with it, really :)

cp2004, 2021-02-08:
There's a 'Back to Top' arrow? I have never seen one in OctoPrint - are you sure this isn't a plugin? I don't even think I've seen it in the code.

foosel, 2021-02-08:
There definitely isn't a back to top arrow. Just be either a plugin or a completely different product 😳

@KenLucke care to solve the mystery? 😂

KenLucke, 2021-02-08:
OK, I'll bet it's in Dashboard

Sorry.

closed 2021-02-08 by KenLucke

KenLucke, 2021-02-08:
https://user-images.githubusercontent.com/57936294/107292750-30c9e400-6a1f-11eb-89ac-a8d8520f77a1.mp4

jneilliii, 2021-02-09:
That's my floating navbar plugin...

jneilliii, 2021-02-09:
Or consolidated tabs maybe...

KenLucke, 2021-02-09:
I'm not using consolidated tabs, so it must be floating navbar