OctoPrint/OctoPrint #4011: All processing just stops and Octoprint appears to restart

KenLucke, 2021-02-09:
**The problem**

Twice now, same model, at relatively the same height, but resliced the second time, so the code is not the same, I have had a model just stop in the middle of the night, and Octoprint has apparently restarted and rebooted the printer in the process:  hotend and bed are cooled, nozzle is stuck in the model where it stopped and cooled down, OctoPrint shows connected but "reload now" dialog on main screen, no layer or height  information on the printer screen (which prevents me from continuing from that point), etc.  This is many, many hours into the print.  Printer/pi are on a UPS system, so power glitches are out of the question, and another printer right next to this one is still going strong with an even longer print.

I just installed a newer build of the firmware before this happened the first time, so I am ass/u/ming it has to do with that, but I need a clue what to be looking for in the build that could be causing this.

Can someone more familiar with following the logs than I, take a look and see what might have happened?  This morning's event must have taken place somewhere around 0430, as when I found it at 0445, the nozzle was cold, but the bed was still cooling off.

**Did the issue persist even in safe mode?**

I cannot test this issue in safe mode (state why below)

**Version of OctoPrint**

1.5.3

**Operating system running OctoPrint**

Mac OS 10.15.7

**Printer model & used firmware incl. version**

CR-10v3/Marlin 2.0.7

**Browser and version of browser, operating system running browser**

 browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:85.0) Gecko/20100101 Firefox/85.0

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

OP log: [octoprint(1).log.zip](https://github.com/OctoPrint/OctoPrint/files/5951519/octoprint.1.log.zip)

gcode file: https://www.dropbox.com/s/u9fr1ax0utp625b/Tavern_Mug_no._3_-_Mug_-_Threaded_aw.gcode.zip

serial log was not enabled

foosel, 2021-02-09:
You said you cannot test this in safe mode but didn't state why. Please do so, because frankly the log is somewhat full of third party plugin errors, especially complaints from TSD, but also regular webdav issues.

From what I see in the log, the server must have crashed around 4:19, because that is when it suddenly started again right after a trace back:

```
2021-02-09 04:19:04,508 - octoprint.util.comm - ERROR - Error while calling hook from plugin actiontrigger with action command notification 37% L=427/1543
Traceback (most recent call last):  
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2298, in _monitor
    params=action_params,  
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
TypeError: hook_actiontrigger() got an unexpected keyword argument 'name'
2021-02-09 04:17:09,086 - octoprint.startup - INFO - ******************************************************************************
2021-02-09 04:17:09,088 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-02-09 04:17:09,089 - octoprint.startup - INFO - ******************************************************************************
2021-02-09 04:17:15,528 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: offline
2021-02-09 04:17:15,528 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
2021-02-09 04:17:15,528 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
```

The fact that the system time also resetted could indicate a system restart, in which case whatever crashed, it crashed hard outside of OctoPrint. You might want to dig into `syslog`. From the looks of it I'm currently leaning towards this being not a case of an OctoPrint but a system crash.

KenLucke, 2021-02-09:
The only reason I said I could not try it in safe mode is that it's like 18 hours into the print to the approximate spot it failed.

That is appears to be a server crash seems consistent with the clues I could see.  I'm going to try reverting the firmware and see if it makes it through.  I'm burning a lot of filament on this one :(.   I appreciate the help, even though it may not be OP itself - thank you.

foosel, 2021-02-22:
I'm going to mark this as "not octoprint" and close this for now, since there's nothing actionable for me as far as I can see. If you find anything new however, e.g. in `syslog`, feel free to comment that here and I'll be happy to reopen if it actually DOES look like an OctoPrint core issue.

closed 2021-02-22 by foosel