OctoPrint/OctoPrint #4013: [Request] Feed and Flow rate is readable (from Marlin 1.9)

AndiP1234, 2021-02-09:
At http://octopi.local/?#control Octopi shows me a empty inputfield for feed and flow rate, at the info it will show, that it´s not possible to get the current values from the printer.

**Describe the solution you'd like**
with M220 and M221 the printer reports the current rate.
see at: https://marlinfw.org/docs/gcode/M220.html
M221 is also working with Anycubic Mega S Marlin 1.9

jneilliii, 2021-02-09:
I know I've seen this discussion somewhere before, possibly on the forums, but from what I remember for OctoPrint to consistently be able to handle these types of things, it would require that the firmware report that it is possible via the capabilities report (M114) somehow so it can be detected as possible to do. I don't think Marlin does that currently and is why this hasn't already been implemented. The other caveat, is if you change that on the printer's control panel itself the firmware does not tell the connected host software (OctoPrint in this case, but pronterface would have the same issue) what it was changed to if I remember correctly.

foosel, 2021-02-09:
> from what I remember for OctoPrint to consistently be able to handle these types of things, it would require that the firmware report that it is possible via the capabilities report (M115) somehow so it can be detected as possible to do

Correct. Otherwise I'd be spamming two commands to the printer needlessly. Additionally I'm missing a specification here about the output format (which will determine if that can even be parsed in without issues) and the fact that I'd need to constantly spam two commands to keep this stuff updated also doesn't exactly entice confidence that it would be a good thing to implement in the first place frankly. Pushed data would make more sense really, or a command with which I could query *all* of the current state, sd printing, temperatures, feed rate, flow rate, currently selected tool, fan speed and so on. It can't be that a host needs to send 5+ commands every couple of seconds just to stay in sync with the printer when it could just... push that kind of information. It already does this for temperatures and sd status, it's time to move away from the constant polling based communication that primarily saturates the already struggling serial interface.

foosel, 2021-05-03:
Still unclear on whether that is included in any kind of capability report, if not it won't be implemented. Requesting information.

mtaumike, 2021-05-30:
Validation info:

- Prusa MK3S+ responds to `M220` with no parameters with (3.9.3):
> Send: M220
> Recv: 100%

- Prusa MINI+ responds to `M220` with no parameters with (4.3.1):
> Send: M220
> Recv: ok

- Ender 3 V2 responds to `M220` with no parameters with (Marlin 2.0.7):
> Send: M220
> Recv: FR:100%

So, until everyone can play nice I don't see how this can be done yet in a fashion that works for **most** yet.

rsheasby, 2022-06-01:
Would it be possible to add a simple "firmware type" dropdown box to a printer's profile, allowing the user to pick what firmware they're running, and thereby support more features like this more universally?

Of course, it could default to a "generic" with the current, safe behaviour. But if the user selects "Marlin 2" for example, then octoprint would know that it *does* support M220 reading, and knows how to parse the response, so it can then poll for it every few seconds.

cp2004, 2022-06-05:
OctoPrint won't be implementing any polling for the fanspeed, it will just slow down an already busy serial communication with the printer. Auto reporting is the only way, and this should be detected from a capability report rather than asking the users to make a selection. Firmware versions become even more confusing when you find out that Creality versioned their Marlin 2.0 fork as 'Marlin 1.0'...

frogale, 2022-08-06:
OctoPrint could only remember the last value and update it when gcode sends it to the printer when printing. Why is the value removed when set in OctoPrint? I think RepetierServer does it the same way.

cp2004, 2022-08-06:
@frogale it is avoiding creating confusion when the value is updated on the printer without OctoPrint, which leads to two different values being shown.

frogale, 2022-08-15:
@cp2004 I use OctoPrint because I don't want to touch my printer at all. How many users plan to change print settings through the printer while using OctoPrint? Isn't it enough to just point out this eventual inconsistency?

cp2004, 2022-08-16:
It was done this way in the past and still caused regular questions through the support channels. That may be how you intend to use your printer, but there's a *lot* more OctoPrint users out there that do different things.

frogale, 2022-08-17:
I don't think this way is good for everyone. If I'm one of those users who does different things, I'd love to be able to set this behavior somewhere. It's a dealbreaker for me now and I have to use the feature-limited RepetierServer.

trayracing, 2023-07-26:
> kind of firmware support like this requires that OctoPrint can detect that it is supported,

or a checkbox in settings->features, like `Use M220 to query feed rate`. Or key off the presence of Marlin/etc that do support feed rate query. The key is to make a feature easy when it's applicable, and non-toxic when its not.

> Additionally, polling information like this continuously spams 

Just update it when the feed rate input box acquires input focus, and if unedited, clear it when it loses it. No polling needed.

jneilliii, 2023-07-26:
This plugin may be interesting for you guys: https://plugins.octoprint.org/plugins/marlinslider/

foosel, 2023-07-26:
> Just update it when the feed rate input box acquires input focus, and if unedited, clear it when it loses it. No polling needed.

It might take a bit to get this value back from the firmware, depending on what else is going on in the background (e.g. when a printjob is running just right now), so now we've got a UX issue because someone there IS a value in the field right after the user clicks in there, sometimes it is there when the use waits a second, sometimes it will take multiple seconds to show up. That's not going to work.

> Or key off the presence of Marlin/etc that do support feed rate query.

Which is *precisely* what a capability report does, so we are back at

> I've now been requesting information on whether this is part of the capability report that's part of the `M115` response for over a year, and still not an answer.

only that now it's been two years. You want this in OctoPrint? So do I, but I can't work with what I currently have from the firmware side in a way it doesn't cause issues with other firmware variants.

Add and/or document a capability report for this functionality, ideally make it autoreportable as well so no polling is needed, and we are golden.

trayracing, 2023-07-26:
>sometimes it will take multiple seconds to show up. That's not going to work.

That posits that anything not knowable instantly isn't worth knowing. You could have a "pending" icon. The only alternative for the user is to pause the print and manually entry the g-code to find out what the feed is. That potentially scars the print so that really doesn't work.

trayracing, 2023-07-26:
>This plugin may be interesting

Thanks. Will check it out. Had a print go awry last night because of non-default feed rate left set on a shared printer, so feed rate visibility is important.