OctoPrint/OctoPrint #4016: First layer(s) printed partially

wimg, 2021-02-11:
**The problem**

1. Printing a set of hinges, created in Tinkercad
2. Saved as an STL file, then created a .gcode file in Cura. Everything looks fine in the Cura preview, with the entire hinge on the first layer
3. When loaded in Octoprint, the first 3 layers print only part of the hinge (about 80%)

Logically, this leads to a bad print

**Did the issue persist even in safe mode?**

Yes, it did persist

**If you could not test in safe mode, please state why**

_No response_

**Version of OctoPrint**

1.5.3

**Operating system running OctoPrint**

OctoPi 0.18.0

**Printer model & used firmware incl. version**

Ender 3 v2

**Browser and version of browser, operating system running browser**

Chrome v88 in WIndows 8.1

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5966544/octoprint.log)
[image]
[image]

[hinge4.zip](https://github.com/OctoPrint/OctoPrint/files/5966571/hinge4.zip)

cp2004, 2021-02-11:
Could you enable the serial.log, reproduce the issue and then upload it?

This will mean that we can compare what OctoPrint told the printer with what the gcode says.

My suspicion is that there is some sort of communication issue, a serial.log would show this hopefully.

jneilliii, 2021-02-11:
Would probably be beneficial getting the entire octoprint.log as well from a fresh restart until you exhibit the issue. The one provided appears to be abridged.

wimg, 2021-02-11:
To be clear : what is actually printed is a 100% match with what the GCode Viewer shows (as seen in the attached screenshot). So I doubt it's a serial/printer issue, but a GCode interpretation issue ? Or does that happen through the printer ?

foosel, 2021-02-11:
There's no GCODE interpretation going on in OctoPrint. It sends to your printer what is in your file, 1:1 (third party plugins might change that, but you said it does also happen in safe mode). 

It would help to know what object is supposed to look like vs what your printer prints. I don't see a difference in your screenshots, and a quick load of the file you attached (thanks!) into S3D also shows nothing that immediately springs to eye.

cp2004, 2021-02-11:
The screenshot of Cura and OctoPrint's gcode viewer looked suspiciously the same to me. 

I'm not seeing an issue when the gcode is uploaded to OctoPrint or opened locally - it's the same.

Could you clarify what the actual issue you are seeing is? You say ' the first 3 layers print only part of the hinge' - but that the gcode viewer shows this, but I don't see it showing this.

foosel, 2021-02-11:
We keep on doing that it seems 😂🙌

cp2004, 2021-02-11:
What can I say, great minds think alike...

wimg, 2021-02-11:
I've added a different print I just did. In the zip you'll find :
- The gcode file (hinge2.gcode)
- A picture of the print, which shows a perfect print on one side, but no so good on the other (20210211_203234.jpg)
- The 5 first layers from Octoprint. I might not interpret this correctly, but it looks like in the first 3 layers the contours are not done completely ? (layer1-5.jpg)
- A picture of a similar (slightly larger) hinge, also showing a similar thing (20210211_203240.jpg)
- The serial.log and octoprint.log files

The bed was leveled right before the print.

[debug.zip](https://github.com/OctoPrint/OctoPrint/files/5968004/debug.zip)

foosel, 2021-02-12:
That looks like a classic bed adhesion problem to me and not a software issue.

foosel, 2021-02-15:
Closing because it looks like an adhesion issue and the GCODE viewer matches what is being printed, so even if it is not an adhesion issue, it's then a slicing issue. You are welcome to get in touch on the [community forums](https://community.octoprint.org) to get help debugging this further.

closed 2021-02-15 by foosel