OctoPrint/OctoPrint #4020: Could not update printer profile

dchamb55, 2021-02-14:
**The problem**

Attempting to create a custom bounding box for a Prusa MK3S profile. 
I go to the "Print bed & build volume" tab, click the "Custom bounding box" box and enter -3 for Y coordinates minimum. Then click "Confirm" and the rotating circle on the Confirm button never goes away.

If I press Abort and go back to printer profiles, the edit profile button is greyed out.

Restarting the server is the only way to get the edit profile button to return. During shutdown this pops on the screen:

Could not update profile
There was unexpected error while updating the printer profile, please consult the logs

I tried disabling the Spaghetti Detective plugin and trying again, to no avail.
I also disconnected Octoprint from the Prusa and tried again, to no avail.

**Did the issue persist even in safe mode?**

Yes, it did persist

**If you could not test in safe mode, please state why**

_No response_

**Version of OctoPrint**

OctoPrint 1.5.3 

**Operating system running OctoPrint**

OctoPi 0.17.0

**Printer model & used firmware incl. version**

Prusa i3 MK3S, Firmware version 3.9.2

**Browser and version of browser, operating system running browser**

Chrome Version 88.0.4324.150 (Official Build) (64-bit); Windows 10 64-bit

**Checklist of files to include below**

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[
[image]
[image]
[octoprint (8).log](https://github.com/OctoPrint/OctoPrint/files/5978140/octoprint.8.log)
[serial (2).log](https://github.com/OctoPrint/OctoPrint/files/5978148/serial.2.log)

](url)

kantlivelong, 2021-02-15:
Hi @dchamb55 ,

The js console log would be helpful here. Could you provide?

foosel, 2021-02-15:
I agree, especially since I cannot reproduce this here at all.

dchamb55, 2021-02-15:
I am not certain what the js console log is called or where it is located. I don't see it on the logging screen.

I thought about creating a new profile to see if it would take the bounding box. I will give it a try and let you know.

Dale

foosel, 2021-02-15:
> I am not certain what the js console log is called or where it is located

https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers

dchamb55, 2021-02-15:
[octopi-1613405651205.log](https://github.com/OctoPrint/OctoPrint/files/5982851/octopi-1613405651205.log)
Here is the js console. 

I created a new printer profile with the custom bounding box enabled and it saved it. I closed the connection to the printer and chose the new profile, reconnecting and saving the connection settings. Then I could delete the old profile. I think my original problem is taken care of, but it still doesn't explain what happened when I tried to update the original profile. Perhaps the js console log might shed some light?

Dale

foosel, 2021-02-22:
Sounds like there might be something going on with the original profile. Certainly good to know that it only happened with that, because that means it's not a general problem. I'll tag it as unreproduced for now and see if I can maybe find a way to trigger it, or maybe someone else chimes in.

foosel, 2021-03-09:
I tried reproducing this with a fresh OctoPrint instance, but I can't. Adjusting the bounding box works just fine, regardless of which profile I try it on. Marking it as unreproduced for now.

foosel, 2023-01-16:
No re-report in well over a year, closing this. If it is indeed a bug still present in current versions, I'm sure at some point we'll see it reported again, hopefully with better reproduction ability.