OctoPrint/OctoPrint #4021: Force disable login page/Documenting OpenID-Connect based login configuration

szymonrychu, 2021-02-15:
**The problem**

In my home lab I'm running octoprint behind OIDC combined of kong and keycloak (total overkill, I know, but this duo guards more resources than this). Therefore I want to disable login screen in octoprint (currently I have to login twice).
I've tried two recommended approaches:
1. editing `config.yaml`:

```
accessControl: 
  autologinAs: admin 
  autologinLocal: true 
  localNetworks: 
    - 0.0.0.0/0
```

2. Installing `OctoPrint-ForceLogin` and trying to set it up with it (which does the same thing AFAIK).

Both methods doesn't get me where I want to get. 

That's why I believe that method of disabling login screen (or setting up openid-connect based login) and proper documentation around it should be available openly. OpenID-connect/oauth2/SAML offer more security/flexibility over what's built in the Octoprint. Moreover such documentation would be great to enable integrations with facebook/github/google.

Also I really don't believe that the login functionality secures anything in that regard. Current octopi installation doesn't offer valid TLS encryption for it's web-interfaces. Credentials necessary to operate Octoprint can be easily sniffed out without it.

**Did the issue persist even in safe mode?**

Yes, it did persist

**If you could not test in safe mode, please state why**

_No response_

**Version of OctoPrint**

1.5.3

**Operating system running OctoPrint**

raspbian (official octopi image)

**Printer model & used firmware incl. version**

N/A

**Browser and version of browser, operating system running browser**

Firefox 85, Chrome 88.0.4324.150 (probably others too)

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5980168/octoprint.log)

foosel, 2021-02-15:
Your use case should be covered by a combination of `accessControl.(trustBasicAuthentication|checkBasicAuthenticationPassword|trustRemoteUser|remoteUserHeader|addRemoteUsers)`, please check and confirm. See also https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control

I'm always happy about contributions, especially with documentation improvements.

> Current octopi installation doesn't offer valid TLS encryption for it's web-interfaces.

It ships with a self-signed certificate and that's about the only thing that's possible to ship with considering that there's no way to set up a browser-trusted certificate for local, self hosted, end user installed software, unless one goes the route of plex and sets up a convoluted, expensive and frankly for OctoPrint unobtainable DNS infrastructure that relies on DNS rebind attacks to work.

>     1. Installing `OctoPrint-ForceLogin` and trying to set it up with it (which does the same thing AFAIK).

I don't know what you want with that plugin or where you got the impression it might help, that has been superseeded by the built-in login mechanism. Uninstall it again please. If you could tell me what made you install this, I'd be happy so I can fix whatevern guide/docs/whatever needs fixing, if it's under my control.

szymonrychu, 2021-02-15:
Let's start from the bottom:
> I don't know what you want with that plugin or where you got the impression it might help, that has been superseeded by the built-in login mechanism. Uninstall it again please. If you could tell me [...]
There are numerous tutorials on how to disable this feature. Some of them ask to edit configuration file, others send to this particular plugin. At the same time the plugin is still available through manager, so maybe it's good idea to make obsolete plugins appear differently, than others? (good idea for separate bug/improvement)

> It ships with a self-signed certificate and that's about the only thing that's possible to ship with considering that there's no way to set up a browser-trusted certificate for local, self hosted, [...]

Unless you provide tutorial on how to import private part key of the cert into the browser (generally- system's trust-store), the tls handshake doesn't happen and such TLS doesn't add much besides another prompt user has to go through (besides login page).
When it comes to seting up more reliable DNS/TLS combo- it is possible, but will require a bit of admin/ops-ish experience. It can be done for free using combination of https://www.duckdns.org/ and certbot (example tutorial: https://forum.openwrt.org/t/duckdns-lets-encrypt-certificates-made-simple/38931/3 ).

The configuration you've mentioned:

> Alternatively take a look at https://docs.octoprint.org/en/master/features/accesscontrol.html#if-you-are-using-a-vpn-and-your-setup-absolutely-requires-disabling-internal-octoprint-access-controls

```
While access controls can no longer be disabled in OctoPrint 1.5+, this can be approximated by an Autologin configuration that automatically logs in all users, that is by using subnets that match all possible IP addresses. By specifying the 0.0.0.0/0 subnet (for IPv4) and ::/0 for IPv6 in the AutoLogin configuration, you can achieve this. This configuration is permitted, but highly, highly discouraged.

Please don’t do this. You will almost certainly regret it. You alone are responsible for your actions.
```

is very hard to find and didn't work for me or is poorly described. I tried (see below) but it doesn't work. It would be easier with example configuration at least, especially that other docs doesn't mention any config.yaml entry about ` AutoLogin configuration`, or it's not clearly written.

 > Your use case should be covered by a combination of accessControl.(trustBasicAuthentication|checkBasicAuthenticationPassword|trustRemoteUser|remoteUserHeader|addRemoteUsers), please check and confirm. See also https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control

In result of last two suggestions I tried two configs- both didn't work for me.

current configuration:
```
accessControl:
    autloginAs: admin
    autologinAs: admin
    autologinLocal: true
    enabled: false
    localNetworks:
    - 0.0.0.0/0
    salt: <redacted>
```

previous configuration:
```
accessControl:
    autloginAs: admin
    autologinAs: admin
    autologinLocal: true
    enabled: true
    localNetworks:
    - 0.0.0.0/0
    salt: <redacted>
```

Please advise.

foosel, 2021-02-15:
> There are numerous tutorials on how to disable this feature. Some of them ask to edit configuration file, others send to this particular plugin.

Which ones and where? I need links please. Keep in mind though that anything that's not under an octoprint.org domain is not official and cannot be kept up to date by me.

> At the same time the plugin is still available through manager

It's not published on the plugin repository, it's not pre-installed and the [source code repository](https://github.com/OctoPrint/OctoPrint-ForceLogin) is both archived, has a note that this functionality is now included and that the code is merely provided for historic reasons and "unmaintained" tags, so I need clarification about "readily available" here.

> Please advise.

I can't because I don't know your specific setup, can't reproduce it and can't spend hours on setting up your specific software chain either just to figure this out, sorry, that simply doesn't scale at a project this size. OctoPrint has configuration options to allow you to automatically login users based on various HTTP headers or client addresses. That should cover cases like yours. However, that requires the correct configuration, and I cannot tell you what that should be because I don't know how you've set up stuff in front of OctoPrint.

The config you shared above SHOULD disable the auto login reliably. If it doesn't for you then I'd need more details on your reverse proxy setup in front of OctoPrint, because then either there is a bug or this is misconfigured and not transporting the client IP as needed by this feature to work. 

> is very hard to find

Where should it be? Where did you look? It's literally located in the documentation's section on access control.

> [missing documentation, documentation needs to improve, etc]

I agree, there is a lot of room for improvement there. That is something that I need help from the community however. I cannot anticipate every single specific workflow someone might want to implement, I cannot advise on every single software combination under the sun, I have a way too development focused point of view to be able to provide good end user documentation (and have *repeatedly* pointed this out in OctoPrint On Air and asked for help without much response) and I frankly also have my plate full with the project as a whole way past 100% already. 

This is an open source project. If things need improvement, then please try your best to contribute to improve them. Merely pointing out that stuff "could be better" does sadly do nothing to make it better when the available developer resources are stretched razor thin already.

foosel, 2021-02-15:
> Unless you provide tutorial on how to import private part key of the cert into the browser

Almost forgot this. I don't see why OctoPrint's documentation should have to explain this, especially since it is browser and OS dependent. If someone wants to do this, they should be capable of doing a simple internet search to figure it out more reliably then OctoPrint's docs could ever explain to them.

szymonrychu, 2021-02-15:
Ok- again from the bottom:

> Almost forgot this. I don't see why OctoPrint's documentation should have to explain this, especially since it is browser and OS dependent. If someone wants to do this, they should be capable of [...]

I mean- if Octoprint (or octopi in that regard) requires complementary setup to happen to serve it's purpose (TLS encryption of traffic on login page created for security purposes) I believe it would be reasonable to at least mention necessity of additional setup steps somewhere in documentation in order to take full advantage out of it (or any advantage in that regard- since without this self-signed certs are untrusted and the traffic- containing credentials- is unencrypted).

> This is an open source project. If things need improvement, then please try your best to contribute to improve them. Merely pointing out that stuff "could be better" does sadly do nothing to make it better when the available developer resources are stretched razor thin already.

I would love to <3 thank you for invitation. At the same time I don't see how. In my opinion there should be easier way to disable this whole module and I believe the problem relies in forcing people to use it with current state of documentation/project. At the same time I see everywhere that this AccessControl functionality is mendatory and I still don't know how to turn it off properly. When It's resolved I'm happy to extend the documentation with proper info and examples of configuration.

> Where should it be? Where did you look? It's literally located in the documentation's section on access control.

I've used links you've provided and I agree, everything is on the same page and simple `ctrl+f` would eventually get me the notes you've mentioned. At the same time the notes doesn't write explicitly what should be set and how to disable/bypass login page. That's why I believe this particular information is hard to find/not easy to understand and not fully working.

> I can't because I don't know your specific setup, can't reproduce it and can't spend hours on setting up your specific software chain either just to figure this out, sorry, that simply doesn't scale [...]

I agree with you, such setup would be too complex to set-up in order to replicate issue I have. At the other hand I don't believe the issue is there. And at the same time I need to apologize you because the log I've attached originally has the information (necessary to validate my claims) about IPs redacted.

At the same time I can see in my file, that Octoprint captures proper publicly available IPs, therefore I believe my proxy is transparent in that regard and necessary IPs are available for Octoprint for autologin to work properly (I'm logging in from two locations - work and home, both IPs are present in logs and are OK). I really don't want to upload log file with my public IPs, so hopefully you can trust me on that. At the same time if there is a way to increase verbosity of requests in octoprint and send out headers/cookies informations to logs, I'm happy to do that to help out with debugging process (again- with some redaction done, so my systems won't get compromised).

> It's not published on the plugin repository, it's not pre-installed and the source code repository is both archived, has a note that this functionality is now included and that the code is merely [...]

Sorry I've meant "AutoLoginConfig" plugin, which is noted in the repository and allows to alter options regarding authentication with a promise of accepting CIDRs https://plugins.octoprint.org/plugins/autologin_config/ . AFAIK the plugin relies of the config.yaml options from the official documentation and CIDR I've provided is fine. With that said I believe underlying mechanism might not work properly/as expected.

> Which ones and where? I need links please. Keep in mind though that anything that's not under an octoprint.org domain is not official and cannot be kept up to date by me.

True. I was desperated and started searching outside of octoprint.org pages. I'm sorry on that.

cp2004, 2021-02-15:
This guide may be of use to you to setup and trust self-signed certificates:

https://community.octoprint.org/t/setup-a-new-self-signed-ssl-certificate-on-octoprint-enable-https/30256

I created the AutoLoginConfiguration plugin, and it does exactly what you mention - it just validates and sets the correct configuration for you, to save typos and wrong yaml syntax from causing people lots of frustration. It doesn't do more than just configure it for you. And it is *completely* different from the old ForceLogin plugin, and they do the exact opposite of each other. That was just a little confusing to read about...

From your log, the IPs are coming through like this:
```
::ffff:<ip discarded>
```

I'm no expert in networking, but the additional `::ffff:` could be what's upsetting it. I don't know how to define something that would match here, but it might be that you have to look at the IPv6 range instead of IPv4.

szymonrychu, 2021-02-15:
> I'm no expert in networking, but the additional ::ffff: could be what's upsetting it. I don't know how to define something that would match here, but it might be that you have to look at the IPv6 range instead of IPv4.

I will google it and check whether it's relevant.

At the other hand more verbose logging in main Octoprint, that would write out headers to file might be useful also for other issues in the future. :)

foosel, 2021-02-15:
Well spotted @cp2004, that detail escaped me indeed and I also overlooked that the v6 version was missing from the config example there. I would suggest to try adding both ipv4 AND ipv6 to the network definitions, as also explicitly mentioned in the linked docs.

In general though, to me it sounds like having whatever reverse proxy you use in front of OctoPrint that takes care of your authorization set an according header and evaluating THAT in OctoPrint via the aforementioned additional access control config settings would be the clean approach here, as it wouldn't open up the whole instance on the network, but that's down to personal preference.

---

Now, to respond to some of the other points:

> I mean- if Octoprint (or octopi in that regard) has a feature, that requires complementary setup to happen to serve it's purpose (TLS encryption of traffic on login page created for security purposes)

OctoPi, yes. OctoPrint doesn't do TLS termination, and it's not its task either. 

It doesn't really require that though. All it requires you is to accept that your browser can't confirm the identity of the certificate (which is logical as it's self-signed), it will still encrypt the traffic just fine, but will not show you a green lock because it can't confirm that *your* octopi host is the *only valid* octopi host (and that's as designed). Does it do that in a very scare manner? Yes, indeed. Can I fix that? Nope. Though I have certainly climbed on a soapbox in public about it a bunch of times, alas, browser vendors are entirely deaf to the troubles of open source end user facing projects that have user bases consisting primarily of people who don't know what a CA is, don't own a domain and so on.

I have never added one of OctoPi's certificates to my various browsers, I always just click "Accept" and that's that. Traffic encrypted. Not secure against MITM attacks, yes, but as secure as one can get it without having to own a domain and/or setting up one's own private CA.

> At the same time the notes doesn't write explicitly what should be set and how to disable/bypass login page. That's why I believe this particular information is hard to find/not easy to understand and not fully working.

It does write it as explicit as possible without making it so that users who do *NOT* understand the risk will just copy paste it without understanding what it actually does, and then once again get their naively port forwarded instances owned for which they then blame me (or for which cases I then get the flood of mails from security researchers who want to report "open instances they found on shodan"). Believe me, I've tried everything else BUT enforcing access control altogether before I did this step. It didn't work, people were still getting hurt, and frankly I was sick and tired of constantly getting blamed for that. There's still ways to get out of it, but they require some putting together of two pieces of information (how to define networks for autologin, and which networks to put as universal wildcards) so that people need to stop and think at least for a minute or two before they potentially shoot themselves into their collective feet.

> Sorry I've meant "AutoLoginConfig" plugin

Well, that certainly explains the utter confusion on my part here 😂

szymonrychu, 2021-02-15:
Heey!
the hint about v6 IP solved it!
```
accessControl:
    autloginAs: admin
    autologinAs: admin
    autologinLocal: true
    enabled: false
    localNetworks:
    - 0.0.0.0/0
    - ::/0
    salt: <redacted>
```

allowed me to bypass the login page entirely!
Thank you for help!

At the same time I understand your initial frustration around this topic. I was another person asking for impossible- conform to opposing requirements and needs. Yours who doesn't want to get the blame for someone else's mistake of running Octoprint openly and others who doesn't want to need to log into it. 

Thanks for supporting my case anyway!

As said I will update docs (once I find those) with the necessary `- ::/0` (or something resembling home networks- if it's even possible), or add note there to set only v4 networking for now. 

Thank you again!

foosel, 2021-02-15:
You're welcome and thanks for understanding. Also, given that this is already documented as far as I want it to be documented (due to the aforementioned "force people to think" requirement), I don't think there's any doc action needed here:

[image]

In any case, doc sources are here: https://github.com/OctoPrint/OctoPrint/tree/master/docs, explanation on how to build docs is included here: https://docs.octoprint.org/en/master/development/environment.html

szymonrychu, 2021-02-15:
Then I guess the topic can be closed. Docs are there (in-place), I'm satisfied with the result of this conversation.
Thank you! You and your work is awesome!

closed 2021-02-15 by szymonrychu