OctoPrint/OctoPrint #4022: Passive auth requires a cookie

ctacke, 2021-02-16:
**The problem**

Octoprint 1.5.3
Raspberry Pi 4

Attempting to authenticate using an API key fails unless a cookie is first generated using a user/password authentication.  This reproduces with just Postman (i.e. no code required to show it).

I have 1 user.  User is an admin.

- Use postman to authenticate with a username/password by POST to /api/login
- a `session_P80` cookie is generated (or P443 if using ssl)
- use postman to authenticate passive with API key
- a full auth response is returned with a `session` value
- delete the cookie
- use postman to resend the same request : authenticate passive with API key
- auth will return only a `needs` array.  Will not return a session again until you auth with user/password

**Did the issue persist even in safe mode?**

Yes, it did persist

**If you could not test in safe mode, please state why**

_No response_

**Version of OctoPrint**

1.5.3

**Operating system running OctoPrint**

Raspbian (raspberry Pi OS)

**Printer model & used firmware incl. version**

Lulzbot TAZ 6

**Browser and version of browser, operating system running browser**

Not using a browser.  Postman only

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5991041/octoprint.log)

shadycuz, 2021-02-20:
I can verify this exists in `Version 1.6.0.dev333+ga628549ad`

Here are my debug logs:

```
2021-02-20 10:44:14,486 - octoprint.server.util.flask - INFO - Passively logging in user shadycuz from ::1
2021-02-20 10:44:14,486 - octoprint.access.users - INFO - Logged in user: shadycuz
2021-02-20 10:44:14,492 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {'username': 'shadycuz'})
2021-02-20 10:44:14,492 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f31b00732e0>>
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-02-20 10:44:14,493 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-02-20 10:44:14,494 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-02-20 10:44:14,494 - octoprint.plugin - DEBUG - Calling on_event on tracking
2021-02-20 10:45:29,130 - octoprint.events.fire - DEBUG - Firing event: UserLoggedIn (Payload: {'username': None})
2021-02-20 10:45:29,130 - octoprint.events - DEBUG - Sending action to <bound method PrinterStateConnection._onEvent of <octoprint.server.util.sockjs.PrinterStateConnection object at 0x7f31b00732e0>>
2021-02-20 10:45:29,131 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2021-02-20 10:45:29,131 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2021-02-20 10:45:29,131 - octoprint.plugin - DEBUG - Calling on_event on announcements
2021-02-20 10:45:29,131 - octoprint.plugin - DEBUG - Calling on_event on file_check
2021-02-20 10:45:29,132 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2021-02-20 10:45:29,132 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2021-02-20 10:45:29,132 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2021-02-20 10:45:29,132 - octoprint.plugin - DEBUG - Calling on_event on tracking**
```

I will poke around and see if I can fix this.

shadycuz, 2021-02-20:
@ctacke 

After poking around for a bit I actually don't think this is a bug.  I'm new too the API and had to read through the docs. The section I found most interesting was on [authorization](https://docs.octoprint.org/en/master/api/general.html#authorization). 

After reading through the source code for passive logins that's when I noticed it doesn't actually read the value of the `passive` field. It only checks if it exists. The documentation for `login/api` can be confusing if you skipped over the `authorization` section like I did 😊. 

I'm guessing you and I both read the documentation for login. Then we both did this:

```json
{
	"passive": "F1r9hWwCRmC3wiisETOJ"
}
```

Then what we got back was an `AnonymousUser`
```
2021-02-20 11:14:50,805 - octoprint.server.util.flask - INFO - User is AnonymousUser(groups=[Group("guests", "Guests", description="Anyone who is not currently logged in", permissions=[], default=False, removable=False, changeable=True)])
```

Which is why we both got back the "needs" json body. 

I think you are supposed to take your token and instead send it as a header.

[image]

Now send your post body like this:

```json
{
	"passive": ""
}
```

You will get back the expected response body:

```json
{
    "_is_external_client": false,
    "_login_mechanism": "apikey",
    "active": true,
    "admin": true,
    "apikey": null,
    "groups": [
        "users",
        "admins"
    ],
    "name": "shadycuz",
...
}
```

I do think we could improve the documentation for the `login` endpoint. Let me know if this fixes your issue and if it does one of us can clean up the documentation.

foosel, 2021-02-22:
Can indeed confirm what @ctacke outlined:

```
$ curl -X POST "http://localhost:5000/api/login?passive=true&apikey=5a73835b4c040b589d64f92a4b92d451"
{
  "_is_external_client": false,
  "needs": {
    "group": [
      "guests"
    ]
  }
}
```

So there is a confirmed bug here, this should return the user associated with the API key (an admin user), not an anonymous one. I'm pretty sure that worked at some point, so I need to dig into why it no longer works.

foosel, 2021-02-22:
Never mind, I have to correct myself, I was looking at a deactivated key.

```
$ curl -X POST "http://localhost:5000/api/login?passive=true&apikey=B8C426309A4C4C80A256937545E98F42"
{
  "_is_external_client": false,
  "_login_mechanism": "apikey",
  "active": true,
  "admin": true,
  "apikey": "D92C34FFE9224124A077131453D520CC",
  "groups": [
    "admins",
    "users"
  ],
  "name": "test",
  "needs": {
    "group": [
      "admins",
      "users"
    ],
    "role": [
      "plugin_action_command_prompt_interact",
      "admin",
      "status",
      "timelapse_delete",
      "plugin_pluginmanager_install",
      "control",
      "plugin_firmware_check_display",
      "plugin_softwareupdate_configure",
      "print",
      "plugin_backup_access",
      "timelapse_download",
      "plugin_appkeys_admin",
      "plugin_logging_manage",
      "connection",
      "gcodeviewer",
      "plugin_announcements_read",
      "settings_read",
      "plugin_action_command_notification_show",
      "plugin_softwareupdate_check",
      "files_select",
      "plugin_softwareupdate_update",
      "files_download",
      "timelapse_admin",
      "settings",
      "files_delete",
      "files_upload",
      "system",
      "plugin_action_command_notification_clear",
      "monitor_terminal",
      "plugin_announcements_manage",
      "files_list",
      "plugin_pluginmanager_manage",
      "slice",
      "webcam",
      "timelapse_list"
    ]
  },
  "permissions": [],
  "roles": [
    "user",
    "admin"
  ],
  "session": "60A838673B8445F4BB08C409F47957F2",
  "settings": {
    "interface": {
      "language": "_default"
    }
  },
  "user": true
}
```

So I can't reproduce this after all.

foosel, 2021-03-09:
@ctacke ping? :)

closed 2021-05-03 by foosel