OctoPrint/OctoPrint #4023: [Request] Printer USB port re-connect by Vendor / Device ID

SaKiEQ, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
While using PSU control, the printer obviously looses USB connection to Raspi. When powering back-up I get every once in a while a miss match between chose USB port and other USB devices that are connected to the same Rapi. 
Depending if other devices where powered-up before or after the printer, or even if the Raspi was power-cycled between, the remembered port, eg. 'dev/ttyUSB0' does not always match and Octoprint connects to a port occupied by another device. 

**Describe the solution you'd like**
For my own USB devices and scripts I am running under Octoprint, I am matching port ID to Vendor/Device ID when scanning the USB ports for connection. This way I can make sure, that regardless of the numbering hierarchy my scripts connect every time to the correct port and do not interfere with the printer port. 

pid="0403"
hid="6001" 
my_comm_port = None 

ports = list(serial.tools.list_ports.comports()) 
for p in ports: 
	if pid and hid in p.hwid: 
		my_comm_port = p.device 

try:
        #open serial port
        ser = serial.Serial(my_comm_port)

I believe if Octoprint would memorize the Vendor/DeviceID and/or device serial, it would become more easy to auto reconnect reliably on re-start of PSU/USB connection. 

**Describe alternatives you've considered**
There are no real alternatives that I could see, since USB devices and Port-IDs are pretty dynamic under Linux.

**Additional context**
Using the PSU plugin / function obviously powers down the control board of the printer in most cases, hence the USB connection gets lost.
On systems that have more then just the printer connected via USB this leads to problems on reconnecting. In my case I run a controller for all the enclosure features (Temp, Humidity, Fan, Lights) on a Arduino connected via USB to the same host. 

When rebooting the Pi, the PSU controls goes low, eg. controller looses power, while the other USB device powers-up relatively quick. Who ever comes back online first occupies dev/ttyUSB0. 

Blacklisting ports in Octoprint does only help if the other devices would remain at all times static, to keep only the printer port whitelisted - now that is not always the case.

jneilliii, 2021-02-18:
Unfortunately, this would only work in linux, and not in the other OSs that OctoPrint supports. Chris Riley even mentioned recently that doing udev rules with printers/cameras that have serial numbers is the best option.

https://youtu.be/FIPbiImixZM

SaKiEQ, 2021-02-18:
The backend / OS is surely a problem,  but the USB device info is OS agnostic, only the method on how its retrieved depends on the serial / USB backend used. The device info structure on the other hand is pretty standardized. 
Granted a lot of UARTs/FTDIs use generic IDs, like all Arduino USB ports will more or less show the same info, more 'consumer' devices will have a unique serial or hardware ID too. 

In my opinion the wast majority of people are not running 2 or more of the same printer on one Pi. The example of 2 USB cameras is most likely even pushing the limits on computing power, paired wirth Octoprint and other services, making it an edge case, 

Most people will have a mouse, keyboard, printer connected - which have unique enough VID (vendor ID) and PID (product ID) to make a significant difference on a day to day basis vs. the current manual/auto selection of port by port path. 

Binding a VID:PID to a printer profile should be possible cross platform, only the functional part of port path / ID is different, but that is different in any case from OS to OS.

foosel, 2021-02-19:
The problem here is that it would complicate the whole auto connect and connection management code by a LOT. So this isn't simple by far and is something I definitely will not take into consideration for 1.x, it is simply not feasible. However, I will take it into consideration for 2.0.

In the meantime, this *could* actually be done quite easily through a plugin.

SaKiEQ, 2021-02-19:
Ok I get that, thanks for the clarification. So I did the right thing - to modify my own devices and 'drivers' so that Octoprint stays in the clear :). 

As I understand AutoConnect does work only on server start, but not printer start - which is the case when using PSU controls.

It is what it is... at the end my solution worked for my self for the time being. I just thought I asked the question, since I did not dig deep enough into the bowels of Octoprint to see how the connection is handled.