OctoPrint/OctoPrint #4024: [Potential Bug] Wrong error message when software update is blocked by undervoltage

cp2004, 2021-02-18:
**The problem**

Seeing a few instances of the software update plugin failing to show the correct error message when undervoltage states are present, it shows 'Update not started. Is it already active? Please consult octoprint.log for details.'

Links:
https://discord.com/channels/704958479194128507/708230190462926898/811727359601541180
https://community.octoprint.org/t/plugin-updates-cant-be-installed/30811

There have been others, but these are two most recent I can remember.

The one in discord mentioned the undervoltage state was not showing up in the navbar either.

**Did the issue persist even in safe mode?**

I cannot test this issue in safe mode (state why below)

**If you could not test in safe mode, please state why**

Yet to investigate/reproduce

**Version of OctoPrint**

1.5.x

**Operating system running OctoPrint**

Raspberry Pi / OctoPi 0.18, probably others

**Printer model & used firmware incl. version**

NA

**Browser and version of browser, operating system running browser**

NA

**Checklist of files to include below**

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Posting as a bug report so that this is not forgotten about, so that the SW update plugin can be poked to make it do the right thing. Have not yet confirmed problem, but it has come up a couple of times on support channels.

cp2004, 2021-02-20:
This is not a bug, but more of an improvement - the error shown is generated here:
https://github.com/OctoPrint/OctoPrint/blob/27af9721e5bfb4d5d788cf79050c769e9a087d68/src/octoprint/plugins/softwareupdate/static/js/softwareupdate.js#L670-L693

It probably would make sense to change the error message, will maybe look at embedding the short snippet from the error response in the Pnotify dialog.

foosel, 2021-03-09:
@cp2004 You've marked that as grabbed - need anything from me?

cp2004, 2021-03-09:
Only a deadline to encourage me to get on with it 😄 

I was thinking just simply quoting the error in the notification, or at least fixing the currently misleading message that says 'check the log' given that the software update [never actually logs it's failures](https://github.com/OctoPrint/OctoPrint/blob/2fc287768bb097035cd26826362016c50d0701a7/src/octoprint/plugins/softwareupdate/__init__.py#L1059-L1094). PR has not yet left the station but it will be on it's way soon.

foosel, 2021-03-15:
Merged and ready for 1.6 👍

PR #4048 (2021-03-15): Show meaningful message when software update is blocked
...and log when it is undervoltage.

The problem here was twofold: first, a very generic message asking to check the log confused users. Second, there was nothing logged when it said check the logs.

So, this changes that so the message is escaped and displayed in the pnotify and when the update is blocked for undervoltage this is logged.

This does not add logging for other errors, since it seems unnecessary to me.

Closes #4024
commit aa9a2335 (2021-03-15): :pencil2: Show meaningful message when software update is blocked

And log when it is undervoltage.

The problem here was twofold: first, a very generic message asking to check the log confused users. Second, there was nothing logged when it said check the logs.

So, this changes that so the message is escaped and displayed in the pnotify and when the update is blocked for undervoltage this is logged.

This does not add logging for other methods, since it seems unnecessary to me.

Closes #4024
closed 2021-04-27 by foosel