OctoPrint/OctoPrint #4027: Change filament (M600) disconnects octoprint from printer

jpokorny12, 2021-02-20:
**The problem**

Hello,
i have problem with filament change. 
I add to gcode M600. When the printer reads the M600, the filament change begins. After a while, communication with the printer is disconnected.
I try send M600 via terminal and it works.
When replacing the filament, I can't control the filament change via the dialog box in octoprint (continue, purge more). Could there be an error associated with this?
I try send M600 via terminal and it works.
in octoprint i  have installed filament change plugin.
In marlin i have uncomment host_action_commands and in octoprint i have M600 in long running comands.

I attached log. I download and install new clean octoprint
Thank you

**Did the issue persist even in safe mode?**

Yes, it did persist

**If you could not test in safe mode, please state why**

_No response_

**Version of OctoPrint**

OctoPrint 1.5.3

**Operating system running OctoPrint**

OctoPi 0.18.0

**Printer model & used firmware incl. version**

_No response_

**Browser and version of browser, operating system running browser**

_No response_

**Checklist of files to include below**

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[image]
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/6015418/octoprint.1.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6015419/serial.log)

[CE3_pikachu_3per 0infill - kopie.txt](https://github.com/OctoPrint/OctoPrint/files/6015421/CE3_pikachu_3per.0infill.-.kopie.txt)

cp2004, 2021-02-20:
I can see some communication timeout in your octoprint.log file, however the serial log (a) contains no issues and (b) contains no M600 commands sent to the printer.

When the M600 command is issued, the printer **must** continue to answer with busy messages and when prompted by OctoPrint, otherwise how does OctoPrint know it is alive? If the firmware randomly stops doing this (such as in the middle of a print) and it stops responding, there is likely a firmware bug, and this will have to be fixed in Marlin. It may have already been fixed, in the development `bugfix-2.0.x` branch. 

Please reproduce the issue *with serial logging enabled* and then upload the new serial.log file here, so that we can see what is happening.

jpokorny12, 2021-02-20:
When I manualy send M600 to terminal, printer change filament and after change octoprint works.  But i dont control filament change throu octoprint popup window with (continue, purge more...)

```
Recv:  T:20.31 /0.00 B:20.31 /0.00 @:0 B@:0
Send: M600
Recv: //action:paused
Printer signalled that it paused, switching state...
Recv: //action:prompt_end
Recv: //action:prompt_begin Pause
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv:  T:20.00 /170.00 B:20.00 /0.00 @:127 B@:0
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:20.31 /170.00 B:20.31 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:20.62 /170.00 B:19.84 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:20.62 /170.00 B:19.53 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:20.94 /170.00 B:20.00 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:32.47 /170.00 B:19.53 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:33.15 /170.00 B:19.69 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:35.67 /170.00 B:20.31 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:38.42 /170.00 B:20.16 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:45.07 /170.00 B:20.16 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:51.82 /170.00 B:19.22 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:55.10 /170.00 B:20.16 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:60.14 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:65.51 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:70.61 /170.00 B:20.16 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:75.89 /170.00 B:20.31 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:80.70 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:85.81 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:90.14 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:95.45 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:100.00 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:104.98 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:109.43 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:113.92 /170.00 B:20.16 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:118.43 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:123.09 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:127.32 /170.00 B:20.47 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:131.72 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
Recv:  T:135.70 /170.00 B:20.62 /0.00 @:127 B@:0
Recv: echo:busy: processing
[... 90 lines cut ...]
When the filament change starts, the communication is disconnected. After changinge filament, the printer display shows printing. But the printer head stay because it is disconnected from the octoprint.
```
Send: N788 G1 F1350 X119.862 Y111.246 E51.15907*62
Recv: //action:paused
Printer signalled that it paused, switching state...
Changing monitoring state from "Printing" to "Pausing"
Recv:  T:200.00 /200.00 B:49.59 /50.00 @:52 B@:0
Recv: echo:busy: processing
Recv:  T:200.17 /200.00 B:49.31 /50.00 @:49 B@:0
Recv: echo:busy: processing
Recv:  T:200.21 /200.00 B:48.19 /50.00 @:49 B@:0
Recv: echo:busy: processing
Recv:  T:200.38 /200.00 B:48.68 /50.00 @:46 B@:127
Recv: X:10.00 Y:225.00 Z:20.00 E:49.16 Count X:800 Y:18000 Z:7823
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Pausing" to "Paused"
Send: M105
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: M105
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Paused" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
Connection closed, closing down monitor
```

I sen logs. I try manualy send M600 and print twice gcode with M600

jpokorny12, 2021-02-20:
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6015878/serial.log)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6015879/octoprint.log)

cp2004, 2021-02-20:
There's still no issue that I can see that OctoPrint can fix. I've done some digging, and this seems to keep being reported to Marlin but no one has yet fixed it as far as I can find.

https://github.com/MarlinFirmware/Marlin/issues/18565
https://github.com/MarlinFirmware/Marlin/issues/18854
https://github.com/MarlinFirmware/Marlin/issues/20431

I can't find one where there is a 'hey, this was a bug, now fixed'. Mostly stale issues that never came to a resolution.

Reviewing the output snippets you provided (BTW, there is still nothing in the serial.log) the problem is clear - the printer stops responding. OctoPrint tries to restart communication 3 times, then gives up. There's nothing more that it can do as far as I can tell.

Please try and report this to the Marlin community, in the hope that at some point the issue will be fixed.

Also linking
https://github.com/OctoPrint/OctoPrint/issues/3862
https://github.com/OctoPrint/OctoPrint/issues/3877
where this issue has been reported before.

jpokorny12, 2021-02-20:
Thank you for your time. I couldn't figure out what was causing this problem
I try ask in Marlin community.

closed 2021-02-20 by jpokorny12