OctoPrint/OctoPrint #4035: [Request] Add ability to filter /api/plugin/pluginmanager

JeffB42, 2021-02-28:
**Is your feature request related to a problem? Please describe.**

Calling /api/plugin/pluginmanager returns info on all plugins.  When looking for info on a specific plugin, the call can be expensive.  Also, this was posted as a FR in https://community.octoprint.org/t/request-filtering-of-api-plugin-pluginmanager-giant-file/23465/5

**Describe the solution you'd like**

It's up to you, but maybe something like:
/api/plugin/pluginmanager?key={key-of-plug-in}

**Describe alternatives you've considered**

_No response_

cp2004, 2021-03-01:
I think at the bare minimum the plugin repository needs to be stripped out of this, since it is weighing in for me (with a reasonable number of plugins installed) at 300kb - *way* too much. Removing this drops 260kb out of the endpoint. It could be loaded async when the plugin repository is opened, and just display a loading indicator there. Obviously once this is done once, it doesn't need to be done again until next reload of the page, so it would not be a frustrating loading time IMO.

foosel, 2021-03-04:
Implemented. To quote the commit message:

> `api/plugin/pluginmanager` has been deprecated in favor of
> `plugin/pluginmanager/plugins`, `plugin/pluginmanager/orphans` and
> `plugin/pluginmanager/repository`. Additionally, there's now
> `plugin/pluginmanager/plugin/<key>` to retrieve info for a specific plugin
> by its identifier.
> 
> Repository and orphans are only fetched on opening of the Settings
> dialog, plugins and notifications on login with required permissions.
> All list endpoints make use of ETag headers.
> 
> The former simple API endpoint `api/plugin/pluginmanager` has been left in
> the code for now but marked as deprecated. Same goes for the clientJS
> API.

JeffB42, 2021-03-05:
Excellent!  Thanks Gina!

closed 2021-03-05 by JeffB42