OctoPrint/OctoPrint #4036: Model Height in .metadata.json wrong

Jack77777777, 2021-03-01:
### The problem

1. Sliced an model in SuperSlicer
2. Model has a height of 10mm and this height is shown in SuperSlicer
3. Uploaded to Octoprint
4. Octoprint shows 0.2mm less total height -> 9.8mm
5. .metadata.json shows 9.799999997019768

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

Windows 10 Pro 20H2 19042.804

### Printer model & used firmware incl. version

Anycubic i3 Mega S | Marlin 1.1.9 (Github, davidramiro) MARLIN

### Browser and version of browser, operating system running browser

Google Chrome 88.0.4324.150 (Official Build) (64-bit)

### Checklist of files to include below

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[image]
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6063919/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6063921/serial.log)
[metadata.json.zip](https://github.com/OctoPrint/OctoPrint/files/6063925/metadata.json.zip)
Gcode just remove the .txt
[Nut.gcode.txt](https://github.com/OctoPrint/OctoPrint/files/6063934/Nut.gcode.txt)

cp2004, 2021-03-01:
Confirmed, reproduced with the provided gcode file

foosel, 2021-03-03:
The problem seems to be that the GCODE never homes, so the minimum Z coordinate never gets set to 0 but instead the next lowest value ever actually seen in the code, which is 0.2. I'll see that I make it default to a minZ of 0 to avoid this situation. Should not cause issues I hope.

foosel, 2021-03-04:
Ready for 1.6.0

Jack77777777, 2021-03-06:
@foosel Do you have a approx. relase date doe 1.6.0 ?

cp2004, 2021-03-06:
There's no specific release date yet, first we have to go through release candidates as well so there it is still quite a long way off.

Jack77777777, 2021-03-09:
My custom end script lifts the nozzle 20mm. This is also added as height in octoprint. Is there a way to exclude this ?

cp2004, 2021-03-09:
You can't have everything without creating some massively heavy analysis of the gcode to work out what is the model and what is not. It's not that simple - and excluding it for your custom end script would work but my custom end script might be different. You can add custom scripts to OctoPrint that would not be taken into account on the analysis.

closed 2021-04-27 by foosel

Jack77777777, 2021-05-07:
@cp2004 @foosel this is not fixed with OP 1.6.0

cp2004, 2021-05-07:
You would have to re-upload any existing files for the fix to take affect, since analysis is only run once at upload.

However, if you have done that then please open a bug report with your new findings.

Jack77777777, 2021-05-07:
@cp2004 i deleted all files before updating to 1.6.0. The issue is still present with new gcodes uploaded.

Jack77777777, 2021-05-07:
[image]

Layer Height 0.2x160=32. OP shows 31.8

cp2004, 2021-05-07:
> However, if you have done that then please open a bug report with your new findings.

^ This still applies. Please open a new bug report with all the new information.

Jack77777777, 2021-05-10:
> > 
> _Edited_: The reason I ask this, is because your original test case is working. So this likely means we have a different issue at hand.

@cp2004 excuse me but i don´t understand. If i download the Nut.gcode from the initial post of this issue and upload to OP one Layerheight is still missing. 
[image]
50x0.2mm=10mm. OP shows 9.8mm.

foosel, 2021-05-10:
This is a freshly flashed OctoPrint 1.6.1, newly created about 2h ago while I was doing final tests for the 1.6.1 bugfix release. As you can see, the analysis returns the correct value:

[image]

``` json
$ cat .metadata.json
{
  "issue_4036.gcode": {
    "hash": "124c2711d3783049d6a78488619831c054686272",
    "analysis": {
      "estimatedPrintTime": 604.3541325803479,
      "printingArea": {
        "maxZ": 10.0,
        "minX": 95.861,
        "maxX": 119.139,
        "maxY": 119.139,
        "minY": 95.861,
        "minZ": 0.0
      },
      "dimensions": {
        "width": 23.27799999999999,
        "depth": 23.27799999999999,
        "height": 10.0
      },
      "filament": {
        "tool0": {
          "volume": 1.5600075203548323,
          "length": 648.5757600000055
        }
      }
    }
  }
}
```

The highlighted parts in your screenshot are from third party plugins, I do not know where they get their data from. What I can tell you is that the data in metadata.json is now correct, and this ticket was about it being wrong, so this bug is solved as far as I can currently say.

Jack77777777, 2021-05-10:
Got it. Thank you pointing me to that. @OllisGit I thought DLP catches the information from OP.

commit 4c4570e2 (2021-03-04): 🐛 GCode analysis & viewer: Assume bed at Z 0.0

People might not actually always have a home command in their GCODE
files, so let's just depend on the so far pretty universal truth that
printers have a bed at z=0 and that is just always the lowest Z of a
printed object. And just in case, let's make this value configurable
because the printer gods only know what some people might come up with
in terms of experimental printer mechanics that end up with negative z
coordinates or a bed at z=100.

Closes #4036