OctoPrint/OctoPrint #4037: ETA wrong

dheera, 2021-03-03:
### The problem

ETA shows 2.5 days for a 5-hour print

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

Raspbian

### Printer model & used firmware incl. version

Prusa i3 MK3

### Browser and version of browser, operating system running browser

Version 88.0.4324.150 (Official Build) (64-bit), Ubuntu 20.04

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[image]
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6073547/octoprint.log)
[image]

cp2004, 2021-03-03:
The ETA in OctoPrint is not known for it's reliability in certain situations, and this is one of them...

Your print has so far taken 1 hour, however it is only 1% through. Very basic estimations mean that the ETA would be about 100 hours. Now it is not quite that bad, that was an extreme example. Obviously this is wrong, just trying to show how things can go wrong.

Since each line of gcode can have massively varying length, some lines will take ages and some will be really short, say on a curve. So the speed is not linear, through the file, it varies by quite a lot - and things like first layer speed or minimum layer time can make this even worse.

OctoPrint's analyser is a trade off between speed and accuracy - and there are other options for more accurate ETA, such as [Print Time Genius](https://plugins.octoprint.org/plugins/PrintTimeGenius/), [Slicer Estimator](https://plugins.octoprint.org/plugins/SlicerEstimator/) and others. Print time genius runs a simulation of the print to determine very accurate print times.

foosel, 2021-03-03:
Thanks for the summary @cp2004 

This is a horse that has been beat to death repeatedly and is not something that will be fixed in OctoPrint core for the foreseeable future. Trying to do reliable estimation of a completely non linear process when all you have is linear extrapolation is bound to cause issues in very non uniform prints. This seems to be such a case. Providing additional information through simulation (see the aforementioned plugin) could help here, but it's highly dependent - again - on the printed object.

Closing.

closed 2021-03-03 by foosel