OctoPrint/OctoPrint #4038: Paused print (Prusa M601) leads to timeout, even though OctoPrint recognized the paused state

zeroflow, 2021-03-04:
### The problem

1. Print GCode with M601 in it
2. M601 is reached
3. Octoprint switches state ("Printer signalled that it paused, switching state...")
4. Wait
5. Octoprint times out ("No response from printer after 6 consecutive communication timeouts")

Expectation:
1. Print GCode with M601 in it
2. M601 is reached
3. Octoprint switches state ("Printer signalled that it paused, switching state...")
4. Insert parts (Magnets, Nuts, ...) into the print
5. Select "Resume" in the printers menu
6. Printer continues printing

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Prusa Mk3S MMU2S, Prusa FW 3.9.3

### Checklist of files to include below

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Sometimes, when acting fast enough for the timeout, the print is able to continue.
But this is not long enough to insert parts into the print.

Additional Settings from default:
Pausing Commands: M0, M1, M25, M600, M601
Long running commands: G4, G28, G29, G30, G32, M400, M226, M600, M601, M602
After Print Job is paused: -empty-
Before Print Job is Resumed:  -empty-

GCode:
[Shape-Box_0.2mm_PLA_MK3SMMU2S_2m.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/6085397/Shape-Box_0.2mm_PLA_MK3SMMU2S_2m.gcode.zip)

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6085458/octoprint.log)
[serial.log.2021-03-04_17-12-22.log](https://github.com/OctoPrint/OctoPrint/files/6085456/serial.log.2021-03-04_17-12-22.log)
[failed_serial.log.2021-03-04_17-34-10.log](https://github.com/OctoPrint/OctoPrint/files/6085597/failed_serial.log.2021-03-04_17-34-10.log)

Serial log of successful M601 + Resume from Menu
There I resumed as fast as possible and everything worked fine.

[success_serial.log.2021-03-04_17-28-11.log](https://github.com/OctoPrint/OctoPrint/files/6085568/success_serial.log.2021-03-04_17-28-11.log)

cp2004, 2021-03-04:
We have this issue over and over - there is nothing better that OctoPrint can do, the printer firmware might as well have crashed. It should continue to respond to commands even in the paused state, it should not say 'I'm paused' then just disappear off and not respond anymore.

Previous examples are here: #4027, #3862, #3877

Please try and report this to Prusa, as it is a bug that they should fix their end, to make the communication not hang when the printer is paused.

However, you say that the M601 command is flagged as a 'long running command'? I wonder why there was still a timeout, admittedly it waited a long time - I am not entirely sure how that is supposed to work.

zeroflow, 2021-03-04:
Thanks for the feedback.
I'll try and implement some "busy" response in prusa firmware.

closed 2021-03-04 by zeroflow