OctoPrint/OctoPrint #4040: [Request] Record last_fan_speed like last_position and last_temperature

jantman, 2021-03-05:
### Is your feature request related to a problem? Please describe.

I'm trying to work on a print-pausing feature using gcode scripts. We have access to the last temperatures (``pause_temperature`` in the Jinja context) and last positions (``pause_position``), but not the fan speed when paused.

### Describe the solution you'd like

I'd like a ``pause_fan_speed`` variable available in the Jinja context, like the temperature and position, so that I can also pause and resume the fan.

### Describe alternatives you've considered

For the time being, I'm just not changing the fan speed when paused. But this isn't optimal.

jneilliii, 2021-03-06:
Couldn't this be implemented with a single file plugin like in the example [here](https://github.com/OctoPrint/Plugin-Examples/blob/master/gcode_script_variables.py)? It would have to add a gcode phasing hook as well to keep track of the last fan speed sent to the printer to store internally to provide to the script variable return I'm guessing. The largest problem here is that this doesn't take into account any manual adjustments made on the printer panel.

jneilliii, 2021-03-06:
So I tool a stab at it and couldn't figure out how the variables work with teh scripts callback, but I was able to get the fam speed to prefix teh beforePrintStart script to achieve the same thing. You can move from prefix to postfix if you want it to set fan speed at the end of the script rather than the end of the script.

```
# coding=utf-8
from __future__ import absolute_import

import octoprint.plugin
import re

class FanSpeedScriptVariable(octoprint.plugin.StartupPlugin):
	def __init__(self):
		self.last_fan_speed = 0
		self.fan_speed_match = re.compile("^M106.+?S(\d+)")

	def gcode_script_variables(self, comm, script_type, script_name, *args, **kwargs):
		if not script_type == "gcode" or not script_name == "beforePrintResumed":
			return None

		prefix = "M106 S{}".format(self.last_fan_speed)
		postfix = None
		return prefix, postfix

	def gcode_callback(self, comm_instance, phase, cmd, cmd_type, gcode, subcode=None, tags=None, *args, **kwargs):
		if gcode and not gcode == "M106":
			return

		test = self.fan_speed_match.findall(cmd)
		if test is not None:
			self.last_fan_speed = int(test[0])

__plugin_name__ = "Fan Speed Script Variable"
__plugin_pythoncompat__ = ">=2.7,<4"
__plugin_version__ = "0.0.1"
__plugin_implementation__ = FanSpeedScriptVariable()
__plugin_hooks__ = {
	"octoprint.comm.protocol.scripts": __plugin_implementation__.gcode_script_variables,
	"octoprint.comm.protocol.gcode.sent": __plugin_implementation__.gcode_callback
}
```

FedericoNembrini, 2022-02-26:
I think I can contribute. I finish implementing the changes and open a pull request.

foosel, 2022-06-07:
Implemented by @FedericoNembrini in #4531, to be rolled out with 1.9.0.

foosel, 2023-05-23:
1.9.0 has been released.

closed 2023-05-23 by foosel