OctoPrint/OctoPrint #4042: MemoryError during a print

ahernsean, 2021-03-07:
### The problem

1. I had a print running, then OctoPrint stopped listening on port 5000. haproxy was listening just fine on port 80, but would receive a 503 when it tried to connect to the OctoPrint python process on port 5000. OctoPrint was still sending data to the printer over the USB bus, and I confirmed this with strace. But something as simple as `nc -v 127.0.0.1 5000` would not connect.

2. After several more hours, the print eventually failed. OctoPrint crashed with a python MemoryError. The log says that I should report this as an OctoPrint bug, so here I am. Here's the octoprint log of interest:

```
2021-03-07 09:34:41,674 - octoprint.plugins.tplinksmartplug - INFO - MoeSizzlac_-_Mandalorian_Chest_Armor_2.stl
2021-03-07 09:41:56,445 - octoprint.plugin.core - ERROR - Exception while sending plugin data to client
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 2268, in send_plugin_message
    client(plugin, data, permissions=permissions)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/sockjs.py", line 426, in on_plugin_message
    "plugin", payload={"plugin": plugin, "data": data}, permissions=permissions
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/sockjs.py", line 569, in _emit
    map(lambda p: self._user.has_permission(p), permissions)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/sockjs.py", line 569, in <lambda>
    map(lambda p: self._user.has_permission(p), permissions)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/access/users.py", line 1307, in has_permission
    return self.has_needs(*permission.needs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/access/users.py", line 1310, in has_needs
    return set(needs).issubset(self.needs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/access/users.py", line 1302, in needs
    needs = needs.union(group.needs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/access/groups.py", line 721, in needs
    needs = needs.union(p.needs)
MemoryError
2021-03-07 09:41:56,640 - octoprint.plugins.tplinksmartplug - INFO - Object
2021-03-07 09:41:56,641 - octoprint.plugins.tplinksmartplug - INFO - NONMESH
2021-03-07 09:42:02,987 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2542, in _monitor
    self.last_temperature.custom,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1407, in on_comm_temperature_update
    custom=copy.deepcopy(custom),
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1177, in _addTemperatureData
    self._temps.append(data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1727, in append
    self._data = self._invariant(self._data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 2016, in temperature_invariant
    data.sort(key=lambda x: x["time"])
MemoryError
2021-03-07 09:42:03,037 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline (Error: See octoprint.log for details)"
```

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I'm not even sure I can reproduce this in normal mode, let along safe mode. I will try the print again and see if it recurs.

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Creality CR-10S Pro v2, but this doesn't seem relevant

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6097391/octoprint.log)

foosel, 2021-03-09:
A `MemoryError` is sadly something outside of the control of OctoPrint. It might be caused by a plugin you have installed, but as you said, it's probably not trivial to reproduce.

We had something similar late last year with #3831 but that never saw a follow up from the OP. I fear unless we see this happening in safe mode there's not really anything that can be done here 🤷‍♀️

---

Probably unrelated, but I also saw this in your log:

```
2021-03-07 10:01:42,460 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 305994}
2021-03-07 10:16:35,273 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2021-03-07 10:16:42,523 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 306894}
2021-03-07 10:26:22,975 - octoprint.startup - INFO - ******************************************************************************
2021-03-07 10:26:22,978 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-03-07 10:26:22,978 - octoprint.startup - INFO - ******************************************************************************
2021-03-07 10:26:23,375 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
```

Note how there's no shut-down log entry from OctoPrint, it just suddenly restarts 10min after the last heartbeat. What happened there, did you just pull the plug? If so, don't do that, that can cause file system corruption.

foosel, 2021-03-09:
That sounds like something was seriously up with it then if even SSH was no longer accessible. Yeah, please keep us in the loop should you find more hints as to what's going on. 

And good to know, sorry to be so verbose on this, it's just that a ton of OctoPrint's users simply pull the plug all the time and then open bug reports after they've successfully shot the file system to hell through that, so I rather warn when I see something 😅