OctoPrint/OctoPrint #4044: When Connecting/Autohoming Printer overtravels

Q-Bot2021, 2021-03-12:
### The problem

Printer Mode: CR-10 V2
Pi: Pi4 8GB
Octopi Version: 0.18
Octoprint Version: 1.5.3

When connecting to the printer or initialising the auto-home function from the control page the X and Y axes always travel by 50% of the maximum build dimensions in the positive direction.  If the axes are too far to the positive direction this results in overtravel condition.  

Work around is to always be next to printer when connecting and move the axes to 0 before connecting/auto-homing.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

OctoPi  0.18.0

### Printer model & used firmware incl. version

CR10 V2 Stock Firmware

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Printer Mode: CR-10 V2
Pi: Pi4 8GB
Octopi Version: 0.18
Octoprint Version: 1.5.3

When connecting to the printer or initialising the auto-home function from the control page the X and Y axes always travel by 50% of the maximum build dimensions in the positive direction.  If the axes are too far to the positive direction this results in overtravel condition.  

Work around is to always be next to printer when connecting and move the axes to 0 before connecting/auto-homing.
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6129465/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint/files/6129468/serial.log)

foosel, 2021-03-12:
Your firmware is responsible of correctly executing home commands and to not do weird stuff upon connect. This isn't an issue with OctoPrint but rather your firmware. 

---

```
2021-03-12 10:44:58,420 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin Creality 3D"
```

The firmware shipped with Creality printers has proven quite problematic and unreliable in the community, you are better off by flashing a stock Marlin.

https://community.octoprint.org/t/printers-known-to-ship-with-broken-or-unsafe-firmware-or-broken-electronics/25911

---

```
2021-03-10 18:17:30,686 - octoprint.plugins.pi_support - WARNING - This Raspberry Pi is reporting problems that might lead to bad performance or errors caused by overheating or insufficient power.
!!! UNDERVOLTAGE REPORTED !!! Make sure that the power supply and power cable are capable of supplying enough voltage and current to your Pi.
```

Either your Pi's power supply is insufficient, or your printer is being backpowered.

https://community.octoprint.org/t/printers-known-to-have-backpowering-issues/25913

You need to solve this.

closed 2021-03-12 by foosel

Q-Bot2021, 2021-03-12:
Hi Foosel, 

This behaviour changed when I upgrade from an earlier version of octoprint (I did not record the version number).  On the earlier version of octoprint the printer only did this when connecting, not during auto-homing.

Also surely if this was the printer firmware then the same thing would happen when I connect to the COM port using a non-octopi device? When I use another serial based printer application (eg. cura) this does not happen.

foosel, 2021-03-12:
Solve the undervoltage issue first. When the issue still persist after, enable `serial.log` (the one you posted is empty), reproduce, provide the log. Also check your configured GCODE scripts.

In general OctoPrint doesn't send any move, temperature or other potentially dangerous commands to the printer out the box.