OctoPrint/OctoPrint #4051: /api/files/sdcard returns stale list of files

JeffB42, 2021-03-14:
### The problem

1. With an SD card in the printer, call /api/files/sdcard
-> a list of files is retuned (expected)
2. Remove the SD card
3. Call /api/files/sdcard again
-> the same list of fils is returned
expected: since the SD card is not in the printer, I expected an empty list to be returned.

On the other hand, if I do the following:
a. With an SD card in the printer, send M20
-> a list of files is retuned (expected)
2. Remove the SD card
3. send M20 again
-> no files are returned (expected)
...so to me, this appears to be an OctoPrint bug, not a firmware bug

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prisa i3

### Browser and version of browser, operating system running browser

n/a

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

cp2004, 2021-03-15:
We will require both the OctoPrint log and an enabled serial log to look into this issue.

foosel, 2021-03-15:
I agree. Unless the printer *tells* OctoPrint that the SD has been removed, in cannot possibly know, in which case this is expected behaviour really.

JeffB42, 2021-03-16:
Here are some log files.

I rebooted my RPi and OctoPi
Then sent a GET /api/files/sdcard via Postman
Here is the first dump:
[octoprint-first-time.log](https://github.com/OctoPrint/OctoPrint/files/6146134/octoprint-first-time.log)
(and as I mentioned, a list files is returned, which is expected) 

I then removed the SD card
and then sent another GET /api/files/sdcard request, and here is the updated log
[octoprint-second-time-no-sd-card.log](https://github.com/OctoPrint/OctoPrint/files/6146137/octoprint-second-time-no-sd-card.log)
(and as I mentioned, the same list of files is returned, which is unexpected, since the files don't exist anymore since the SD card was removed)

Interesting, in that in both times errors are being reported in the logs.  The printer is a Prusa i3 MKS2 is if matters, firmware 3.9.3-3556
Also worth mentioning, is that I'm only performing the steps I listed above.

JeffB42, 2021-03-16:
> I agree. Unless the printer _tells_ OctoPrint that the SD has been removed, in cannot possibly know, in which case this is expected behaviour really.

I don't understand your reply.  As a consumer of this API, I'm expecting it to return the _current_ list of files that are available.  There is no more SD card, so there _are_ no more files.  I'm consuming the API and treating it as a back box - I don't know what the implementation is, or what takes place behind the scenes, but I'm expecting that when I call /api/files/sdcard, that I get back the (current) list of files, not a cached list.

The docs for this API says, "Retrieve information regarding the files currently available on the selected location..._" and that doesn't seem to be the case.

cp2004, 2021-03-16:
The serial log was probably the more important, so it could show if the printer has told OctoPrint that the SD card is out. My printer, with Marlin 2.0.7.2 sends a message when the SD card is in or out and OctoPrint can immediately update the files list, and this problem does not exist.

The problem we *think* you are facing is that the printer does not do that, and so as far as OctoPrint knows the SD card is still inserted in the printer. Since it will not send the commands to list all the files from the printer on the API (that would be *slow*) they are saved, and only refreshed when told otherwise.

cp2004, 2021-03-16:
I just remembered,I have SD autoreport status enabled in Marlin. *That's* the name of the feature I was trying to remember.

foosel, 2021-03-16:
File lists are expensive, and SD card file lists even more so (since they involve actual communication with the printer, which for example during a print job might cause performance issues, depending on how the firmware side is implemented). Usually during operation, the SD card situation on the printer should not change so OctoPrint caches the result. If it wouldn't do that, you'd see a constant stream of `M20` on every page reload or on every file list request from a third party client - that would be an utter waste of time and detrimental to print operations. You can override the internal cache with the `force` parameter to the API request (which I just noticed is indeed missing from the docs, I'll rectify that in a second), and that is also what the reload button in the UI uses. Additionally, when the printer tells OctoPrint "hey, the SD card just got ejected" by an `SD init fail` message or similar it will also remove the SD file list and set the SD availability to false.

foosel, 2021-03-16:
Fixed for 1.6.0. Supplying the `force=true` request parameter will now correctly ignore even the SD card cache. Documentation has also been updated.

JeffB42, 2021-03-17:
> Fixed for 1.6.0. Supplying the `force=true` request parameter will now correctly ignore even the SD card cache. Documentation has also been updated.

Thanks Gina!  This sounds like it will work.

JeffB42, 2021-03-17:
> The serial log was probably the more important, so it could show if the printer has told OctoPrint that the SD card is out. 

I'm going to close this ticket, but before I do, in case it's helpful here is a log with serial turned on:

1. I booted up my system, with serial logging on, and the SD card already inserted
2. I then called /api/files/sdcard
[octoprint-first-time.log](https://github.com/OctoPrint/OctoPrint/files/6153315/octoprint-first-time.log)
[serial-first-time.log](https://github.com/OctoPrint/OctoPrint/files/6153324/serial-first-time.log)
3. Removed the SD card
4. Called /api/files/sdcard again
[octoprint-second-time.log](https://github.com/OctoPrint/OctoPrint/files/6153318/octoprint-second-time.log)
[serial-second-time.log](https://github.com/OctoPrint/OctoPrint/files/6153325/serial-second-time.log)

commit a24d3f26 (2021-03-16): 🐛 Fix caching behaviour on SD card API

When `force=true` is supplied, the SD card files should be refreshed.
Also finally documented the `force` parameter in the API docs.

Closes #4051
closed 2021-03-17 by JeffB42