OctoPrint/OctoPrint #4053: [Request] Allow users to add metadata to print files (such as expected duration and filament used)

emilyboda, 2021-03-16:
I'd love to be able to record information that Cura calculated about my print when I upload it to my octoprint server. Currently users usually create a file naming convention that incorporates this info or keep a separate excel or text file log.

foosel, 2021-03-16:
Pretty sure that yes. @emilyboda could you please elaborate this request? The type of data you are quoting is already covered by existing plugins such as PrintTimeGenius through data extraction. Additional data defined by the user already exists as a request in #519 and is waiting for someone to implement it (as a plugin, the backend hooks for this are there).

emilyboda, 2021-03-16:
As far as I know, the plugins for estimating the length of time only allow me to see how much time a print will take after I start the print. I'd like to be able to see this information for all of my files in the file list at a glance. While a notepad would work for this, the notes should be available to the user while they scroll through their list of files, in the same way that the size of the file and the layers in the file are visible.

[Here's a photo of what I'm thinking.](https://photos.app.goo.gl/vccctfBvtiBAuBKdA)

foosel, 2021-03-16:
I think you want to click on the little "v" shaped button ;)

[image]

For this you don't even need a plugin, that IS already built in. This metadata is extracted by OctoPrint in the background, however its calculated anew instead of taking something from the slicer because there's no standard in slicer comments. Some plugins however extract data from at least the most popular slicers, which improves accuracy.

emilyboda, 2021-03-16:
Wow, that's an incredible feature! I had no idea it was there. That's exactly what I wanted, I will close this!

closed 2021-03-16 by emilyboda