OctoPrint/OctoPrint #4054: Cura and octoprint

Ruissilva, 2021-03-22:
### The problem

1. Cura sends file but its terrebly slow and sometimes it doesn't send. I think time waiting is proportional to the size of the file.
2. If i replace a file with the same name, octprint freezes. I have to delete the file in octoprint and then put the file

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

latest

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

1. Cura sends file but its terrebly slow and sometimes it doesn't send. I think time waiting is proportional to the size of the file.
2. If i replace a file with the same name, octprint freezes. I have to delete the file in octoprint and then put the file

Ruissilva, 2021-03-22:
And also, just remebered, octoprint completely freezes if the google chrome extension Rakuten is turned on.

fieldOfView, 2021-03-22:
Please open separate issues for separate problems.

> Cura sends file but its terrebly slow and sometimes it doesn't send. I think time waiting is proportional to the size of the file.

Please check and confirm if saving the file to disk and using the webinterface to upload is faster/more reliable. If it is, then this is likely not an OctoPrint issue, but an issue with the Cura OctoPrint Connection plugin (which I created), so this issue belongs here:
https://github.com/fieldOfView/Cura-OctoPrintPlugin

> If i replace a file with the same name, octprint freezes. I have to delete the file in octoprint and then put the file
There is nothing that we can say about this unless you provide logs.

Ruissilva, 2021-03-24:
I made a new octoprint install. I had a plugin issues. I still have one plugin that I suspect is giving mi errors, that is the smart filament sensor plugin. I say this because I discovered It has confits with all the plugins that affects gpio pins. In relation to cura, I need to do new tests now that i have the new and clean image. I'm just making my first print with this image and I still have the "tornado.access - WARNING", that I also had before, but I suspect it is because of the smart filament sensor plugin. What do you think? Please see the log.
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6194305/octoprint.log)

jneilliii, 2021-03-24:
I don't think this is related. A lot of the 404 warnings you have in your log are related to thumbnails it looks like. Only other oddity inside your log is related to plugins, not specifically to smart filament sensor. The one odd one is 

```
2021-03-23 20:42:39,434 - tornado.access - WARNING - 404 GET /nice%20ports%2C/Tri%6Eity.txt%2ebak (::ffff:192.168.1.76) 3.77ms
```
Which leads me to believe you've made your octoprint publicly accessible and someone is attempting script hacks on your pi or you may have used nmap to find your pi on the network?

Ruissilva, 2021-03-24:
Ok. Im not using any thumbnail plugin. 
I do have a octoprint public and I don't use nmap :S

Ruissilva, 2021-03-24:
But 192.168.1.76 is my personal Pc, that has a shared folder in the network, and that IP is in message. :)

Ruissilva, 2021-03-24:
Im going to remove the public acess, its better and i never use it. What can i do about the thumbnails?

Ruissilva, 2021-03-24:
Cura and octoprint connection seems to be just fine :)

fieldOfView, 2021-03-24:
I think this is better handled on the community forum than in the issue queue.
https://community.octoprint.org/

closed 2021-03-24 by fieldOfView