OctoPrint/OctoPrint #4057: Very serious bug, being rude and missing the point that is help people and not close issues.

Ruissilva, 2021-03-24:
### The problem

I was in the middle of a conversation, being help by a member and the issue was suddenly closed because it was not on the right place

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

latest

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Ender 3

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Just wanted to leave my complaint, because I didn't like the way my issue previous terminated. I was in a conversation with jneilliii that was helping me, which i thank in advance and then suddenly Aldo Hoeben closed the issue because it was not on the right place. 

Mr. Aldo Hoeben, you may be right but that does not justify what you did. Please intervien if you must but in a nice way and ask to terminate a issue ormove it to another place, not suddenly terminate it. Mr. jneilliii was refering that I may be target of hacking, wich is a very serious issue and pointing some problems in my octoprint image, and I really wanted to see the end of that discussion. 

I just send this e-mail to remind you that the important thing is to help people, the comunity because we need your help, but don't forget that you also need us, the people that love 3d printing. 

You can close this non issue, that for me was one of the worst issues, being unrespectfull.

cp2004, 2021-03-24:
It was recommended that you discuss the issue on the community forums, since it was determined that the issue was no longer a bug.

You are more than welcome to get in contact via the support channels, on the forums and the discord server but this issue queue is reserved for bugs with OctoPrint, not your individual setup. Both of the user's mentioned are active in the community helping out many people, there is no disrespect intended and they will be more than happy to help you further, just in a more suitable place.

closed 2021-03-24 by cp2004

fieldOfView, 2021-03-24:
I'm sorry you were offended by my closing your issue. 

Note that you can continue posting to an issue after it is closed, and even request it be reopened if you feel the issue was closed for the wrong reason.

Ruissilva, 2021-03-24:
Thanks, keep up the good work :)