OctoPrint/OctoPrint #4059: [Request] Timelapse rendering: Adjust -b to match current ffmpeg docs

foosel, 2021-03-29:
the option -b should be used as -b:v or -b:a (video/audio),
but maybe for the resulting command-line it should not be empty..

fiddling with my error i found that this option-value seems also depend on the encoder, too:
[image]

_Originally posted by @ozett in https://github.com/OctoPrint/OctoPrint/issues/3972#issuecomment-808751090_

foosel, 2021-03-29:
- [ ] Find out what `-b` is actually supposed to be doing these days. `man ffmpeg` on OctoPi 0.18.0 and https://ffmpeg.org/ffmpeg.html does only mention it in the examples and it is otherwise completely undocumented, not even `-b:v` or `-b:a` are mentioned outside of examples.
- [ ] Find out if the codec options are indeed overridden via `-b` and if the different units apply here

ozett, 2021-04-01:
dont forget that leaving -b blank in the gui seems to me breaking the resulting ffmpeg command.
also -b seems to requires not only numbers like "10000k" (so maybe gui-check this input?)

foosel, 2021-06-23:
Switched to `b:v` and also added more sanity checking for the bitrate parameter at both API level and timelapse processing. Ready for 1.7.0.

commit 87ddc846 (2021-06-23): ♻️ Switch to `b:v` for bitrate

Seems more fitting to do add a stream selector for the video stream
based on the ffmpeg docs and #4059

Closes #4059
closed 2021-10-11 by foosel