OctoPrint/OctoPrint #4060: Can't upload files

neilyoung, 2021-03-29:
### The problem

When trying to upload files from my local PC to octoprint the progress bar moves right but then again to the left, so the upload seems to fail. Directories are not show. The files are on the PI. Don't have registered anywhere, no app key. Is this the problem?

Running on a PI 3A+.

Connectivity OK, can control the bed, camera is working.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

I'm using your image

### Printer model & used firmware incl. version

Creality Ender 3D

### Browser and version of browser, operating system running browser

macOS Big Sur, Chrome 89.0

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

I have "tailed" the logs mentioned above. None gives a clue. The files are on the PI in the proper directories. Directories are created.

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6221640/octoprint.log)

BTW: Genius bug tracking interface...:)

neilyoung, 2021-03-29:
Triage... Nice...

cp2004, 2021-03-29:
Hmm, from the server's perspective everything is OK - which aligns with what you were saying about how the file makes it to the Pi and is stored sucessfully.

For example: I can see the analysis of the file always coming through:

> 2021-03-29 13:14:00,349 - octoprint.filemanager.analysis - INFO - Starting analysis of local:pi-coral-lid.gcode
2021-03-29 13:14:00,351 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/pi-coral-lid.gcode
2021-03-29 13:14:19,357 - octoprint.filemanager.analysis - INFO - Analysis of entry local:pi-coral-lid.gcode finished, needed 19.01s

Could you open the Javascript console in your browser (F12 or `ctrl-shift-I` and select console) and paste it's contents?

Another question, does hitting the refresh button after bring back the files?
[image]

neilyoung, 2021-03-29:
Did that already. There is nothing :) Refresh does not help either.

neilyoung, 2021-03-29:
Rebooted everything. Now the files are shown. I suppose I must not have been logged in, since now, with the restart, I was required to do that.

neilyoung, 2021-03-29:
Not related, but since I'm seeing it right now: 

The "GCode Viewer Window" tells me, my model has X and Y dimensions of 140 x 138 mm. This is not true. Z is correct.

[image]

The preview window seems to show the proper dimensions:

[image]

The same time the Prusa Slicer tells the truth:

[image]

Even though Blender:

[image]

And the real world dimensions are 50 x 50 x 2

neilyoung, 2021-03-29:
Hmm. This is also kind of funny :)

[image]

neilyoung, 2021-03-29:
Anyway, well done. Nice tool

neilyoung, 2021-03-29:
I see the duration adapt over time, so it is OK I guess :)

cp2004, 2021-03-29:
The first issue is more than likely a prime line, and when the gcode is processed this is part of the model so gets included in the total size. Prime lines are commonly generated by Prusa Slicer.

For the time, this is an issue where OctoPrint is optimized for performance rather than accuracy. Install the PrintTimeGenius plugin to run a simulation of the print to work out an accurate print time.

Is this issue solved now? Can it be closed?

neilyoung, 2021-03-29:
Sure, I will get used to it. Thanks for the quick replies.

neilyoung, 2021-03-30:
Nope, the problem returned. This morning the same. This time restart/refresh/re-login did not help. Need to re-open the issue. Available for help

neilyoung, 2021-03-31:
Anybody?

foosel, 2021-03-31:
Not really right now, no, or someone would have said something.

The only things I see in your original log are what looks like two power cuts (log ends suddenly, never just cut power to the pi, always shut it down through the menu first) and one 400 while retrieving files. Also no idea right now what it could be tbh, never seen an issue like this before when the network was actually working.

jneilliii, 2021-03-31:
Have you tried in an incognito browser window?

neilyoung, 2021-03-31:
No not yet. Didn't expect this to be of help.... Let me try...

neilyoung, 2021-03-31:
Hmm. This works. I mean with all caution. I also had a "this works" situation already in non-incognito mode. Any explanation?

cp2004, 2021-03-31:
Certain browser extensions can conflict, incognito disables them. At least, I think that was the intention there.

neilyoung, 2021-03-31:
Wait, can conflict with what? The upload works. It is just the JS and maybe the communication with the box wich is problematic. This answer does really not satisfy me and if it shall be a solution than I would be sceptical.

cp2004, 2021-03-31:
I have debugged one such issue where selecting a file and printing it wouldn't work with ublock enabled (the adblock extension) because of the file name. So yes, it can happen.

neilyoung, 2021-03-31:
It's not the adblocker. Neither putting octoprint on the whitelist nor disabling adblock works.

jneilliii, 2021-03-31:
yes, the reason for testing was to see if there was a browser extension interfering as well as a fresh clear browser cache. That would be the other question, if you shift+f5 to force a reload of the browser cache makes a difference or not. I've seen noted issues personally with the DuckDuckGo Privacy Essentials and my bed level visualizer plugin causing slowness, and I think I remember an alibaba extension causing someone troubles before with the UI.

neilyoung, 2021-03-31:
I have none of these extensions installed and cleaning the cache was the first I did yesterday. Repeated right now via developer console. No change.

neilyoung, 2021-03-31:
All extensions deactivated. No change. Just incognito works. BTW: The upload bar moves way faster in normal mode, just to switch back after 5 seconds.

neilyoung, 2021-03-31:
An attempt to create folder `123`, previous attempts `abc` and `test`. All folders are on the PI, not shown in the browser. 

```
[1 bytes missing in capture file].POST /api/files/local HTTP/1.1
Host: octopi.local
Connection: keep-alive
Content-Length: 144
Accept: application/json, text/javascript, */*; q=0.01
Cache-Control: no-cache
X-Requested-With: XMLHttpRequest
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36
Content-Type: multipart/form-data; boundary=----WebKitFormBoundaryIy6GLf9Lf6ctDudf
Origin: http://octopi.local
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
Cookie: session_P80=.eJxNjs1qwzAQhN9F51L0t7uSbzGJX8PI0qoWOI6x5EMIffe65NAev5mPYV5izDvXWXRtP_hDjCWJTjDjxJpMMnpyLvpgZeYks03oVVLSgYnSe1CREqqcEa0ETmCnmHXGIJW28mSK0Ujikxw5AoTgJ9KRjEMwSSnyCI5RgdMELqjIZAlDFOeRo_L-frOVk0vitZX2_AxHm8f23Fh067Es_5o_eXl8lXW8c5zDWur9jOfWtrP4Ha1ca3msb90BeCfxOtheWn-hix5ueujBmStcbqYX3z8aCFWQ.YGTAww.WmVU4_v3LVEKAfaHg9V2xfxYR4o

------WebKitFormBoundaryIy6GLf9Lf6ctDudf
Content-Disposition: form-data; name="foldername"

123
------WebKitFormBoundaryIy6GLf9Lf6ctDudf--
HTTP/1.1 201 CREATED
Content-Type: application/json
Location: http://octopi.local/api/files/local/123
Cache-Control: no-store, no-cache, must-revalidate, post-check=0, pre-check=0, max-age=0
Pragma: no-cache
Expires: -1
X-Clacks-Overhead: GNU Terry Pratchett
Vary: Cookie
Set-Cookie: session_P80=.eJxNjs1qwzAQhN9F51L0t7uSbzGJX8PI0qoWOI6x5EMIffe65NAev5mPYV5izDvXWXRtP_hDjCWJTjDjxJpMMnpyLvpgZeYks03oVVLSgYnSe1CREqqcEa0ETmCnmHXGIJW28mSK0Ujikxw5AoTgJ9KRjEMwSSnyCI5RgdMELqjIZAlDFOeRo_L-frOVk0vitZX2_AxHm8f23Fh067Es_5o_eXl8lXW8c5zDWur9jOfWtrP4Ha1ca3msb90BeCfxOtheWn-hix5ueujBmStcbqYX3z8aCFWQ.YGTA-g.njSn1OKMjDHbbGfBSzI6ZOm83Sw; HttpOnly; Path=/
X-Robots-Tag: noindex, nofollow, noimageindex
X-Content-Type-Options: nosniff
X-Frame-Options: sameorigin
Transfer-Encoding: chunked
Content-Encoding: gzip

76
..........=.A.. .E...k3.....,.........&.|....6................e.....s..$..E.v.W.XG.T..0...8D	&.?O..0FNT..q.Z.n..a)....
0

GET /api/files?recursive=true HTTP/1.1
Host: octopi.local
Connection: keep-alive
Accept: application/json, text/javascript, */*; q=0.01
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_2_3) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36
X-Requested-With: XMLHttpRequest
Accept-Encoding: gzip, deflate
Accept-Language: de-DE,de;q=0.9,en-US;q=0.8,en;q=0.7
Cookie: session_P80=.eJxNjs1qwzAQhN9F51L0t7uSbzGJX8PI0qoWOI6x5EMIffe65NAev5mPYV5izDvXWXRtP_hDjCWJTjDjxJpMMnpyLvpgZeYks03oVVLSgYnSe1CREqqcEa0ETmCnmHXGIJW28mSK0Ujikxw5AoTgJ9KRjEMwSSnyCI5RgdMELqjIZAlDFOeRo_L-frOVk0vitZX2_AxHm8f23Fh067Es_5o_eXl8lXW8c5zDWur9jOfWtrP4Ha1ca3msb90BeCfxOtheWn-hix5ueujBmStcbqYX3z8aCFWQ.YGTA-g.njSn1OKMjDHbbGfBSzI6ZOm83Sw
If-None-Match: "d4d6af712d3b44d69776abbb1991e534958d5a15"

HTTP/1.1 200 OK
Content-Type: application/json
Etag: "cf737ea70a39402d22f518484e95bf8ae2852ca8"
Last-Modified: Wed, 31 Mar 2021 18:35:37 GMT
Cache-Control: max-age=0
X-Clacks-Overhead: GNU Terry Pratchett
Vary: Cookie
Set-Cookie: session_P80=.eJxNjs1qwzAQhN9F51L0t7uSbzGJX8PI0qoWOI6x5EMIffe65NAev5mPYV5izDvXWXRtP_hDjCWJTjDjxJpMMnpyLvpgZeYks03oVVLSgYnSe1CREqqcEa0ETmCnmHXGIJW28mSK0Ujikxw5AoTgJ9KRjEMwSSnyCI5RgdMELqjIZAlDFOeRo_L-frOVk0vitZX2_AxHm8f23Fh067Es_5o_eXl8lXW8c5zDWur9jOfWtrP4Ha1ca3msb90BeCfxOtheWn-hix5ueujBmStcbqYX3z8aCFWQ.YGTA-g.njSn1OKMjDHbbGfBSzI6ZOm83Sw; HttpOnly; Path=/
X-Robots-Tag: noindex, nofollow, noimageindex
X-Content-Type-Options: nosniff
X-Frame-Options: sameorigin
Transfer-Encoding: chunked
Content-Encoding: gzip

207
...........S.n.0...3xVd.%..?....F.4EY.hR.......RR...S$.I\..pfwuB..vB...Z.,j.$.%.s\..M..G.....8j.......E....A............DvH}.ce-........Q.@....U.Pz....;...at!}.*.p*....."RrQd......J1z....}&..r\..(W........2.!.....#..5.gxt.o......L.........~...........@&.V..Yt=B..f.^O 
...DK.v.e.*c...].....J.....CG.x...8.....h...Ag.W.3.n.@|	>......f..&...3..].i3..f..6..N.i4Y.5"=..P@7&.+...Jb....C.;h..`........_..t.........fl......K...V....O.........N..[;...&..wu............\.?u'.e/.............a;.......j...0....j.`5.j..=..
........
[... 80 lines cut ...]
03a0   6c 6f 63 61 6c 22 2c 22 70 61 74 68 22 3a 22 31   local","path":"1
03b0   32 33 22 2c 22 72 65 66 73 22 3a 7b 22 72 65 73   23","refs":{"res
03c0   6f 75 72 63 65 22 3a 22 68 74 74 70 3a 2f 2f 6f   ource":"http://o
03d0   63 74 6f 70 69 2e 6c 6f 63 61 6c 2f 61 70 69 2f   ctopi.local/api/
03e0   66 69 6c 65 73 2f 6c 6f 63 61 6c 2f 31 32 33 22   files/local/123"
03f0   7d 2c 22 73 69 7a 65 22 3a 30 2c 22 74 79 70 65   },"size":0,"type
0400   22 3a 22 66 6f 6c 64 65 72 22 2c 22 74 79 70 65   ":"folder","type
0410   50 61 74 68 22 3a 5b 22 66 6f 6c 64 65 72 22 5d   Path":["folder"]
0420   7d 2c 7b 22 63 68 69 6c 64 72 65 6e 22 3a 5b 5d   },{"children":[]
0430   2c 22 64 69 73 70 6c 61 79 22 3a 22 74 65 73 74   ,"display":"test
0440   22 2c 22 6e 61 6d 65 22 3a 22 74 65 73 74 22 2c   ","name":"test",
0450   22 6f 72 69 67 69 6e 22 3a 22 6c 6f 63 61 6c 22   "origin":"local"
0460   2c 22 70 61 74 68 22 3a 22 74 65 73 74 22 2c 22   ,"path":"test","
0470   72 65 66 73 22 3a 7b 22 72 65 73 6f 75 72 63 65   refs":{"resource
0480   22 3a 22 68 74 74 70 3a 2f 2f 6f 63 74 6f 70 69   ":"http://octopi
0490   2e 6c 6f 63 61 6c 2f 61 70 69 2f 66 69 6c 65 73   .local/api/files
04a0   2f 6c 6f 63 61 6c 2f 74 65 73 74 22 7d 2c 22 73   /local/test"},"s
04b0   69 7a 65 22 3a 30 2c 22 74 79 70 65 22 3a 22 66   ize":0,"type":"f
04c0   6f 6c 64 65 72 22 2c 22 74 79 70 65 50 61 74 68   older","typePath
04d0   22 3a 5b 22 66 6f 6c 64 65 72 22 5d 7d 5d 2c 22   ":["folder"]}],"
04e0   66 72 65 65 22 3a 31 32 37 39 37 38 35 33 36 39   free":1279785369
04f0   36 2c 22 74 6f 74 61 6c 22 3a 31 35 33 38 32 37   6,"total":153827
0500   38 31 39 35 32 7d 0a                              81952}.

```

neilyoung, 2021-03-31:
So you create a REST request and get a response. What can prevent you from showing this in your GUI? Don't get it.

neilyoung, 2021-03-31:
BTW: Works in the world's stupidest browser: Firefox...Jeesss

neilyoung, 2021-03-31:
...and in a way more stupid browser: Safari... Not in Chrome :)

LazeMSS, 2021-03-31:
@neilyoung  have you tried connecting to your rpi using its ip instead of the http://octopi.local?

neilyoung, 2021-03-31:
Sure, but I can try again tomorrow.

neilyoung, 2021-04-01:
OK, thank you all. I have abandoned OctoPrint

closed 2021-04-01 by neilyoung