OctoPrint/OctoPrint #4061: [maintenance] Error in all file analysis

cp2004, 2021-03-29:
### The problem

Since it was mentioned that 1.6.0rc is going to start tomorrow... Here's a bug:
```2021-03-29 14:04:23,720 - octoprint.filemanager.analysis - DEBUG - Got output: "DONE:0.0s\r\nERROR:Invalid analysis result, please create a bug report in OctoPrint's issue tracker and be sure to also include the GCODE file with which this happened\r\n"
2021-03-29 14:04:23,720 - octoprint.filemanager.analysis - ERROR - Analysis for local:Connection_Test.gcode ran into error: Invalid analysis result, please create a bug report in OctoPrint's issue tracker and be sure to also include the GCODE file with which this happened
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.6.0.dev404+g631085f78

### Operating system running OctoPrint

Windows 10

### Printer model & used firmware incl. version

VIRTUAL

### Browser and version of browser, operating system running browser

Microsoft Edge Version 89.0.774.63 (Official build) (64-bit)

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

Any gcode file will do it, I'm always getting some kind of logged error. It can't process them at any time, if the server is restarted they are still not processed. 

Full log:
[octoprint - 2021-03-29T140714.476.log](https://github.com/OctoPrint/OctoPrint/files/6221893/octoprint.-.2021-03-29T140714.476.log)
[bundle viewer](https://bundleviewer.octoprint.org/?url=https%3A%2F%2Fgithub.com%2FOctoPrint%2FOctoPrint%2Ffiles%2F6221893%2Foctoprint.-.2021-03-29T140714.476.log)

gcode:
Any I have tried it with, but here are some of them:
https://github.com/OctoPrint/OctoPrint/blob/master/tests/cypress/fixtures/e2e-test.gcode

[Connection_Test (1).zip](https://github.com/OctoPrint/OctoPrint/files/6221925/Connection_Test.1.zip)
^ not sure where that one came from, probably the forums or discord. It looks like valid gcode to me.

foosel, 2021-03-29:
That *actually* looks like it is doing what it's supposed to be doing, but being a bit too scary about it. Those test codes don't contain a printed model and thus the analysis result contains no volume data. 1.6.0 now detects that and refuses to write an analysis result for such cases, since it would result in invalid data to end up in the metadata (e.g. infinity or null sizes). 

The error message seems to need some work however.

cp2004, 2021-03-29:
Ok that makes sense. I was just doing as I was told 😄

foosel, 2021-03-29:
And I'm grateful for it!

foosel, 2021-03-29:
Fixed by the above commit.

commit 72dfa803 (2021-03-30): 🔇 Don't log error for empty analysis results

Closes #4061
closed 2021-03-29 by foosel