OctoPrint/OctoPrint #4062: GCode Viewer does not return to life view after checking previous layers

neilyoung, 2021-03-29:
### The problem

After checking previous layers in GCode Viewer the view does not return to the life view, so a browser refresh is required. This is not really a bug, just an annoyance

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.5.3

### Operating system running OctoPrint

I'm using your image

### Printer model & used firmware incl. version

Creality Ender 3D

### Browser and version of browser, operating system running browser

macOS Big Sur, Chrome 89.0

### Checklist of files to include below

- [X] octoprint.log
- [ ] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2021-03-29:
Not even if you recheck this box?

[image]

neilyoung, 2021-03-29:
Hi Gina,

Thanks for the quick answer. In fact, it is checked already:

[image]

foosel, 2021-03-29:
That sounds weird, because it should uncheck itself when you scroll to prior layers or otherwise disconnect the view from the live print. Try toggling it, see if this fixes things. Also check safe mode please and provide octoprint.log and the contents of your JS error console.

neilyoung, 2021-03-29:
OK, will do. Next print is on the way.

neilyoung, 2021-03-29:
Ah ok, you are right: I wasn't aware that I have to set this flag again. I thought it would be a permanent setting. So yes, behaviour confirmed and if I recheck the thing, it shows the current state.

Thanks

closed 2021-03-29 by neilyoung