OctoPrint/OctoPrint #4064: [RC Feedback] Feedback on the 1.6.0rc1 Release Candidate

foosel, 2021-03-30:
Please provide *general* feedback on your experience with the 1.6.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2021-03-30:
1.6.0rc1 installed successfully. I do have a question. I didn't click on all the release note links before I clicked on upgrade. Is there an easy way to find the release notes sorted by date of install?

foosel, 2021-03-30:
Currently not, no. Might make sense to add some sort of install log to the software updater maybe and keep those there.

cp2004, 2021-03-30:
That's a good idea to be fair, many many 'i updated and now there is X issue', 'what did you update?, 'i can't remember' conversations about...

foosel, 2021-03-30:
In principle we do now also have `plugin_softwareupdate_console.log` and `plugin_pluginmanager_console.log` included in the systeminfo bundle, so that should at least allow us to see what was updated and what kind of issues it ran into, but something more human readable would probably also be nice. I'll extract this into a request.

zeroflow, 2021-03-30:
Works fine on a Pi 3B & Prusa MK3S

The-EG, 2021-03-31:
Seems to be working well on my setup: rpi4b, 2 instances of OP, 2 ender 3's, 2 cameras, etc

One note, as expected upgrading one instance to the RC and restarting it causes the weird behavior from the other instance until it's manually restarted (they share a single venv, different --basedirs). Yep, it's a weird/unsupported setup; normally I can do something to one instance and then use the web interface to restart the other, but in this case I had to ssh and manually restart the systemd service, the UI of the 'other' instance wasn't usable.

Another thing is that the new panel in the plugin manager looks like it's introducing styles that aren't accounted for by Themeify and likely other UI customization plugins:
[image]

cp2004, 2021-03-31:
> Another thing is that the new panel in the plugin manager looks like it's introducing styles that aren't accounted for by Themeify and likely other UI customization plugins:

That would be my changes to the UI, I couldn't figure out a way to inherit the colour on the sticky navigation bar.

Using custom CSS if I remember right, the selector is `#settings_plugin_pluginmanager_navbar`. I meant to send a PR to the plugins at the time but forgot....

JohnOCFII, 2021-03-31:
TL;DR - 1.6.0rc1 seems to be working fine.

I decided to do a backup and then re-image from an rc of Octopi 0.18 to the current release, and then do a restore.  Other than the understandable need to upgrade from the OctoPi version of OctoPrint to the 1.5.3 version used for my backup, that process worked fine.

- Uploaded new GCODE
- Printed GCODE on Prusa MK3S/wMMU2 in single extruder mode
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- The Prusa MMU2 Select Filament plugin worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Telegram plugin worked well

JohnOCFII, 2021-03-31:
I did have a question about this gray slot icon in the Rendering Timelapse window.  Is it supposed to be indicating progress?  The oblong slot did not change while it was displayed.  There was a similar oblong slot for the next green pop-up window as well.
[image]

ChrisHeerschap, 2021-04-01:
Installed on rpi4 (Prusa MK3S) rpi3 (Prusa Mini) and rpi2 (Anycubic i3 Mega) with no issues. Have sent prints to the first two with no issues. Didn't see the "gray slot icon" mentioned by @JohnOCFII but am wondering if that's a scroll bar? Ah, yes, the print from the rpi3 showed the rendering timelapse and that is a scroll bar made visible by the long filename:

[image]

I guess safari just makes the scroll bar really fat which does make it look like a progress bar.

Haven't sent a print to the rpi2 only because I think I don't have filament loaded and there's no light on that webcam. Will try tomorrow, but am assuming there will be no problems based on the first two working. Won't update if everything goes well.

Installed Active Filters Extended to suppress temp messages by default, no issues. Clicked through all the menus and the only thing that looked wonky was the previously mentioned white background on the installed plugins page. Only an issue on the rpi4 with the dark theme, otherwise the other two with the default light theme you'd never know.

Tested upload via API from PrusaSlicer, that worked well. Realized I had it set to start printing with the wrong filament loaded and something on the print bed, so tested the emergency stop M112 plugin which also worked. Reconnected to the printer and all is well. 

Nothing I saw that caused me any concerns.

foosel, 2021-04-01:
This slot is indeed a scrollbar. Safari by default doesn't show scrollbars when they are needed, that caused issues (see #3987) so when Safari is detected OctoPrint will now enforce scrollbars. Apparently with that particular notification there's need for some CSS to keep the filename from making it to have to scroll to get rid of the bar in that case.

Taomyn, 2021-04-01:
Upgrade went fine for my installation, not seeing any major issues but have not had the chance to do any actual printing yet. and so far just a minor issue with Themeify which I reported here: https://github.com/Birkbjo/OctoPrint-Themeify/issues/118

arontaupe, 2021-04-01:
Update from 1.5.3 to 1.6rc1 destroyed the distribution entirely? after update pi 4 refuses to boot and shows firmware missing error. still have to check for sd failure but it appears normal in windows.

loskexos, 2021-04-01:
Upgrade went fine. 4 prints so far as well (rpi3b/modified mk3s). Upload from prusaslicer, prusaslicer thumbnails, print, abort, print again, timelapse, multicam, heater timeout, tempsgraph, prusa levelling . All worked. Logs do not show anything unaccounted.

Remark: When I did hit the upgrade button I thought ..... wooops, should've made a config backup first :). Might be a good thing to ask or automatically do a config backup during upgrade. In case something goes utterly wrong and revert/reinstall needed.

jneilliii, 2021-04-01:
So @cp2004 and I have been discussing the possibility of an issue with the settings dialog opening on Discord. He's going to investigate further in the morning, but when opening settings it seems to be laggier/slower to swipe down from the top of the screen, it's jittery while it's swiping down or in some cases doesn't animate at all and just shows up. 

My assumption is it's related to #3949 as an additional API call was added the settings modal open callback, but would think those were done asynchronously.

jneilliii, 2021-04-02:
Here's a video example of what I mean comparing 1.6.0rc1 with no plugins and 1.5.3 with several plugins.

https://youtu.be/vXaNZwPINPY

jneilliii, 2021-04-02:
Here's an example video showing the comparison between 1.6.0rc1 with no plugins and 1.5.3 with several plugins.

https://youtu.be/Eol6i5O4UaU

Patbonamy, 2021-04-02:
Works fine with Ender 5 Pro btt skr mini E3 V2, pi 3 b+.
9 hours print without problems.
Thank you for the update.

KenLucke, 2021-04-03:
Running on 3 RPi 4b 4GB / CR-10v2 setups, with Meatpack and ArcWelder working well together, and other than the Themeify issue already mentioned above, seems to be functioning fine, although on one instance the GCODE Viewer races several layers ahead of the actual progress as it used to, the others don't, even with identical plugins.  I'm ass/u/mining it's something I'm not catching that is specific to that one setup, so I will investigate further before calling it a bug.

mobby999, 2021-04-04:
There is a long delay now with the RC1 and webcams (even after restart of the systems).
I have two Raspberries 4.0 here with 2 GB and identical setup (also for Octoprint). Printers are a Caribou and a Prusa MK3S with C270 Webcams from Logitech.

Have a nice Easter!

Greetings from Ratingen
Raimund

popshansen, 2021-04-05:
Running the 1.6rc1 on one printer and still have 1.53 on another to compare. :-)
The only thing I have encountered so fare is that the 1.6rc1 have knocked out the "Action Commands" v0.3 plugin,
have reinstalled it but with out success, it shows installed but does not show up in the plugin list and it is not running.
So OctoPrint doesn't react on my custom commands from Marlin.

benlye, 2021-04-05:
> Running the 1.6rc1 on one printer and still have 1.53 on another to compare. :-)
> The only thing I have encountered so fare is that the 1.6rc1 have knocked out the "Action Commands" v0.3 plugin,
> have [...]

Will have a new version out which fixes this in a few minutes.  Apparently I'm one of the very few plugin owners who were using the very old, deprecated, plugin property `__plugin_init__`.  My bad.

benlye, 2021-04-05:
Action Commands patched and v0.4 is available.

RC1 seems pretty good for me too.  I've updated the Firmware Updater plugin to take advantage of the new `octoprint.printer.handle_connect` and it works well - no more OctoPrint serial connections interfering with firmware updates.

popshansen, 2021-04-05:
Confirm that Action Commands v0.4 is working with 1.6rc1 :-)

gege2b, 2021-04-06:
Installed RC1 a last week, and I now have a lot of sluttering while printing via octoprint (Ok via SD card)

octoprint process is > 100% CPU time on the pi most of the time
[image]

I still need to figure out a few things :
 - I upgraded from py2 to py3 circa two weeks ago, but I printed some stuffs without issue after that
 - I'm not sure how it could affect this, but the two gocde where I got this problem was sliced from AMFs and not STLs

foosel, 2021-04-06:
@mobby999 I see you edited your comment to say "it's gone now" - what changed? In any case, OctoPrint doesn't directly interface with the camera, so any kind of loading issues or such can only be a case of wrong configuration, network problems or are caused by whatever webcam server is being accessed.

@gege2b please open a complete bug report. Be sure to try in safe mode & share a system info bundle.

foosel, 2021-04-06:
> So @cp2004 and I have been discussing the possibility of an issue with the settings dialog opening on Discord. He's going to investigate further in the morning, but when opening settings it seems to [...]

Interesting that y'all are seeing this. I've been seeing this off and on for about a year now, but chalked it up to my development environment because that was the only place I could actually reproduce it. I wouldn't call it a regression, however it could have been made worse by changes. I'll take a close look at that as well, though if I understood @cp2004 on Discord correctly, it sounded like there was no specific reason to pinpoint.

In any case, as a bit of a timeline feedback here... I'm not sure I'll manage to take care of all analysis and fixes for a second RC until tomorrow, and I'd rather like to avoid even RC releases past Wednesday (general rule after I had some bad release experiences in the past). So RC2 will *likely* not launch until next week, giving everyone some more time for testing. I'd especially love to see a ticket from @gege2b on the stuttering issue, and also get some more details from @mobby999 on the webcam issue.

In general, let me reiterate that if you observe any kinds of issues with an RC, **always provide a system info bundle at the very least**. Better yet, fill out a full bug report. That saves us from time consuming back and forth while trying to iron out issues. Thanks :)

gege2b, 2021-04-06:
@foosel ticket was in progress but the slutter seems to be gone while I was testing O_o

cp2004, 2021-04-06:
> I'll take a close look at that as well, though if I understood @cp2004 on Discord correctly, it sounded like there was no specific reason to pinpoint.

That was my conclusion - there was no single thing slowing the settings dialog down, the `onSettingsShown` callback is the only one that's doing anything. Between that and the sheer amount of markup that the browser is trying to throw around with the settings modal, it is not a surprise the performance is not great. But yeah, intermittent and not always the same thing.

foosel, 2021-04-06:
@gege2b definitely sounds like a third party plugin issue then though... let's see if it comes back

@cp2004 yeah... I'm wondering if it might be best to just disable animations altogether here.

kazibole, 2021-04-06:
Updated from OctoPrint 1.5.3 to 1.6.0rc1. Was not able to start a print from the printer's sd card (clicking "Load and Print" did nothing).

Opened issue #4082

berrystephenw, 2021-04-07:
Updated to 1.6.0.rc1 - no issues during the update and it seems to run fine (1 print only at this point). All my plugins kept their settings!
I did see one thing in the log:

> 2021-04-07 20:42:16,480 - zeroconf - WARNING - Choked at offset 0 while unpacking b'\n\xba\x10\x00'
> 2021-04-07 20:42:16,481 - zeroconf - WARNING - Exception occurred:
> Traceback (most recent call last):
>   File "/home/pi/oprint/lib/python3.7/site-packages/zeroconf/__init__.py", line 713, in __init__
>     self.read_header()
>   File "/home/pi/oprint/lib/python3.7/site-packages/zeroconf/__init__.py", line 736, in read_header
>     ) = self.unpack(b'!6H')
>   File "/home/pi/oprint/lib/python3.7/site-packages/zeroconf/__init__.py", line 723, in unpack
>     info = struct.unpack(format_, self.data[self.offset : self.offset + length])
> struct.error: unpack requires a buffer of 12 bytes

full log attached.
[octoprint.log.txt](https://github.com/OctoPrint/OctoPrint/files/6275069/octoprint.log.txt)

I do have spotty internet on this Pi - a starlink satellite connection that has regular dropouts.

foosel, 2021-04-08:
@berrystephenw That looks like some invalid network package that got caught by the zeroconf dependency, outside of OctoPrint's reach. It just ended up in the log due to logging level settings.

foosel, 2021-04-12:
Another quick update on the progress, I'm currently trying to find a file with which I can reproduce #4086. Any help welcome, if you've seen any kind of weirdness with the gcode viewer while "zoom in on model" was enabled then please add reproduction steps to the ticket. This is currently blocking the go forward with RC2. Idea is to see if I can figure something out tomorrow, otherwise push RC2 with a slight refactoring I just did to the involved code in the hopes that this will solve it.

popshansen, 2021-04-12:
> Another quick update on the progress, I'm currently trying to find a file with which I can reproduce #4086. Any help welcome, if you've seen any kind of weirdness with the gcode viewer while "zoom [...]

I tried the "zoom in on model" on a print this evening and set the show previous layer, and some flicker with different layer was rendered. I suspect the z-hop/lift causing this as it is basicly a layer shift up and down and the GCode viewer render the layer it is one for a moment. Maybe this could be a hint on the hunt for the #4086.

foosel, 2021-04-13:
@popshansen I've also been suspecting the zhop, however, the viewer *should* always render that on the layer it's triggered on, not the next one (that is, as long as it's a z-hop that looks like retract, z move, move, z move, extrude -- if extrusions happen with the z change, which should usually *not* happen during a z hop but only at start and end on the originating layer, detection will fail, as there's *nothing* else in the gcode to go on to detect this stuff, not even slicer specific comments)

foosel, 2021-04-14:
1.6.0rc2 is out, new ticket is here: #4093

closed 2021-04-14 by foosel