OctoPrint/OctoPrint #4069: [Request] Meaningfull errormessage for disabled install of Plugin

ozett, 2021-04-01:
### Is your feature request related to a problem? Please describe.

i wanted to install a new plugin, but the install is blocked. 
The errormessage does not help to solve this

### Describe the solution you'd like

like to have an link from the error to solutions

### Describe alternatives you've considered

alternative i used: searched the forum and found manual install. this workaround helped

as i [wanted notes in my installaiton](https://github.com/OctoPrint/OctoPrint/issues/4068#issuecomment-811369115) and tried to install this plugin, but install is blocked

[image]

i went to the issues and found [this solution](https://github.com/OctoPrint/OctoPrint/issues/3849#issuecomment-743824970):

```
The following command line installation workaround, when logged in via SSH, works for me:

cd ~/oprint/
source bin/activate
pip install "INSERT_PLUGIN_INSTALL_URL_HERE"
sudo service octoprint restart

```
maybe [the whole issue is related](https://github.com/OctoPrint/OctoPrint/issues/3849) and worked on in v1.6.x ?

i also have a **raspi 3B** !!!!!

my specs:
```
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.90 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 916779008
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0xD0005
env.python.pip: 20.2.4
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Prusa-Firmware-Buddy 4.3.0-RC1+2691 (Github)
```

foosel, 2021-04-01:
```
env.plugins.pi_support.throttle_state: 0xD0005
```

Your Pi has undervoltage, the power supply you are using with it is insufficient. Refer to https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256

I agree though, this should be linked from the block message.

---

> (maybe the octoprint-info page for the specs should list the installed 3rd party plugins & the diskspace & platform (octopi-os) as source of information for bug-hunting?)

If there's an issue with a third party plugin, that will come up when you test in safe mode (which you are always asked to do in a bug report and a support request). It's also contained in the logs (or system info bundle) you are always asked to share.

The OctoPi version is in there:

```
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0xD0005
```

Diskspace might make sense indeed.

ozett, 2021-04-01:
> Your Pi has undervoltage, the power supply you are using with it is insufficient. Refer to https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256

my [solution-thread](https://github.com/OctoPrint/OctoPrint/issues/3849) stated that he had the similar problem with plugin install even supplying power from a perfekt lab-bench.

somehow illogical that power causes this , because the pi-power is unchanged but allowed my to install already 3 third-party plugin via gui. only for the 4th plugin it becomes picky about unobservable "undervoltage" problem an block more plugins? 
i did not see any effects of the undervoltage problem until this

 would be more logical if blocking would be on clear critera like on/off=> 

underpower == no octotprint at all

instead of 

underpower == allow octoprint without notice and up to three plugins, but than block further plugins ?

cp2004, 2021-04-01:
If OctoPrint refused to start when it was undervoltage, we would upset a *lot* of people.

Plugin installs and updates are disabled when the system is throttled and unstable due to undervoltage, it won't allow 3 and then block the 4th. This is coincidence that the power supply started failing around that time. There should also be a lightning bolt icon in the navbar, with a link to the forum FAQ post.

The very person you quote about undervoltage issues from their 'perfect' lab supply confirmed this is the issue in later comments: https://github.com/OctoPrint/OctoPrint/issues/3849#issuecomment-743912312

ozett, 2021-04-01:
ok, i see... 
i run more than 4 other PIs since years with undervoltage without observing any effect in their (simple) functions.
might be critical for 3d-printing with octoprint and serial-communication.

i would like to see a little more obvious warning to change the powersupply instead of blocking some sub-functions like the plugin-install on a somehow random day or time. and the flash-icon is sooo mini..., As the PI runs headless, there is no startup-warning to see for underpower and maybe the whol GUI should become Warning-RED ? or at leat the uppber bar? so that i feel forced to action ...

[image]

foosel, 2021-04-01:
> i would like

That is great. A lot of people don't, and when I still do it who do you guess gets made their life miserable over it? I have received personal attacks just for including a lightning bolt at all. I cannot just consider what you want, I have to consider what everyone wants, what impact it might have, and so on.

The current solution is the result of one such considerations. It isn't big, but it blinks. It is red. OctoPrint screams at you in the log file, it tells you there's an underpower condition with a big yellow warning box on the pages where functionality is disabled due to the undervoltage. Yes, future version will include a link to the FAQ on that THERE as well.

And as @cp2004 already said, there's no countdown or anything in there when there's undervoltage. If there's an active condition (like it was here) stuff will get blocked, because the alternative would be to let things proceed and potentially corrupt the installation like it has happened in the past. If there's no active condition, stuff won't be blocked. If you have intermittent failure, you might intermittently be able to install plugins or not. 

I can't hold the hand of everyone without alienating large parts of the userbase. Pay attention to stuff that is red, blinking or otherwise screaming for attention and consult the FAQ which usually is linked from all places necessary (as already admitted, this one here slipped through).

ozett, 2021-04-01:
thanks for long explanation.

as i understand that the gui-layout  is the result of requests from multiple voices of the userbase from the past, the "i would like" were my 3 add-words to this userbase sound. i found my old posts from 2015 in the issues and was reminded that i am missing some years of discussion here. but i get along with the latests quirks to bite my way through the issues and FAQ.  sorry to bother your here in the issues if i dont find solutions or answers there and than consider some annotations as request or bugs. my fault. but the long answer shed some light to the history of the small flash... 

by the way 5 minutes ago i changed the powersupply after going to this lecture https://elinux.org/RPi_Hardware#Power_Supply_Problems and after your issue-comments i now have a clue to the difference of "0xD0005" vs "0x80000". 💭 

[image]

thanks for all. and the great octoprint. 
should i close this?

ozett, 2021-04-02:
my hardware issues showed up with blocked plugininstall. i investigated now further and my raspberry 3b seems to need some explicit cooling while intense rendering with ffmpeg for the timelapse, the SOC felt hot and rebootet even with new powersupply. i exchaned the powersupply again with different type and again beefier one and will print a cooling-case to mount a cooler.

maybe underpowering/inusfficent powersuppy for special models of raspberries and also in combination of certaing quality state of cooling affects in resulting to endanger the stability of octopi-os and the octoprint-gui. hard to investigate.

As i understand now that there are reasons not to show/monitor  this system-states on the genuine octoprint-gui. 
but i found it helpful while printing to be informed about the progress of few system-key-values like CPU-Temp and processor-state while running some octoprint functions in background and therefore i here only mention this not as a REQUEST but as an idea for considering it maybe helpful to have this all in one place at the gui. 
especialy on octopi-os, wich makes the raspberry a special-tool and hides somehow the rest of the raspberry-os.

i myself make no REQUEST instead keep this separate from octoprint and use [rpi-monitor](https://github.com/XavierBerger/RPi-Monitor) and will try also [webmin](https://www.webmin.com/).
But if anybody else ever find it helpfult to have SOC-STates on the Octoprint-gui i will give it my 👍 

[image]

cp2004, 2021-04-02:
I saw the notification and was going to suggest resource monitor plugin, I see you've found it 😄. 

> should i close this?

You can leave it open for the time being, the link to the FAQ in the message that Gina said should be added will still get done, it is on the 1.7.0 milestone.

The throttle state can be decoded using the table in the Raspberry Pi docs: https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md. I agree it is hard to work out, which is why the icon should show up as OctoPrint can work it out.

In your case, with 0x80000 it is reporting that 'Soft temperature limit has occurred'. OctoPrint should show the icon for overheating as well. But anyway, you now have an awesome plugin to show you the exact temperature 😉.

foosel, 2021-06-22:
FAQ link has been added to both Plugin Manager and Software Update alerts.

commit 146c8315 (2021-06-22): 🚸 Add FAQ link to throttled messages

Closes #4069