OctoPrint/OctoPrint #4070: Print Time Left Shows Wrong value

mrspaceman, 2021-04-01:
### The problem

I start a print and the "Approx Total Print Time" shows a value similar to my slicer (Cura 4.8.0)

part way through a print (exactly when varies) the "Print Time Left" will increase to a ridiculous value

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint: 1.6.0rc1 

### Operating system running OctoPrint

Rasbian

### Printer model & used firmware incl. version

Ender 3 v2

### Browser and version of browser, operating system running browser

chrome Version 89.0.4389.90 (Official Build) (64-bit)

### Checklist of files to include below

- [X] octoprint.log
- [X] serial.log or contents of your terminal tab (always include in cases of issues with printer communication)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

[Screenshot showing times left](https://user-images.githubusercontent.com/378813/113262333-65904400-92c8-11eb-87bd-17b004ce25c1.PNG)
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6241965/octoprint.log)
[octopi-1617264531106.log](https://github.com/OctoPrint/OctoPrint/files/6242030/octopi-1617264531106.log)
cant upload gcode file.

Terminal tab:
`Recv: ok
Send: N6392 G1 F3000 X108.865 Y68.314 E42.48993*58
Recv: ok
Send: N6393 G0 F9000 X107.252 Y68.313*68
Recv:   TT::210.43210.43  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::8383  BB@@::3030
Recv: 
Recv: ok
Send: N6394 G1 F3000 X84.913 Y151.684 E44.21237*50
Recv: ok
Send: N6395 G0 F9000 X85.841 Y151.684*65
Recv:   TT::210.00210.00  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::9191  BB@@::3030
Recv: 
Recv: ok
Send: N6396 G1 F3000 X58.691 Y124.534 E44.97861*58
Recv: ok
Send: N6397 G0 F9000 X58.692 Y125.239*66
Recv: ok
Send: N6398 G1 F3000 X161.306 Y97.744 E47.09862*63
Recv: ok
Send: N6399 G0 F9000 X161.307 Y98.401*68
Recv:   TT::209.92209.92  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::9292  BB@@::3030
Recv: 
Recv: echo:busy: processing
Recv:   TT::209.37209.37  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::104104  BB@@::3030
Recv: 
Recv: ok
Send: N6400 G1 F3000 X147.03 Y151.684 E48.19944*54
Recv: ok
Send: N6401 G0 F9000 X145.454 Y151.685*127
Recv: ok
[... 246 lines cut ...]
Recv:   TT::209.45209.45  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::100100  BB@@::2828
Recv: 
Recv: echo:busy: processing
Recv:   TT::210.00210.00  //210.00210.00  BB::60.0060.00  //60.0060.00  @@::8888  BB@@::2828
Recv: 
Recv: ok
Send: N6477 G1 F3000 X161.307 Y125.206 E113.05059*63
Recv: ok
Send: N6478 G0 F9000 X161.307 Y122.678*112
Recv:   TT::210.47210.47  //210.00210.00  BB::59.9959.99  //60.0060.00  @@::7878  BB@@::3232
Recv: 
Recv: echo:busy: processing
Recv:   TT::210.62210.62  //210.00210.00  BB::59.9759.97  //60.0060.00  @@::7676  BB@@::3737
Recv: 
Recv: echo:busy: processing
Recv: ok
Send: N6479 G1 F3000 X58.691 Y150.174 E115.17064*15
Recv:   TT::210.62210.62  //210.00210.00  BB::59.9759.97  //60.0060.00  @@::7777  BB@@::3838
Recv: 
Recv: echo:busy: processing
Recv: ok
Send: N6480 G0 F9000 X58.691 Y149.87*125
Recv:   TT::210.62210.62  //210.00210.00  BB::59.9759.97  //60.0060.00  @@::7777  BB@@::3939
Recv: 
`

cp2004, 2021-04-01:
Can you confirm that this didn't happen in OctoPrint 1.5.3 or earlier?

It's well known that the analysis can be inaccurate, if it is showing a red dot beside it then OctoPrint *knows* it is inaccurate.

foosel, 2021-04-01:
I didn't touch print time estimation between 1.5 and 1.6, so it cannot be a regression.

Sounds more like the file so non uniform that half way through the linear approximation diverges SO much from the estimate it will assume the estimate was wrong in some way and rather go with linear altogether. Would need the file however to confirm, so please share, at the very least in private to gina -at- octoprint.org

foosel, 2021-04-06:
@mrspaceman still waiting for a file here to analyse. I can't look into this without that.

mrspaceman, 2021-04-06:
Sorry, the print time issue happened with this (as well as all my other) files

[Everest.zip](https://github.com/OctoPrint/OctoPrint/files/6263183/Everest.zip)

foosel, 2021-04-06:
Then please share some of the others as well. Looking at that it is no wonder that OctoPrint's attempt at linear estimating based on progress and calculated total will be deemed invalid after a while, the shape gets constantly smaller with a big AF bottom, so is anything but linear and thus a prime example of cases where the estimate will produce invalid results.

Have you tried the same files through 1.5? Is there a difference in plugins that are enabled and/or installed? Do you maybe used to have Print Time Genius installed and it didn't process these files?

Again, estimation wasn't really touched apart from one bug fix that I don't see how it could impact things here.

mrspaceman, 2021-04-06:
heres another one 
[paint-dish_v2-1.zip](https://github.com/OctoPrint/OctoPrint/files/6266025/paint-dish_v2-1.zip)

mrspaceman, 2021-04-07:
I am new to Octoprint and the only plugins I have installed are:
- FileManager
- Octolapse (disabled)
- Creality-2x-temperature-reporting-fix (0.0.4)
- Telegram Notifications

foosel, 2021-04-07:
That still leaves open the question whether you see the same behaviour under 1.5.3 or not.

mrspaceman, 2021-04-07:
I upgraded to 1.6.0rc1 as soon as I installed Octoprint, so I dont know if the same would happen in earlier version.

foosel, 2021-04-07:
Then please downgrade to 1.5.0 and verify if the same issue exists there. 1.6.0rc1 - as the name suggests - is still a release candidate and not a stable release. I need to know whether the issue is something introduced by 1.6 (and thus a regression) or if it already existed in 1.5. The whole point of release candidates is to test new functionality and to identify regressions in existing functionality before stable rollout - so a comparison point with earlier stable releases is pretty important.

foosel, 2021-04-07:
FWIW, I just took a look at the analysis results of the Everest file against 1.5.3 and 1.6.0rc2.dev.

1.5.3:

[image]

1.6.0rc2.dev:

[image]

The only difference there is the calculated printed model height, and that is due to a bugfix in the model size calculation (#4036). 

With that, the shape of the model, the described issue and the screenshot this reeks of the usual issue with non uniform and non linear models where OctoPrint's analysis backed linear estimate while printing will be deemed invalid after it leaves the limits of sane expected values based on the estimate and fall back to a full linear estimate. 

Suggestion: Install the Print Time Genius plugin, that does some more (resource intensive) stuff than the bare minimum that OctoPrint can support for all possible runtime scenarios. If that *doesn't* fix it, request a reopen of this ticket. For now I'll close this as known issue and won't fix since the limitations of the print time estimation are well known, I've explained where the issues lie repeatedly here and elsewhere and there exists a solution that works for most people (the aforementioned plugin).

closed 2021-04-07 by foosel