OctoPrint/OctoPrint #4073: strange sorting behavior when navigating folders

jneilliii, 2021-04-02:
### The problem

1. Have multiple folders created, and file list sorted by upload date descending.
2. Click on 2nd folder in list and then click back.
3. Notice previously exited folder is at top of list instead of in the second position.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why

n/a

### Version of OctoPrint

1.5.3, 1.6.0rc1

### Operating system running OctoPrint

octopi 0.18

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

Google Chrome Version 89.0.4389.114 (Official Build) (64-bit), Windows 10 64 Bit

### Checklist of files to include below

octoprint.log
Screenshots and/or videos showing the problem (always include in case of issues with the user interface)

I would expect the folders at the top of the list to be sorted by name when they show at the top of the list all the time with the "sort by folders, files" mode enabled.

Here's a quick video demonstrating the issue.

https://user-images.githubusercontent.com/5249455/113379549-d32f8500-9347-11eb-9524-1a032b168f8a.mp4

No errors in js console.

[octoprint (70).log](https://github.com/OctoPrint/OctoPrint/files/6247630/octoprint.70.log)

foosel, 2021-04-08:
Can confirm, but only in Chrome, Firefox doesn't show this behaviour. Investigating.

foosel, 2021-04-08:
Fix ready for 1.7.0

commit 216568d2 (2021-04-08): 🐛 Fix non deterministic sorting bug

If folders are empty (no upload date) & sorted by upload, Chrome would
produce arbitrary sorting orders. The sorting now falls back on sorting
by name in such cases.

Closes #4073
closed 2021-10-11 by foosel