OctoPrint/OctoPrint #4074: Failure analyzing a gcode file results in a http 500 in the Frontend  and prevents loading the file

nicolas-lang, 2021-04-02:
### The problem

Please bear with the partially incomplete description, since I was able to trace the issue to a specific sourcecode line, I didn't really bother with reproducing and providing complete debug data.
1. Use Prusaslicer Version 2.3.0+win64 to create and upload gcode
2. Select/Load File to be printed in web UI
3. HTTP-500 in Http-Console

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

I didn't really bother to check, sorry.

### Version of OctoPrint

OctoPrint 1.5.3 Python 3.7.3 

### Operating system running OctoPrint

OctoPi 0.17.0

### Printer model & used firmware incl. version

Ender 3 with custom Marlin

### Browser and version of browser, operating system running browser

Chrome / Win 10 

### Checklist of files to include below

octoprint.log

[Octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6249836/Octoprint.log)

I have traced this issue to `/OctoPrint/src/octoprint/printer/standard.py`, line 1279.
Changing
` if "analysis" in fileData:`
to
` if "analysis" in fileData and fileData["analysis"]:`
fixes the issue. 

I would assume that somehow `fileData["analysis"]` can be set but still be `None`, but I have no clue how/where/if this can happen. 

I'll update with full debug data at a later date. (Very busy at the moment, but still wanted to report)

cp2004, 2021-04-02:
This should already be fixed in 1.6.0, please try the release candidate. (#4064)

Commit: https://github.com/OctoPrint/OctoPrint/commit/4531f44ed5a93684ebaab7cef92887d592583f3f

cp2004, 2021-04-02:
Wait, that is a different issue I think, might have to ignore that

nicolas-lang, 2021-04-02:
Might take a couple of days, since i am bulk-printing easter bunnys at the moment, but I'll do and post an update after testing 1.6.0

cp2004, 2021-04-02:
If you want to send a PR feel free :)
I agree, it makes sense to harden against this somewhere if there is a chance of it being 'None'.

foosel, 2021-04-07:
Hardened and ready for 1.7.0 (not a regression, so no inclusion in 1.6.0).

commit bcf57aae (2021-04-07): 🥅 Protect against invalid analysis & stats

Fixes #4074
closed 2021-10-11 by foosel