OctoPrint/OctoPrint #4077: action:notification being overused

dwery, 2021-04-03:
### Is your feature request related to a problem? Please describe.

Too many notifications. It seems that some firmware is using notifications to convey status messages rather than important events. 

### Describe the solution you'd like

Would be nice to have something to convey regular status updates, maybe action:status

cp2004, 2021-04-03:
This is not something we can sort out here - there's an option to disable the popup notification or you can disable the bundled plugin dealing with them entirely. It's up to the firmware to implement the action commands, if they want to implement an action: status command then that needs to be requested to the firmware vendor.

closed 2021-04-03 by cp2004

dwery, 2021-04-03:
If the firmware implements action:status and there is no corresponding feature on the client, it won’t be much useful.