OctoPrint/OctoPrint #4078: File manager use of slugify breaks multilingual support

richfelker, 2021-04-03:
### The problem

Attempt to upload a file named `འ.gcode` using the API or web interface. Expected behavior: file is uploaded. Actual behavior: mysterious "invalid type" error.

The mechanism of the bug is that, behind the scened, Octoprint renames everything to ASCII using "slugify" and only keeps the original filename to display back to the user. This makes multilingual filenames mostly seem to work, but break in mysterious ways. In the case of the test case here, "slugify" seems to collapse the name to zero characters, breaking the base/extension split and causing `gcode` not to be interpreted as an extension. But there are all sorts of other ways this could break too, possibly including overwriting unrelated files that happen to collapse to the same thing.

Ideally no name transformation should happen at all; names should be honored exactly as uploaded. If there are reasons this is problematic, the behind-the-scenes names should be converted in some one-to-one manner that is not lossy and that does not collapse multiple names to the same thing.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.4.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

octoprint.log

cp2004, 2021-04-05:
> In the case of the test case here, "slugify" seems to collapse the name to zero characters, breaking the base/extension split and causing gcode not to be interpreted as an extension.

That... is definitely an issue. Though it's not an empty filename, it makes it hard to identify the file.

> Ideally no name transformation should happen at all; names should be honored exactly as uploaded.

I think there would have to be some kind of transformation, since I don't believe the FS can work with all characters that are allowed to be uploaded. Maybe it can, I would happily be wrong. For example, emojis - do they work in filenames? OctoPrint will allow it in the 'Display' filename (as far as I know), but I haven't got a clue if I could save a file with a smiley face name to the filesystem. Sounds like it is time for an experiment :)

foosel, 2021-04-05:
Last I checked file systems can have many kinds of encodings for for names, so the only thing that will definitely work is ASCII only and hence it'll stay at transformation to that. However i agree that there needs to be a solution for situations like this one, eg falling back to utf8 symbol names.

richfelker, 2021-04-06:
On basically every system relevant in the past 20 years, filenames either consist of byte sequences agnostic to any encoding except that they cannot contain `/` or the null byte (all unices except OSX) or admit all Unicode characters except a small set of special characters (Windows) and possibly with rules about normalizartion (OSX). There is no reasonable basis for forcing ppl to use ASCII for filenames in this century, or for transliterating through ASCII in a way that makes non-latin filenames clash with one another in very surprising ways; doing so is hostile to multilingual support.

In case anyone is using Octoprint on a really weird host environment where native-language filenames don't work, it may be desirable to keep *an option* for doing what Octoprint is doing now, or just disallowing non-ASCII filenames for such a host environment. But this is not an issue on real-world host environments people are using, especially not rpi's.

It's also possible - and this is in some sense independent of the Unicode issue, but something else slugify might be solving - that there are places where the filenames are pasted into command lines without any escaping, assuming slugify produced names that don't include any shell special characters. If this is the case, removing the slugify would require either fixing that escaping or adding rules to disallow filenames with shell specials in them.

foosel, 2021-04-06:
I distinctly remember issues with files containing UTF-8 characters on *some* OS. There even was a bug for it. Might have been Windows, my memory is a bit fuzzy, but I remember that it was causing severe issues and took some effort to solve.

I can see the issue with slugifying to a collision, however, 1.6.0 has introduced a file name collision detection mechanism (at least on the UI, exposed on the API though and thus usable by third party clients), so the "accidentally overwriting something else" issue can be solved through that. 

With that being said, could you please elaborate why you even care on how OctoPrint stores stuff *internally*? From an API point of view, it could also be storing the file under the hash of the contents, or in a database or whatever. That it stores it with ASCII-fied names on disk in the regular file hierarchy is the way it is now, but that's not guaranteed to always stay that way. And the upload folder itself also is (and always has been) something you should keep your hands out off due to internal housekeeping anyhow. Frankly, I can't justify spending a significant amount of time now throwing the current file name sanitization out with the bathwater, potentially introducing breaking changes in the API, breaking plugins and third party clients in the process, just so the file names look nice on disk (if that doesn't even break in some setups, as I said, it has caused issues in the past).

I'm happy to look into issues like this one with the current approach, but I need some better arguments for the justifying any further changes, so I'd be happy if you'd share any other reasons why filenames *as stored on disk and exposed in the API paths* should not be ASCII apart from "it's 2021".

richfelker, 2021-04-06:
> With that being said, could you please elaborate why you even care on how OctoPrint stores stuff *internally*?

Sorry, I actually don't care a lot how it's done *internally*, as long as there are not nasty surprises with certain non-latin names clashing or being rejected. I'm not sure if this is entirely internal or not though. You mention API paths, and it seems unfortunate, surprising, and m17n-hostile if non-latin filenames can't be used in API calls to reference the files you previously uploaded.

If on the other hand this is entirely an internal matter, I only slightly prefer just storing the names as-is, for the sake of making things easier and less surprising if you login to the box running Octoprint to access the files directly in any way, but it's not a big deal if that doesn't work.

foosel, 2021-05-20:
1.7 will do its best to keep the file names as is, no more slugification. I could no longer reproduce any issues with this, but just in case there's a flag to enforce the old behaviour for maximum compatibility. Note that for all of this to work we rely on the `pathvalidate` library, and that is Python 3 only, so on Python 2 the old behaviour will stay. Since that is on life support anyhow I decided to not put more effort into trying to solve conflict problems there that should now in most cases be handled by 1.6's overwrite dialog anyhow.

commit f912adcd (2021-05-20): ✨ Support unicode file names on Py3

If we are on py3 and thus can rely on pathvalidate, we'll now no longer
slugify uploaded files names, but try to keep them as close to the
original name as possible, including unicode and emoji and whatnot. This
requires the underlying file system to support this, so just in case the
old behaviour can be enforced by setting the config flag
feature.enforceReallyUniversalFilenames to true.

Closes #4078