OctoPrint/OctoPrint #4082: [1.6.0-rc1] Unable to start sd card print (gcode stored on printer's sd card)

kazibole, 2021-04-06:
### The problem

Update from OctoPrint 1.5.3 to 1.6.0-RC1. Connect to Prusa MK3S firmware 3.9.3 (also occurred on 3.10.0-RC1). Select gcode from printer's SD card and click "load and print". Nothing happens.

Error in octoprint.log:

```
2021-04-06 12:29:20,501 - octoprint - ERROR - Exception on /api/files/sdcard/TEST~1.GCO [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/util/flask.py", line 1550, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 791, in gcodeFileCommand
    if not _validate(target, filename):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 1224, in _validate
    map(lambda x: fileManager.sanitize_name(target, x), filename.split("/"))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/files.py", line 1224, in <lambda>
    map(lambda x: fileManager.sanitize_name(target, x), filename.split("/"))
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 1018, in sanitize_name
    return self._storage(destination).sanitize_name(name)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/__init__.py", line 1038, in _storage
    "No storage configured for destination {destination}".format(**locals())
octoprint.filemanager.NoSuchStorage: No storage configured for destination sdcard
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0-RC1

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S firmware 3.9.3 (also occurs on 3.10.0-RC1)

### Browser and version of browser, operating system running browser

Windows 10 20H2, Google Chrome 89.0.4389.114

### Checklist of files to include below

octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/6266542/octoprint.log)

foosel, 2021-04-06:
Ah, that's side effect of some API hardening I did, forgot something there... will take a look, thank you for reporting!

foosel, 2021-04-07:
Fix ready for 1.6.0rc2

foosel, 2021-04-14:
1.6.0rc2 is out

commit 54c5cd26 (2021-04-06): 🐛 Fix SD Card printing

Closes #4082
closed 2021-04-14 by foosel