OctoPrint/OctoPrint #4083: [1.5.3] Change api/job to also return 409 Conflict when printer is offline

DarkGuardsman, 2021-04-06:
### Is your feature request related to a problem? Please describe.

In version 1.5.3 calling GET:api/printer on an offline printer will result in `409 CONFLICT`. While calling GET:api/job will result in a data object with a single status text field not null. This is a bit strange as api/printer can't be used to get get the status... but api/job can be used.

### Describe the solution you'd like

In the event api/printer returns `409 CONFLICT` then api/job should also return `409 CONFLICT`. 

### Describe alternatives you've considered

Another option would be to not return 409 from printer endpoint. Then to simply return offline status in the normal object.

foosel, 2021-04-07:
Agreed, but can't change that without a new version of the API as it would be an backwards incompatible change, so that's going to stay like this for the foreseeable future.

DarkGuardsman, 2021-04-07:
An option is to make a v2 API by doing `api/v2/printer`. This is how I manage API migrations at my day job and works decently well.