OctoPrint/OctoPrint #4084: Breaking changes on the job API, float instead of int, doesn't match docs

foosel, 2021-04-08:
> Would still love details on breaking changes you observed in the past however

Sadly I have not kept a list, the one I noticed recently was with time fields moving from integer to float. I assumed it was unintentional and likely a side effect of types in python. Only frustrating part was documentation not matching but guess this is expected for projects 

https://docs.octoprint.org/en/1.5.3/api/datamodel.html#job-information  and 

https://docs.octoprint.org/en/1.5.3/api/datamodel.html#job-information , and 

I'll take a bit more of a look later to see if I can figure out when it changed. 

_Originally posted by @DarkGuardsman in https://github.com/OctoPrint/OctoPrint/issues/4079#issuecomment-814464013_

foosel, 2021-04-08:
After some initial analysis, this seems to actually have been a side effect of switching to python 3/python 3 style division handling. Which based on `git blame` should have happened with 1.3.0. So so long ago in fact that at this point I think we should rather call this a documentation error maybe 🤔

DarkGuardsman, 2021-04-08:
Odd as I had successful JSON mappings using integers not long ago. Any way in OctoPi I can check when I've updated my machines?

cp2004, 2021-04-11:
> Odd as I had successful JSON mappings using integers not long ago. Any way in OctoPi I can check when I've updated my machines?

There's no current way to view the history I believe, not through OctoPrint, not sure about anything on the system (though a log was requested in #4065). Given how long ago 1.3.0 was released ([Blog post, 2016](https://octoprint.org/blog/2016/12/08/new-release-1.3.0/)) I think it is unlikely that you would have updated from something earlier than that any time recently. So 1.3.0 is probably not the change you are looking for. 

Through due to Python 2 EOL I have found it hard to install things like 1.3.12 due to dependencies dropping support. It takes some work, but is doable, if you go installing multiple versions to test.

~~If we are looking at Python 2 vs. 3, it may also be useful to work out if there's an inconsistency based on Python versions under the same OctoPrint version.~~ I checked, it is the same under both.

foosel, 2021-05-18:
I just flashed an ancient OctoPi 0.15.1 with OctoPrint 1.3.8 on board and that *also* has floats there instead of int. So I think we can safely classify that as a documentation bug at this point.

foosel, 2021-05-18:
Correction: I see this back to 1.3.8 for the job information. I *don't* see it for progress data however, neither for 1.3.8 nor for 1.6.1:

[image]

`printTime` and `printTimeLeft` are both integers here. I remember however that PrintTimeGenius might possibly messing with the API responses here, so maybe that is what was happening there for you. I'll make sure to enforce int in such cases.

commit 884ccee5 (2021-05-18): 📝 Fix some doc bugs

Closes #4084
closed 2021-05-18 by foosel