OctoPrint/OctoPrint #4085: [Request] Look into alternative settings dialogs animations

cp2004, 2021-04-08:
### Is your feature request related to a problem? Please describe.

The default Bootstrap 2 implementation of animations are slow. I'm informed they are using JavaScript and jQuery to animate them, which is not working that well.

Most of us have noticed a bit of a slowdown when trying to open the settings dialogs, coming and going. Potentially made worse by the ever-increasing amount of markup in the settings dialog, it is kinda annoying. So I went looking for a solution...

### Describe the solution you'd like

Looking into alternative animations for the settings dialog:

* Disable animations:
  * + : Quick response to clicking the button, and the dialog *appears*
  * - : There is still a slight delay, and the feeling of the settings dialog *just appearing* is strange. Too much of a flash of content at once.

* Use an alternative CSS animation. I like this one, because it is smooth, and customizable. Some animations I tried are below, including a grow animation, and replacing the slide down from BS2 with a CSS animation, and a fade animation.

The CSS animations all start straight away when the element is created, so there is no need for extra logic. The only thing that I haven't figured out is closing the settings, there is no animation on any of them. This would probably be easy enough to implement, but for proof of concept I didn't. 

### Describe alternatives you've considered

See everywhere above and below...

Slide:
[image]

Grow:
[image]

Fade:
[image]

No animation:
[image]

Default animation:
[image]

cp2004, 2021-04-08:
Any comments of which animations are preferred, or a better solution are welcome, I will probably dig about a bit more and come up with a PR eventually.

JohnOCFII, 2021-04-08:
I'm curious if you'll run into issues with consistent behaviors across different browsers and operating systems if you make an animation change.  That could argue to leaving things as they exist today.

Of the animations listed above, my preferences (in order) are:

1. Fade
2. Grow
3. Default animation (when it is working well)
4. No animation
5. Slide

I'm currently viewing this in Safari 14.0.3 on macOS Big Sur 11.2.3

foosel, 2021-04-08:
One vote from my side for fade. Feels the less gimmicky and is almost "no animation" level, but without the issue of having it just suddenly pop up without warning.

cp2004, 2021-04-08:
> I'm curious if you'll run into issues with consistent behaviors across different browsers and operating systems if you make an animation change. That could argue to leaving things as they exist today.

Most of them are very simple CSS animations - and the `animation` property is [quite well supported](https://caniuse.com/css-animation). I can test using Browserstack across some different platforms, to make sure there is nothing surprising. Good point to take into account.

JohnOCFII, 2021-04-08:
> Most of them are very simple CSS animations - and the `animation` property is [quite well supported](https://caniuse.com/css-animation). I can test using Browserstack across some different platforms, to make sure there is nothing surprising. Good point to take into account.

Thanks for that.  My web development skills never got much past using TABLES and a DIV tag...

jneilliii, 2021-04-08:
+1 for fade

KenLucke, 2021-04-13:
I didn't realize others were having this issue for quite some time as well (although I only get it on one of my setups, which I had not been able to track down so I had not reported it yet).  I often get a long, long response to opening the settings, combined with a frequent "A script is slowing down your browser. Wait or Stop It" warning from Firefox itself, especially the first time I open the settings.

I'd vote for fading, if it's faster, or just doing away with the animation/fade altogether (but I'm used to dialogs opening and closing instantly without fanfare), if it still slows things down perceptibly.

PR #4103 (2021-05-04): 💄 Improve all dialog animations
#### What does this PR do and why is it necessary?

Implements #4085, replaces the default slide-in-fade animation with a simple, more performant CSS animation.

#### How was it tested? How can it be tested by the reviewer?

Applying the patch, opening all the dialogs to view the animation. Make sure none of them are using the slide animation....

#### Any background context you want to provide?

For ages, I was annoyed by the speed of the animation, looked into it to try and find the slow thing, which was jQuery & BS2's JavaScript animations.

#### What are the [... cut]