OctoPrint/OctoPrint #4086: Gcode viewer bounding box weirdness

KenLucke, 2021-04-08:
### The problem

First off, let me thank you for implementing the self-adjusting bounding box for the gcode viewer.  It seems to normally work very well (m y only improvement I could suggest would be a user-configurable buffer of X pixels around it so it doesn't actually touch the viewer box)

However, I've found one bit of weirdness in one file (so far): The Bounding box jumps around for a few layers way into the print.  See the video.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Print in progress.

### Version of OctoPrint

1.6.0rc1

### Operating system running OctoPrint

Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.2

### Printer model & used firmware incl. version

n/a

### Browser and version of browser, operating system running browser

Firefox 8.7.0

### Checklist of files to include below

octoprint.log
Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

*I did not have logging turned on for this, so there is no log file.  If I can duplicate it after this print finishes, I will upload a log separately.

Video of issue: https://user-images.githubusercontent.com/57936294/114075577-89203500-985a-11eb-8f6b-cb3e7bacead3.mp4

gcode file too big, dropbox URL: ﻿﻿https://www.dropbox.com/s/jdxnbsx28yl7fwd/Treasure%20-%20Mug%20%26%20Lid%20Threaded.gcode.zip

==========

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:87.0) Gecko/20100101 Firefox/87.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3925282816
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.0.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.0rc1
printer.firmware: Marlin 2.0.7.9_NIC (Mar  6 2021 16:08:01)
systeminfo.generator: systemapi

KenLucke, 2021-04-08:
Here's some more of it at the end of the print (haven't physically gotten to that point yet, just advancing the layers in the viewer).
https://www.dropbox.com/s/300mrozbviskd58/screencast%202021-04-08%2011-24-21.mp4

KenLucke, 2021-04-10:
Another example of auto bounding box weirdness:

https://www.dropbox.com/s/rdxwcf2uyg16m38/froggy-pot.gcode.zip

https://user-images.githubusercontent.com/57936294/114286889-8c145480-9a17-11eb-9842-242122c7c899.mp4

cp2004, 2021-04-11:
There's no issue that I can see with the second file, at least not in the same way that I can see the issue in the first video. Mine behaves exactly the same, cropping the view to the model all the way up.

It does look like you have an issue in the 1st file from the video, but I can't get the browser to load the file at the moment. Going to try BrowserStack to see if it has more resources than my PC I'm working on.... It should be able to cope, but that is a huge file.

KenLucke, 2021-04-11:
I apologize, I forgot that file was cut off at an angle as it rose, which made it appear to be cut off in the viewer as I scrolled upwards, unless I pulled back from "zooming in on the model."  My fault on that one.  I have deleted that post, as it was an error on my behalf.

foosel, 2021-04-12:
I can't get the file to load either, it just crashes the browser. Not surprising given the size. I looked at the video at which z height you are observing the issue and created an abridged file with only some of those layers. Sadly, I cannot reproduce with that file. I'm a bit stumped on how to proceed right now.

Could you test if you see the same thing with the abridged file?

[treasure_mug_lid_abridged.zip](https://github.com/OctoPrint/OctoPrint/files/6295598/treasure_mug_lid_abridged.zip)

KenLucke, 2021-04-12:
Leaving for a crabbing trip for the day in an hour or so, will check when I get back.

foosel, 2021-04-12:
Still haven't been able to reproduce, however I've refactored some parts which might be related, and I've also added a new debug feature to the viewer, you can now enable a layer specific bounding box:

[image]

[image]

Plan is to push out RC2 by Wednesday the latest, if we don't figure this out until then we'll at least have an additional tool for you to enable to help us hopefully see something.

loskexos, 2021-04-12:
Observation ... not sure if that helps: 

the first video shows a really low number of gcode commands for those layers where it's bouncing around. 
E.g. layer 1412 at 141.3mm height shows "39 gcode commands"

If I cut the file "a little" (e.g. crop layers 1-1399, 1470-end) and upload it, then it displays correct - and shows a much higher number of gcode commands for the same layer/height: 141.3mm height: 1732 gcode commands. 

Advancing from observation to speculation .. maybe something is parsed wrongly or overflows with high line count and small amount of retracts per layer? Or some simplification process for finding boundary box is irritated by that scenario?
The mark where it seems to drop off  is at first lines with a E extrusion value specified with less than 5 digit precision (Layer 1412: G1 X120.545 Y156.806 E144.958). No idea why this would be a problem parsing when it's at line 6mn instead line 360k in the reduced file.

foosel, 2021-04-13:
That's a great observation, thank you. I'm wondering if there might be some commands from a completely different location in the file ending up in an intermediate layer or something that's causing issues here (that would be possible the way that the gcode viewer is implemented, as it divides the whole file into layers based on z-height). 

It would be amazing if we had a file to reproduce this with that wasn't 200MB big, that makes analysis extremely difficult 😅 Will see if I can maybe whip up a small CLI analysis tool to get some form of insight of what is going on here.

foosel, 2021-04-13:
Nope, still absolutely no clue. The z-hop detection looks sound, with analysis I also couldn't find anything looking off, and generating some 50 and 100MB variants from head and tail of the file that I *could* then get to load in the viewer also didn't give me a reproduction either.

So at this point I fear we'll have to wait for @KenLucke to run some tests *in safe mode* with his original file as well as a bunch of abridged ones that I've [uploaded here](https://drive.google.com/file/d/15Z26DvAfa04BZj9PAc6VJsvkPFqYY7sE/view?usp=sharing). In the meantime I'll go forward with rc2, which if anything will at least have the aforementioned visualization of the bounding box that might help in analysis going forward.

KenLucke, 2021-04-13:
> So at this point I fear we'll have to wait for @KenLucke to run some tests _in safe mode_ with his original file as well as a bunch of abridged ones that I've [uploaded [...]

I will do some testing today.  My wife's father just passed away, so I may have to drop what I'm doing and help her deal with some stuff, but I'll try to do shat I can.

Not sure why the large file is causing so  many others problems.  I do have problems with very large files (I often print very large, very complex things, files sometimes in the 4-500MB range), but this one is not what I consider that large.  I have my threshold for the viewer set to 200MB, but often I'll override that at it will go ahead and process the file until I get significantly over that limit.  [As a side note, the Pretty Gcode Viewer plugin, which I have disabled for *other* reasons, never had a problem showing any file of any size, so it might be worth looking at how that plugin handled large files better - and faster - for viewing]

Anyway, I'll be doing some testing on that file on my system today, once I wake up a little more to be coherent.

I totally agree that holding up RC2 on account of this minor issue, that only seems to be happening to one file to me, is not a reasonable thing to do.  This is a minor irritant at worst, and seems to have no perceptible physical world consequences, and no reason to make others wait - it can be traced down (or not) for later fixage if necessary.

foosel, 2021-04-13:
> My wife's father just passed away, so I may have to drop what I'm doing and help her deal with some stuff, but I'll try to do what I can.

Completely understandable & my condolences too. Currently going through some grief myself, though "only" for a beloved pet.

>  [As a side note, the Pretty Gcode Viewer plugin, which I have disabled for _other_ reasons, never had a problem showing any file of any size, so it might be worth looking at how that plugin handled large files better - and faster - for viewing]

Absolutely. The viewer was never tailored towards such large file sizes and I didn't write it myself either, I only adapted it. At some point it would be nice to replace it with something else, but that will probably have to be something the community needs to take care off, to many other construction sites already.

I'm frankly astonished that you are able to load 200+ MB files in it, my FF on a Ryzen 7 with 64GB of RAM here simply locks up. Which isn't surprising at all given the data structure the viewer creates for the *whole* file.

KenLucke, 2021-04-13:
> I'm frankly astonished that you are able to load 200+ MB files in it, my FF on a Ryzen 7 with 64GB of RAM here simply locks up. Which isn't surprising at all given the data structure the viewer creates for the _whole_ file.

Buy a Mac :)

OK, I have  made it possibly easier for you guys to work on.  I have now uploaded an ArcWelded version of the file, which exhibits the same issues (starting at layer 1411).  It is only 97.8 MB, so it should load more easily.

As a start, I have also tested Safe Mode and get the same results.

﻿https://www.dropbox.com/s/f9o07rbt9v7mqa0/Treasure%20-%20Mug%20%26%20Lid%20Threaded.ArcWelded.gcode.zip

KenLucke, 2021-04-13:
> Absolutely. The viewer was never tailored towards such large file sizes and I didn't write it myself either, I only adapted it. At some point it would be nice to replace it with something else, but [...]

Maybe ask that plugin developer to actually rework the main viewer?  He seems to have a handle on how to do it.  We don't need the fancies of the PGCV, but the core routines might be adaptable.

foosel, 2021-04-13:
I can reproduce with the arc welder version, so I'll take another shot at this.

> Maybe ask that plugin developer to actually rework the main viewer? He seems to have a handle on how to do it. We don't need the fancies of the PGCV, but the core routines might be adaptable.

Might be an idea indeed.

cp2004, 2021-04-13:
It is the arc welding - the bounding box of the layer is not taking into account the fact that arcs travel beyond their start and end points.

foosel, 2021-04-13:
It's not just that though, @KenLucke also saw this with the non arc welded file and I just `grep`ed that for `G2` or `G3` with zero results. Something about the layer bounding box is definitely fishy though:

[image]

cp2004, 2021-04-13:
I think that the file in the video *is* the arc welded one.

Take layer 1420 for example. In the video, it says that there are 40 commands for each extrusion. That *does not* match with the non-AW file, but it *does* match with the AW file. Haven't counted exactly, but it is far enough difference for me.

Showing the start and end points of the moves would display it well, but I haven't found a gcode viewer that does this on the internet *and* can handle arc welded files.

KenLucke, 2021-04-13:
> I can reproduce with the arc welder version, so I'll take another shot at this.

Good, so it's not just me or my system.

KenLucke, 2021-04-13:
> I think that the file in the video _is_ the arc welded one.
>
> Take layer 1420 for example. In the video, it says that there are 40 commands for each extrusion. That _does not_ match with the non-AW [...]

The video IS from the Arc Welded version, which I was printing at the time.  But the issue originally came to my notice, at the same spot, when I had loaded the original, un-Arc Welded file and, as I always do, scrolled through it before starting the print.  So, at that point, I AW'ed it, and found the same issue, so just went ahead and printed with that, figuring that it was just visual. 

I uploaded the original file here, so that the raw, unchanged data that caused me to spot the problem initially would be what you were looking at, which is why you're seeing the discrepancy.

KenLucke, 2021-04-13:
Another point on the AW/Not AW'ed thing:  I Arc Weld 95% of my files, just for compression, if nothing else.  That, combined with Meat Pack give me usually over 50% compression for transmission, plus a 250000 baud rate, have virtually eliminated blobs and zits up to 100-125mm/s speeds.

If it were due to the G2/G3 commands going outside the bounding box, I'd be seeing this far more frequently than just this one file.

cp2004, 2021-04-13:
> If it were due to the G2/G3 commands going outside the bounding box, I'd be seeing this far more frequently than just this one file.

I was looking at this - since I was wondering how I didn't spot this testing with G2/G3 files. The explanation I came up with was that most files have some sort of infill or multiple arcs, or maybe not a perfect circle - but this file, and a couple I have sliced to test with (linked below) do have this. Also majority of arc commands don't go beyond their starting points by much - but in this case, they are making most of the circle.

[4068-noinfill.aw.zip](https://github.com/OctoPrint/OctoPrint/files/6305423/4068-noinfill.aw.zip)

KenLucke, 2021-04-13:
At this point, I'll leave it in the capable hands that are working on it... there doesn't seem to be much more *I* can do to track it down.  In any case, being merely cosmetic, and apparently extraordinarily rare, it's certainly nothing to hold up RC2 about.

foosel, 2021-04-13:
I'm a bit confused at this point. So did the issue happen with the shared 200 MB non arc version or not? If not, then mystery solved, because the layer bound calculation definitely is currently not arc aware (and neither is the general model size calculation btw). So that'll need fixing, and in two places to bit boot, but that's going to be decidedly not fun.

KenLucke, 2021-04-13:
> I'm a bit confused at this point. So did the issue happen with the shared 200 MB non arc version or not? If not, then mystery solved, because the layer bound calculation definitely is currently not [...]

Yes, it did.

jneilliii, 2021-04-13:
We should see if we could get @FormerLurker to look at the layer bound and model size calculation stuff maybe for arc welded files? Obviously, not the underlying issue here if it does it on the non-arcwelded file, but still maybe something that could be fixed.

KenLucke, 2021-04-13:
My thoughts are:  if the G2/G3 commands are part of the problem, see what can be done on that part alone, and move forward, and if I ever see the issue again unrelated to that, I can open another ticket.  That way, you're not spending time on the issue that could be devoted to [probably] far more important things.

FormerLurker, 2021-04-13:
@jneillii, i will take a look.  Things will need to change to get g2/g3 to work (the entire path and not just the endpoints need to be taken into account), but i have code that may work for G0/G1 that is quite fast.  I will see what I can do.

KenLucke, 2021-04-14:
OK, checking some other files, this pair shows the same issues on the Arc Welded version, but NOT on the original version (which should be small enough to analyze in the viewer, if you override its warning).  It's looking as if the first file may have been a fluke due to the size of that first file, but the issue is actually due to the AW interaction.

Just another dataset to analyze.

﻿https://www.dropbox.com/s/38a5f2p2jxw1fzg/screencast%202021-04-13%2017-30-57.mp4

Both files are in the .zip file
https://www.dropbox.com/s/jtj04bxcj4fgirv/Crab%20Float.zip

FormerLurker, 2021-04-14:
@KenLucke, I will create a quick python function to calculate the x/y extents of G2/G3 as a first step, which shouldn't be all that difficult.  Should solve the issue.

foosel, 2021-04-14:
So... two things. I pushed an additional debugging option with the GCV in 1.6.0rc2 that also renders little crosses at the segment starts:

[image]

That should help in the future with issues like this one, and also should allow @KenLucke to attempt further analysis should he observe the problem in files w/o ArcWelder.

Additionally I tried my hand at correct bounding box calcuation for the viewer in [regressionfix/issue_4086](https://github.com/OctoPrint/OctoPrint/tree/regressionfix/issue_4086) (commit 8b3cebf) and would even have pushed this with 1.6.0rc2 as well, but I couldn't get the math for the max X to work properly in all cases. Crossing the 0/360° axis is where my math is still off I think, and I need some more time (or at least one other pair of eyes) to see what can be done about that.

KenLucke, 2021-04-14:
Quick question:  What does "Sort Layers" in the advanced options do?

foosel, 2021-04-14:
Sort by increasing Z, otherwise you get the order that occurs in the file.

KenLucke, 2021-04-14:
Ah, OK, so that toggles the situations in which you build multiple objects one at a time (all layers per object), rather than layer by layer (whole build plate).

foosel, 2021-04-14:
Yep.

---

Also, I think https://github.com/OctoPrint/OctoPrint/commit/1170914f626e165a1307dd50244e41b7d370a287 solves this issue now. Would love some more pairs of eyes to look on that however. I tested it with the provided arc welded files and also these:

```
G28
G0 X50 Y150 Z0.2
G2 X150 Y150 I50 J-50 E1.0
G1 Z0.4 E2.0
G2 X150 Y50 I-50 J-50 E3.0
G1 Z0.6 E4.0
G2 X50 Y50 I-50 J50 E5.0
G1 Z0.8 E6.0
G2 X50 Y150 I50 J50 E7.0
G1 Z1.0 E8.0
G2 X50 Y149 I50 J-49 E9.0
```

```
G28
G0 X150 Y150 Z0.2
G3 X50 Y150 I-50 J-50 E1.0
G1 Z0.4 E2.0
G3 X50 Y50 I50 J-50 E3.0
G1 Z0.6 E4.0
G3 X150 Y50 I50 J50 E5.0
G1 Z0.8 E6.0
G3 X150 Y150 I-50 J50 E7.0
G1 Z1.0 E8.0
G3 X150 Y149 I-50 J-50 E9.0
```

cp2004, 2021-04-14:
Appears to work on a couple files I had created yesterday, empty cylinders.

KenLucke, 2021-04-14:
Is this in rc2, or do I need to wait for the next RC to test? (I'm not set up to compile from  scratch)

KenLucke, 2021-04-14:
BTW, I appreciate (greatly) the attention, attention to detail, and work you all do, especially on such a nitpicky thing as this that no one else would probably notice for some time.

cp2004, 2021-04-14:
It's not in RC2. Possibly the easiest way for you to temporarily install it would be to use this URL:
```
https://github.com/OctoPrint/OctoPrint/archive/refs/heads/regressionfix/issue_4086.zip
```
That can be loaded to the plugin manager get more > from URL box, or installed from the command line using pip. Note that the version number will show as `0+unknown` since it does not contain the git metadata. To get back to the release branch, you can 'update' again.

It is possible to configure the software update plugin to do that for you, by switching to 'Github Commit' and typing the branch name into it's options.

foosel, 2021-04-14:
Considering that it's not really a big issue, it'll also depend on whether there'll be need for a third RC for another reason whether this will be part of 1.6.0(rc3) or have to wait until 1.7.0.

It also still needs a corresponding fix in the backend but since @FormerLurker hinted at having some python code for this already I'll see what comes from his side first before I try to wrap my head around the problem there as well ^^

FormerLurker, 2021-04-14:
@foosel, I don't actually have the code, but for you I will create it :)  I should have plenty of time to look at this Friday during the afternoon.

KenLucke, 2021-04-14:
> It's not in RC2. Possibly the easiest way for you to temporarily install it would be to use this URL:
> 
> ```
> https://github.com/OctoPrint/OctoPrint/archive/refs/heads/regressionfix/issue_4086.zip
> ```

LOL, I installed that version, and intended to go Arc Weld the file that originally caused me to notice the issue.... ::poof:: - no Arc Welder plugin, didn't even show in my installed plugins (even as disabled), or under "Get More."  I had to unzip the AW'd file on my local drive and re-upload it  :)

That being said, the issue appears to be cured 100% on several files, including the second set that I uploaded as examples.

cp2004, 2021-04-14:
> Edit: AW returned as soon as I reinstalled RC2

I don't know the specifics, but it is probably checking based on the version number whether it can load or not. When you install a specific commit it doesn't contain the metadata, you need a git checkout for that.

KenLucke, 2021-04-14:
> > Edit: AW returned as soon as I reinstalled RC2
> 
> I don't know the specifics, but it is probably checking based on the version number whether it can load or not. When you install a specific commit it doesn't contain the metadata, you need a git checkout for that.

Kinda what I figured

KenLucke, 2021-04-15:
This probably has nothing at all to do with this issue, but it is involving then gcode viewer, so I'll plop a link to it right here:
https://github.com/paukstelis/Octoprint-Cancelobject/issues/67

foosel, 2021-04-16:
On first glance, I don't believe those two are related. 

Tentatively considering this one as solved and merging the fix branch into `staging/maintenance`. If we'll have another RC, it'll be part of 1.6.0, otherwise it'll have to wait until 1.7.0.

foosel, 2021-04-21:
1.6.0rc3 has been released.

commit bf15c805 (2021-04-12): ♻️ Some repetition reduction of the layer bb calc

Possibly fixes #4086, though that is a very long shot.
closed 2021-04-21 by foosel