OctoPrint/OctoPrint #4092: Inaccessible items in tool selection list due to cropping

fasteddy516, 2021-04-13:
### The problem

(Assuming the active printer profile is configured with more than 2 extruders, i.e. 4 for Prusa MMU, 5 for Prusa MMU2, etc.)

1. Navigate to the `Control` tab on the main OctoPrint interface
2. Press the `Switch Tool...` button

The list of available tools appears, but is cut off halfway through `Tool 2`, making `Tool 3` and higher inaccessible.  (As far as I can tell, there is no way to scroll the list or to navigate the list with a keyboard.)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.5.3 (also present in 1.6.0rc1)

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK2/MMU (Firmware 3.2.3)

### Browser and version of browser, operating system running browser

Microsoft Edge 89.0.774.76 (64-Bit), Mozilla Firefox 87.0 (64-Bit), Microsoft Windows 10 Pro 20H2 build 19042.906

### Checklist of files to include below

octoprint.log
Contents of the JavaScript browser console (always include in cases of issues with the user interface)
Screenshots and/or videos showing the problem (always include in case of issues with the user interface)

cp2004, 2021-04-13:
Removing `overflow: hidden` on the control tab fixes it. The only issue is: why was that there in the first place? It's been there over 8 years as well, looking at the git-blame...

https://github.com/OctoPrint/OctoPrint/blob/c3b5991c4d43caee0da1805d548bd59ad38f6d7f/src/octoprint/static/less/octoprint.less#L817-L819

If no one objects, we might as well remove it....?

Visible is set for the more general selectors for the rest of the tabs, so there is maybe something special about the control tab?

https://github.com/OctoPrint/OctoPrint/blob/c3b5991c4d43caee0da1805d548bd59ad38f6d7f/src/octoprint/static/less/octoprint.less#L508-L511

cp2004, 2021-04-13:
Relevant screenshots:
[image]
[image]

LazeMSS, 2021-04-14:
@cp2004  I think it might be related to the sizing issues when adding multiple custom items to control but not sure

cp2004, 2021-04-15:
I'm not a custom controls user, but can probably add some to the UI to give it a test. I may be not thinking it all the way through, but I can't think of the reason you would stick `overflow: hidden` on there. But then again, custom controls allow for additional classes so it may be a safeguard against that?

foosel, 2021-05-07:
> If no one objects, we might as well remove it....?

I just went down a rabbit hole of old code, and I frankly have *no bloody idea* why that is in there. So we might as well get rid of it.

foosel, 2021-05-07:
Gotten rid of it

Pun-e, 2021-07-22:
Hi there, this has recently become a big issue for us.

Is there some workaround we can use in the interim?

commit f69ab865 (2021-05-07): 🐛 Get rid of an unnecessary overflow: hidden

Fixes #4092
closed 2021-10-11 by foosel