OctoPrint/OctoPrint #4093: [RC Feedback] Feedback on the 1.6.0rc2 Release Candidate

foosel, 2021-04-14:
Please provide *general* feedback on your experience with the 1.6.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

  * [ ] #4086 Wrong Gcode viewer bounding box calculation with `G2` & `G3`

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

KenLucke, 2021-04-14:
Install fine, no problems using release channel in Software Update.  Testing now.

Reperiel, 2021-04-14:
All is working fine. I've printed 2 items without problems.

b-morgan, 2021-04-15:
1.6.0rc2 installed. The OctoPrint Bundle Viewer is bungling the connectivity.connection_check line again. Where did I put that issue? :-)

don-willingham, 2021-04-18:
1.6.0rc2 installed, printing, no issues so far.

Sembazuru, 2021-04-19:
Mostly seems to be working fine. But, I have been noticing that when idle it seems to disconnect from my WLAN more often than 1.5 did. Here is the System Info Bundle: 
[octoprint-systeminfo-20210418213258.zip](https://github.com/OctoPrint/OctoPrint/files/6332670/octoprint-systeminfo-20210418213258.zip)

Sembazuru, 2021-04-19:
Ok, soon after I sent the above report I again lost WLAN, this time part way through a print. I couldn't connect with my browser to the OctoPrint UI, nor to port 8080 where one of my 3 cameras lives. I was also unable to connect with WinSCP nor PuTTY. Once the print finished (but before Octolapse completed rendering) I was able to connect with my browser to the OctoPrint UI. Here is a freshly captured System Info Bundle:
[octoprint-systeminfo-20210419025710.zip](https://github.com/OctoPrint/OctoPrint/files/6333894/octoprint-systeminfo-20210419025710.zip)

foosel, 2021-04-19:
WiFi and other network stuff is handled by the underlying OS and not OctoPrint. There's absolutely no way at all that OctoPrint could be making your WiFi connection drop. 

An already existing inherent problem with WiFi stability could be influenced by the number and frequency of connections from running software (including OctoPrint), but that's about it.

Whatever this is, it needs to be fixed outside of OctoPrint.

kazibole, 2021-04-19:
I had a chance to test whether #4082 was resolved. Unfortunately printing from SD card is still not working. I opened #4102 since the symptoms were different this time.

Please let me know if you need any other information.

LightningShark25, 2021-04-19:
Upgrade from previous stable release to RC2 today.  Printer won't connect regardless of settings tried.
On latest stable Marlin (not nightly)

Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x5fce6cb0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Recv: echo:Unknown command: "øøøøøø\x98\x80\x98\x80\x98\x80ààà"
Recv: ok
Changing monitoring state from "Detecting serial connection" to "Operational"
No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /dev/ttyUSB0@115200, /dev/ttyUSB0@250000, /dev/ttyUSB0@230400, /dev/ttyUSB0@57600, /dev/ttyUSB0@38400, /dev/ttyUSB0@19200, /dev/ttyUSB0@9600
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x5fcd0d10, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 230400
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 57600
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 38400
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 19200
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Trying port /dev/ttyUSB0, baudrate 9600
Handshake attempt #1 with timeout 2.0s
Handshake attempt #2 with timeout 2.0s
Handshake attempt #3 with timeout 2.0s
Changing monitoring state from "Detecting serial connection" to "Error"
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

cp2004, 2021-04-19:
@LightningShark25 you seem to have some sort of interference on the serial line there:

>```
>Recv: echo:Unknown command: "øøøøøø\x98\x80\x98\x80\x98\x80ààà"
>Recv: ok
>```

Please downgrade to 1.5.3 again to verify it is not a regression. If the issue persists, then it is not an issue with 1.6.0 and it is coincidence that your setup broke at that moment. If it does turn out to be fixed, go back to 1.6.0 to get the information requested for a full bug report. I was going to link to it, but it appears to be broken for me right now.

LightningShark25, 2021-04-19:
@cp2004 in the process of downgrading - but confident this is directly related to the upgrade, as I did a print immediately before the upgrade.  Will let you know :)

LightningShark25, 2021-04-19:
AND after updating again - the issue doesn't re-appear :S  I'm a little suspicious that enabling the 'error reporting' plugin had something to do with it - but can't get it to happen again.  Didn't touch the hardware at all.

foosel, 2021-04-20:
Pretty sure this was coincidence. There were no changes that would factor into connection, it DOES look like garbage on the serial line, possibly from interference, and error reporting is a completely separate plugin that doesn't interact with the comm layer either.

foosel, 2021-04-20:
Planning to roll out 1.6.0rc3 tomorrow after putting in a work-around for #4102. That means that the fix for #4086 will make it into 1.6.0 after all.

KenLucke, 2021-04-20:
Are you going to leave the diagnostic options for #4086 (bounding boxes, segment starts) in the release version?  I actually really like them :)

foosel, 2021-04-20:
Yeah, those are going to stay

KenLucke, 2021-04-21:
> Yeah, those are going to stay

Nice.

BTW, I have subsequently examined every file I've sliced both with and without arc welder since my initial report, and even files as large or larger than my initial reported file *do not* exhibit the issue #4086 in the unwelded versions, but do in the welded ones.  I suspect that there was something specific about that particular file, and it was just coincidence that I noticed it on that file.

MacShenanigans, 2021-04-21:
First ever GitHub post so please let me know what else I need to do to make this helpful.

I upgraded from 1.5.3 to 1.6.0rc2 this afternoon on Octopi 0.17:
- Creality Ender 3 Max on Creality firmware 1.0.1.6
- upgrade was smooth
- have done a couple of prints of with no issues
- Chrome 89.0.4389.128 and Safari 14.0.2 work fine
- plugins seem to be working fine (although I've found that the Multicam plugin doesn't switch cams under Safari - I'll report that to the plugin owner)
- I'm pretty new to OctoPrint so I'm not really pushing all its features

Nothing to report really other than its still working nicely after upgrade. Thank you!

foosel, 2021-04-21:
@MacShenanigans Thank you, that sounds great! :)

Still, all of you expect another RC to come your way today - after my ISP just finished working on the line I should hopefully have the required internet connectivity to actually go through with this and push out a new RC with the fixes for #4086 and the workaround for the Prusa firmware bug discovered in #4102.

foosel, 2021-04-21:
1.6.0rc3 is out, new ticket is here: https://github.com/OctoPrint/OctoPrint/issues/4106

closed 2021-04-21 by foosel

klapptnix, 2021-08-17:
Updated from 1.6 to 1.7.0rc2 directly and no issue encountered so far after a few prints using Prusa Slic3r and Prusa i3 Mk3S.

klapptnix, 2021-08-17:
Gina, did you really intend to link to this old page (referring to 1.6 release candidates) from https://github.com/OctoPrint/OctoPrint/releases/tag/1.7.0rc2 for general comments?

cp2004, 2021-08-17:
Link should be fixed