OctoPrint/OctoPrint #4094: Tool Tip bug

KenLucke, 2021-04-14:
### The problem

Under VERY specific circumstances, the tool tip that is connected to the Gcode Viewer layer scroll bar can interfere with expanding/collapsing a panel.

The requirements: 
1) Must have the plugin Widescreen installed and active
2) Must have two or more panels on the right, and at least one on the left
3) Must be in the gcode viewer tab
4) Must have the top 2 panels on the right collapsed
5) A file must be loaded (any size), ready to print, in progress, or finished printing
6) The Gcode Viewer must be scrolled to the highest, or very near the highest, layer

Now, attempt to expand the second panel - the tooltip from the scrollbar in the gcode viewer covers the name of the panel to the extent that it's unclickable if it's a short name, even though the mouse is not actually over the toolbar.  Expand the first panel, and the name now clears the tooltip and is expandable.  However, if you play around trying this a bit, THEN, the top panel won't collapse or expand (depending on its state when it turns stubborn).

Discovered this while testing and exploring the other gcode viewer weirdness I reported elsewhere.  Went back a few versions, it's been there a while.  100% reproducible on my system.  If you need logs, please let me know which ones you need.

https://user-images.githubusercontent.com/57936294/114731358-89a34a80-9cf6-11eb-8e75-2e31f77661e5.mp4

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Widescreen plugin not installed when in safe mode.

### Version of OctoPrint

All checked, including up to latest 1.6.0rc2

### Operating system running OctoPrint

see below

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

see below

### Checklist of files to include below

octoprint.log

### Additional information & file uploads

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10.15; rv:87.0) Gecko/20100101 Firefox/87.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3925282816
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 21.0.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.0rc2
printer.firmware: Marlin 2.0.7.9_NIC (Mar  6 2021 16:08:01)
systeminfo.generator: systemapi

jneilliii, 2021-04-14:
I'd consider this a plugin bug, or the need for the plugin to adapt for it. I'm not looking to maintain widescreen anymore though with the introduction of UI Customizer, which has the same functionality plus a whole lot more options. Does this same thing occur with UI Customizer?

KenLucke, 2021-04-14:
I haven't switched to UI Customizer.  I'll do so and see.  I put it in the main OP issues, as it is the tooltip that appears to be causing the problem, and I considered that a function of the main program.

jneilliii, 2021-04-14:
That is an interesting dilemma for sure.

KenLucke, 2021-04-14:
OK, I've installed it and started playing, because stuff is laid out completely differently, there does not seem to be the same issue, but I did find a couple of other ones, I'll go over to the correct plugin to list them.

cp2004, 2021-04-15:
Doesn't look loads like a core issue to me, but more an unfortunate clash of things - the right hand sidebar is not there by default, so who knows what changes the plugins are inserting into the UI to make it work. CSS bugs are common when you start trying to adapt an existing framework. The fact it doesn't occur with UI customizer suggests that there's more than one way of doing this too.

If it *is* a bug with the default styles, that is kinda unfortunate, but probably won't get fixed unless it is a simple, quick adjustment. Bootstrap 2 has a few little bugs like this, and with a look to rework the UI on the horizon with a modern implementation and framework (rather than the same one from 2012) it is unlikely to be much to write home about.

closed 2021-04-15 by cp2004