OctoPrint/OctoPrint #4095: Hot offset for Tool 1 Bug

DesignGuy98, 2021-04-14:
### The problem

I leveled my bed, set a proper hot adjustment for Tool 0 fine, but when I attempted to offset Tool 1 it was too close to the bed. I went into advanced controls and hit the 0.04 adjustment under the side where it says offset is too low. It proceeded to make the bed even closer to the nozzle. I adjusted by 0.04 twice because it wasn't extruding properly yet and I didn't realize it had actually made the bed closer for this reason. It became apparent that it was a bug on the second reattempt when it gouged my bed.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Possibly caused by this issue now that I'm thinking about it. Bed is making a grinding noise randomly when crossing in the center of Y-axis. 

### Version of OctoPrint

0.18.0

### Operating system running OctoPrint

Windows

### Printer model & used firmware incl. version

Makergear M3-ID-S3d

### Browser and version of browser, operating system running browser

Brave

### Checklist of files to include below

octoprint.log

cp2004, 2021-04-14:
Are you sure this isn't an issue with a plugin? I am not sure what you mean by:

> I went into advanced controls and hit the 0.04 adjustment under the side where it says offset is too low.

There is no such menu that I know of in OctoPrint. Please clarify, and also where you checked the box to upload your OctoPrint log you didn't add the file.

And, please test in safe mode. I am confused by the response there, safe mode will not impact your printer.

DesignGuy98, 2021-04-14:
Yes if you go into maintenance>tool 1 hot z offset and run the first print when you scroll down you can click advanced controls and the 0.04mm large adjustment pops up along with several other options including reprinting the same zigzag. This is my first time on here and I didn't realize it was a forum so I figured that she might run a quick test at Octoprint and make the alteration if there is one to be made. The only plugin I have installed is Octoeverywhere, which was recommended when I went into Privacy Policy at Octo, so I made the assumption this was not the issue. When I plugged my printer in this morning to attempt to reconfigure this setting while the bed was moving to heat it made a terrible scratching noise that I have been trying to fix since. That is why I am unable to attempt this in safe mode for now. The same as the Octoprint.log (maybe I am confused and I can do this without sending any sort of G-Code though?) I posted ASAP because I wanted to avoid other people running into the same issue by having the bug fixed because it seems like a simple enough issue. If someone else is capable of running a test though that would be great. Maybe it is the plugin or Makergear makes alterations to Octoprint, but I could see it being a lightly used button and many people not using it for the bug to be caught.

jneilliii, 2021-04-14:
Sounds like something that is installed/distributed with your specific makergear setup/octoprint distribution? I don't know what you're talking about either in regard to `I went into advanced controls and hit the 0.04 adjustment under the side where it says offset is too low.`. Maybe a screenshot of this `advanced controls` section you're talking about would help.

DesignGuy98, 2021-04-14:
Okay. It sounds like it's something related to Makergear's edition of Octoprint then. I will contact them and see if they have something to say about the software. Thank you for the responses. I did not realize I was running somewhat of a different software.

closed 2021-04-14 by DesignGuy98