OctoPrint/OctoPrint #4096: [Request] Select uploaded file

lukemaurer, 2021-04-15:
### Is your feature request related to a problem? Please describe.

When I upload a file, I _always_ want to select it, so it's irritating to have to select it afterward. Worse, it's easy to forget and start printing the wrong file (I've done this more than once).

### Describe the solution you'd like

Uploading a file should cause it to be selected.

### Describe alternatives you've considered

This could be made an option; if it is, IMO it should be on by default.

I suppose another idea would be to warn the user if they upload a file, then print the file that was selected before the upload. Or combine that and the option: have a warning with the options “print the uploaded file and [optionally] always select uploaded files” and “print the other file and [optionally] never warn me again.”

cp2004, 2021-04-15:
The general request is fine, to select a file when it is uploaded. This might make sense where there is no file selected currently, but I can see it annoying the opposite camp. If I am doing multi-part things, uploading multiple files this would annoy me. So it is not a 'one-size-fits-all' approach. Also what do you do if you upload with a print in progress - it becomes inconsistent.

> I suppose another idea would be to warn the user if they upload a file, then print the file that was selected before the upload. Or combine that and the option: have a warning with the options [...]

This sounds to me like an incredibly specific case, and again would probably annoy anyone who did that deliberately. You have to consider both sides.

Not shooting down the request, just pointing out that it is not as clear-cut.

jneilliii, 2021-04-15:
There is a plugin.

https://plugins.octoprint.org/plugins/autoselect/

closed 2021-04-15 by jneilliii