OctoPrint/OctoPrint #4097: After latest upgrade can't get rid of setup wizard

szabocs, 2021-04-16:
PSU Control just prompted me for upgrade, so I did it. 
After reboot I am getting a setup wizard with only a finish button on it.
If i click this button the screen stays and won't go away, basically rendering my 
octoprint installation unusable and I cannot print at the moment.
I will try to manually remove the plugin for now because i need to print.

[image]

cp2004, 2021-04-16:
Not an OctoPrint issue, a plugin issue.

cp2004, 2021-04-16:
You can use the recovery page (add /recovery to your URL) to remove it.

closed 2021-04-16 by cp2004

szabocs, 2021-04-16:
Thanks managed to start the printing. 
Also sorry I didn't realize I was opening the issue on the octoprint git, 
the plan was to do it on the plugin's git.