OctoPrint/OctoPrint #4099: attempt usage of `serial.Serial.set_low_latency_mode(true)` for ASYNC_LOW_LATENCY on POSIX

draeath, 2021-04-17:
### Is your feature request related to a problem? Please describe.

It's somewhat well known that with some devices, stalls/dwells can occur due to USB/RS232 related problems.

### Describe the solution you'd like

It's been suggested to me that the ASYNC_LOW_LATENCY flag is useful in such cases in other usages of USB/RS232. While it seems undocumented, I was able to discover that pyserial [does have a mechanism to set this flag](https://github.com/pyserial/pyserial/blob/f28543b/serial/serialposix.py#L130-L146) that bears investigation. I find that `dir(serial.Serial)` does list this function, so implementation should be a simple as calling `set_low_latency_mode(true)` on your instance of the serial.Serial class.

### Describe alternatives you've considered

Bypassing USB entirely when possible is currently the best option. For example, on a Prusa MK3S I am using GPIO UART to the appropriate pins on the Einsy Rambo J19 header, and the firmware has a setting to enable using this UART instead of USB.

This does require the end user to have some basic electrical engineering type experience or extreme caution when following directions provided by others.

Should an implementation of ASYNC_LOW_LATENCY prove useful, this is a better option for most users, I think.

### Additional context

I would be willing to try to implement this myself to test if it even works, but your codebase is over my head unfortunately and I'm not sure how best to approach that.

Causual testing via the python console certainly does work, though I am unsure if this actually changes anything as I am using said UART bypass:

    >>> import serial
    >>> port = serial.Serial('/dev/ttyAMA0', 115200, timeout=1)
    >>> port.write('M105\n'.encode('ascii'))
    5
    >>> print(port.readline().decode('ascii').strip())
    ok T:21.4 /0.0 B:22.0 /0.0 T0:21.4 /0.0 @:0 B@:0 P:22.1 A:30.7
    >>> port.set_low_latency_mode(True)
    >>> port.write('M105\n'.encode('ascii'))
    5
    >>> print(port.readline().decode('ascii').strip())
    ok T:21.5 /0.0 B:22.1 /0.0 T0:21.5 /0.0 @:0 B@:0 P:22.0 A:30.4

draeath, 2021-04-17:
Some discussion about this flag may be read [here](https://stackoverflow.com/a/46037550) on StackOverflow.

The text:

> I've used a FT232RL based USB adapter on Modbus at 115.2 kbs.
> 
> I get about 5 transactions (to 4 devices) in about 20 mS total with ASYNC_LOW_LATENCY. This includes two transactions to a slow-poke device (4 mS response time).
> 
> Without ASYNC_LOW_LATENCY the total time is about 60 mS.
> 
> With FTDI USB adapters ASYNC_LOW_LATENCY sets the inter-character timer on the chip itself to 1 mS (instead of the default 16 mS).
> 
> I'm currently using a home-brewed USB adapter and I can set the latency for the adapter itself to whatever value I want. Setting it at 200 µS shaves another mS off that 20 mS.

foosel, 2021-05-07:
Definitely worth a shot I guess. The only thing I'm currently unsure about is whether to make it an option, or just default to try to do it and handle any kind of errors gracefully, with an option to NOT try it.

foosel, 2021-05-07:
It became "try to do it, handle errors gracefully and also allow to disable it". Will be part of 1.7.0, thanks for the heads-up and the details in the feature request! 👍

draeath, 2021-05-07:
Awesome, thank you!