OctoPrint/OctoPrint #4101: [Request] G90/G1 overrides relative extruder on for Marlin

dwery, 2021-04-18:
### Is your feature request related to a problem? Please describe.

 In the setting _G90/G91overrides relative extruder mode
This is used by the GCODE analysis in the backend and the viewer in the frontend to interpret your sliced files correctly._
please add an indication that it needs to be on for Marlin

### Describe the solution you'd like

A label _Marlin_ or any text to the same effect.

foosel, 2021-04-26:
Whether this needs to be on or not highly depends on the firmware configuration. I'm not aware that Marlin *always* behaves such that it needs to be enabled. Has there been a recent change in that regard? Both the commit comment that introduced this setting (dba95ee94fac45052e2edaa190161f46abfa2962):

> Handle G90/G91 and M82/M83 distinct by default  
> 
> By default G90/G91 define relativeMode, M82/M83 define relativeE. The extruder coordinate will be viewed as relative when either relativeMode or relativeE are set. This mirrors the implementation in Marlin (current and legacy forks) as well as Repetier.  
> 
> SmoothieWare's implementation resets the relative mode of the extruder on a G90/G91. To still accomodate this different implementation a new settings has been added under Settings > Feature: "G90/G91 overrides relative extruder mode". Checking this switches to the alternative interpretation for future gcode analysis and visualization.  
> 
> The only difference between the two of them is that the following segment of GCODE:  
> 
>     M83
>     G90  
> 
> will have the extruder still set to relative on Marlin and Repetier but set to absolute on Smoothieware.  
> 
> Please note that the setting in all likelihood will be moved to the printer profile in the future, since it makes more sense to have it on a per printer basis. It's not part of the set of features available for auto detection since it's required for GCODE analysis and hence needs to be known even outside of a connection to the printer.  
> 
> Implements #1818

as well as #1818 suggest this used to be a Smoothieware only setting in fact.

dwery, 2021-04-26:
My understanding of https://marlinfw.org/docs/gcode/G090.html is that Marlin's G90 can be overridden with M83, but M83 needs to come after.

cp2004, 2021-04-27:
Additionally, @FormerLurker's ArcWelder plugin checks this, and recommends enabling this option for Marlin.

FormerLurker, 2021-04-28:
For marlin 2+, g90/g92 overrides the extruder.  I've yet to see a fork of 2.x where this is not true.

foosel, 2021-04-29:
Ah, so this is a change that got introduced in Marlin **2**! Thanks for the clarification. I've merged the PR by @cp2004 that adds a "Marlin 2+" label. Just plain "Marlin" would be wrong for the thousands of firmware forks out there that still carry the old behaviour.

Change will be part of 1.7.0

ManuelMcLure, 2021-05-11:
As far as I can tell, in Marlin 1 (1.1.9 at least) there is a global relative/absolute flag controlled by `G90`/`G91` and a separate extruder relative/absolute flag controlled by `M82`/`M83`. Neither flag is modified by the other set of gcodes. If either flag is set to relative, then the extruder will be relative, otherwise absolute. Thus an `M83` will "stick" past a `G90`/`G91`.

||G90|G91
|-|-|-|
|M82|All axes (including E) absolute|All axes (including E) relative|
|M83|Only E relative|All axes (including E) relative|

In Marlin 2, there is one flag per axis - `G90`/`G91` sets all of the flags, and `M82`/`M83` sets only the extruder flag.

PR #4116 (2021-04-29): ✏️  Marlin 2 uses G90 influences extruder
Closes #4101

#### What does this PR do and why is it necessary?
Adjusts label as set out in #4101 

#### How was it tested? How can it be tested by the reviewer?

It wasn't 🙃 

#### Any background context you want to provide?
See #4101

#### What are the relevant tickets if any?
#4101 

#### Screenshots (if appropriate)

#### Further notes
commit 8435039e (2021-04-29): :pencil2: Marlin 2 uses G90 influences extruder

Closes #4101

:art: pretty