OctoPrint/OctoPrint #4102: [1.6.0rc2] Unable to start sd card print (gcode stored on printer's sd card) v2

kazibole, 2021-04-19:
### The problem

Testing resolution of #4082. Connect to Prusa MK3S firmware 3.9.3. Select G-code from printer's SD card and click "load and print". Nothing happens.

Revert back to OctoPrint 1.5.3 and everything functions as expected.

This time there is no error in octoprint.log.

In the 1.6.0rc2 serial.log there is the following:

```
2021-04-19 15:49:35,333 - Send: M20
2021-04-19 15:49:35,365 - Recv: Begin file list
2021-04-19 15:49:35,401 - Recv: TEST~1.GCO 19691
2021-04-19 15:49:35,403 - Recv: End file list
2021-04-19 15:49:35,404 - Recv: ok
2021-04-19 15:49:40,194 - Send: M105
2021-04-19 15:49:40,205 - Recv: ok T:31.8 /0.0 B:33.2 /0.0 T0:31.8 /0.0 @:0 B@:0 P:28.9 A:32.3
2021-04-19 15:49:40,405 - Send: M23 /TEST~1.GCO
2021-04-19 15:49:40,416 - Recv: ok
2021-04-19 15:49:45,195 - Send: M105
2021-04-19 15:49:45,209 - Recv: ok T:31.7 /0.0 B:32.9 /0.0 T0:31.7 /0.0 @:0 B@:0 P:27.8 A:32.7
2021-04-19 15:49:50,196 - Send: M105
2021-04-19 15:49:50,208 - Recv: ok T:31.6 /0.0 B:33.0 /0.0 T0:31.6 /0.0 @:0 B@:0 P:27.8 A:32.2
```

In the 1.5.3 serial.log there is the following:

```
2021-04-19 15:52:04,189 - Send: M20
2021-04-19 15:52:04,219 - Recv: Begin file list
2021-04-19 15:52:04,256 - Recv: TEST~1.GCO 19691
2021-04-19 15:52:04,257 - Recv: End file list
2021-04-19 15:52:04,258 - Recv: ok
2021-04-19 15:52:09,049 - Send: M23 /test~1.gco
2021-04-19 15:52:09,062 - Recv: echo:Now fresh file: /test~1.gco
2021-04-19 15:52:09,076 - Recv: File opened: /test~1.gco Size: 19691
2021-04-19 15:52:09,077 - Recv: File selected
2021-04-19 15:52:09,100 - Changing monitoring state from "Operational" to "Starting print from SD"
2021-04-19 15:52:09,173 - Recv: ok
2021-04-19 15:52:09,176 - Send: M105
2021-04-19 15:52:09,191 - Recv: ok T:30.4 /0.0 B:31.3 /0.0 T0:30.4 /0.0 @:0 B@:0 P:29.0 A:32.4
2021-04-19 15:52:09,195 - Send: M23 /test~1.gco
2021-04-19 15:52:09,203 - Recv: echo:Now doing file: /test~1.gco
2021-04-19 15:52:09,224 - Recv: File opened: /test~1.gco Size: 19691
2021-04-19 15:52:09,225 - Recv: File selected
2021-04-19 15:52:09,244 - Recv: ok
2021-04-19 15:52:09,246 - Send: M24
2021-04-19 15:52:09,264 - Recv: ok
2021-04-19 15:52:09,266 - Changing monitoring state from "Starting print from SD" to "Printing from SD"
```

At a quick glance, 1.6.0rc2 does not issue an M24 G-code, and the capitalization of the filename is different between versions.

Please let me know if you need more information.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.0rc2

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa MK3S firmware 3.9.3

### Browser and version of browser, operating system running browser

Google Chrome 89.0.4389.128
Windows 10 20H2

### Checklist of files to include below

[octoprint_1.6.0rc2.log](https://github.com/OctoPrint/OctoPrint/files/6339094/octoprint_1.6.0rc2.log)
[serial_1.6.0rc2.log](https://github.com/OctoPrint/OctoPrint/files/6339097/serial_1.6.0rc2.log)

[octoprint_1.5.3.log](https://github.com/OctoPrint/OctoPrint/files/6339093/octoprint_1.5.3.log)
[serial_1.5.3.log](https://github.com/OctoPrint/OctoPrint/files/6339096/serial_1.5.3.log)

cp2004, 2021-04-19:
M24 is the missing link here, that starts the print. The capitalisation was wrong in 1.5.x (you can see it is read upper case, then OctoPrint responds lowercase), so that was changed (#3994). However, the relevant change there (https://github.com/OctoPrint/OctoPrint/commit/dc2b63ba81daa3a615ce7682433244ef140af6d9) shouldn't have had an impact here with the M24 command. Will test against the virtual printer.

cp2004, 2021-04-19:
Ah right, because it doesn't respond with this either:
```
2021-04-19 15:52:09,203 - Recv: echo:Now doing file: /test~1.gco
2021-04-19 15:52:09,224 - Recv: File opened: /test~1.gco Size: 19691
2021-04-19 15:52:09,225 - Recv: File selected
```
I guess OctoPrint doesn't send the M24 command, because the file is not selected. I'll leave you to it 😄.

foosel, 2021-04-19:
Yeah, exactly. No confirmation that the file got selected, no M24.

Honestly, if this wasn't such a popular firmware variant, I'd just leave things as they are, this is clearly broken behaviour on part of the firmware and should also be reported as such.

kazibole, 2021-04-19:
Just to provide more info, here is the output of the SD card on my Windows PC, using "dir /x":

G:\>dir /x
 Volume in drive G is FLASHAIR

 Directory of G:\

2020-08-13  10:26 AM    <DIR>                       DCIM
2021-04-06  11:24 AM            19,691 TEST~1.GCO   test.gcode
               1 File(s)         19,691 bytes
               1 Dir(s)  31,828,705,280 bytes free

kazibole, 2021-04-19:
@foosel I will open a bug report at Prusa.

kazibole, 2021-04-19:
@foosel See https://github.com/prusa3d/Prusa-Firmware/issues/3115.

foosel, 2021-04-20:
Work around implemented. OctoPrint will continue to default to sending the file name as received from the firmware, but to work around this Prusa bug there's now also a flag and that even gets auto-applied if Prusa-Firmware is detected on connect. 

If it turns out there's more firmware that doesn't accept file names in the same case as it sends it, this detection can either be extended or the flag default reverted (though I sincerely hope the latter won't be necessary).

foosel, 2021-04-21:
1.6.0rc3 has been released.

commit d7e39288 (2021-04-20): 🥅 Workaround for prusa3d/Prusa-Firmware#3115

If Prusa firmware is detected, fall back old behaviour of lower casing
files received from M20, instead of using them as provided (as
implemented in #3994).

Fixes #4102
closed 2021-04-21 by foosel