OctoPrint/OctoPrint #4106: [RC Feedback] Feedback on the 1.6.0rc3 Release Candidate

foosel, 2021-04-21:
Please provide *general* feedback on your experience with the 1.6.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

kazibole, 2021-04-21:
Tested resolution of #4102 - all good on Prusa MK3S firmware 3.9.3. Thanks!

kazibole, 2021-04-21:
Oh just an FYI, I have another Prusa bug report open for their 3.10.0-RC1 / RC2 firmware - https://github.com/prusa3d/Prusa-Firmware/issues/3102

It occurs on OctoPrint 1.5.3 and 1.6.0-rc2 so I do not believe it is related to OctoPrint.

b-morgan, 2021-04-21:
Updated from 1.6.0-rc2 to 1.6.0-rc3 successfully.

KenLucke, 2021-04-21:
RC3 install went smoothly.  

I only have physical access to one of my printers right now to actually print (everything is a jumble as I build and wire a custom 15' long printer bench/table with drawers and filament storage racks to house all my current printers plus another printer and a resin printer coming in), but I tested the #4086 solution by just loading and scrolling through some files on another printer that is still connected.  Issue seems to be resolved perfectly - thank you!  

Everything else I have tested without actually firing off a print (my one accessible printer is still in the middle of a print for the next 12 hours) looks good.

LightningShark25, 2021-04-21:
Updated from rc2 to rc3.  Early days, but no troubles. Didn't have the serial connection issue this time either :)  Keep up the good work!

b-morgan, 2021-04-23:
I tried to use the VIRTUAL printer and got errors. Have I done something wrong or is this a bug?

[octoprint-systeminfo-20210423075812.zip](https://github.com/OctoPrint/OctoPrint/files/6365817/octoprint-systeminfo-20210423075812.zip)

foosel, 2021-04-23:
Looks like something in the plugin's config being `None` that shouldn't be `None`, so not a bug but rather a misconfiguration. Would need to see all of `plugins.virtual_printer` though to confirm.

In any case, if it IS a bug, it's only in a debugging tool, so it's not severe enough to justify another RC.

b-morgan, 2021-04-23:
I'm not sure what you mean by see all of plugins.virtual_printer. 

If it is a bug, then shouldn't it be fixed in case there is another RC?

foosel, 2021-04-23:
In your `config.yaml`, there is something that looks like this:

``` yaml
plugins:
  virtual_printer:
```

and below that are the settings of the virtual printer. Since mine works just fine and the exception looks like there's some invalid data in the config, what I think is happening here is that something is amiss in that config on your end, and that's causing this instead of a bug.

> If it is a bug, then shouldn't it be fixed in case there is another RC?

If it's a bug, it of course will be fixed. But first we need to figure out if it is a bug and not just a wrong configuration, which I'm leaning towards to right now given that I've been firing up a virtual printer for most of the past week just to do final testing.

b-morgan, 2021-04-23:
I think this is the relevant section:
```
  virtual_printer:
    _config_version: 1
    enabled: true
    resetLines:
    - null
    - null
    - null
    - null
```
Here's the whole thing in case I didn't get the right section (and so you can verify the spacing).

[config.zip](https://github.com/OctoPrint/OctoPrint/files/6366466/config.zip)

foosel, 2021-04-23:
Turn that into:

```
  virtual_printer:
    _config_version: 1
    enabled: true
```

No idea what the `null` values are doing in there, but those are what's causing this. Left over from some experiment on your end maybe?

foosel, 2021-04-23:
Also, don't share your whole config file, I've removed it. There's keys and such in there.

b-morgan, 2021-04-23:
Thanks for looking out for me. Certainly not an experiment on my end. I just use the virtual printer when I'm too lazy to turn the real printer on. Useful for using the GCode Viewer, etc.

I made those changes and all is well, Thanks again.

cp2004, 2021-04-23:
Virtual Printer Settings - make sure you have 1.5.1 installed, known bug with 1.5.0 that caused resetLines to turn into null.

b-morgan, 2021-04-23:
How is that relevant for 1.6.0rc3?

cp2004, 2021-04-23:
The whole conversation about how your virtual printer was broken?

cp2004, 2021-04-23:
Ah right I see a potential confusion, 'Virtual Printer Settings' is a plugin for configurable settings with the virtual printer. Not clear that that was a *plugin*, not core OP.

b-morgan, 2021-04-23:
Charlie, I trust everything you say but don't try to confuse me with the facts ;-)

MacShenanigans, 2021-04-24:
Upgraded from 1.6.0rc2 to 1.6.0rc3 last night (on octopi 0.17), everything seems to be working as normal, have done a couple of fairly long prints. Thank you! (Ender 3 Max firmware 1.0.1.6)

ejhoness, 2021-04-24:
apparently everything wonderful ... thank you very much ...

foosel, 2021-04-27:
1.6.0 has just been released, thank you everyone for helping test the RCs!

closed 2021-04-27 by foosel